ARMmbed/mbed-os #11218: Wi-SUN causing file corruption and assert erros on LittleFS

felipeLeast, 2019-08-14:
### Description

- What target does this relate to?
This issue is related to Wi-SUN and LittleFS

- What toolchain are you using?
I'm using GCC_ARM as toolchain

- What tools are you using?
Mbed-os 5.12.4, mbed compiler and visual studio code as IDE.

- What is the SHA of Mbed OS?
73f096399b4cda1f780b140c87afad9446047432

- Steps to reproduce

Activate Wi-SUN and littleFS, create some files on littleFS and after a while littleFS files will dissapear or get corrupted.

I'm trying to use littleFS on Wi-SUN project but every time I try to use both together LittleFS get's eventually lost. When I comment the Wi-SUN part of the project it goes smooth. I want to know if the Wi-SUN messes with filesystem at some point, or if the queue event from mbed os could cause problems to the littleFS filesystem.

I've tried to protect my files with a mutex and tried to disable irq on the thread that I write on files (which is not possible to do when writing files).

Besides files corruption sometimes littleFS throws an exception regarding block count, lfs type or pcache block. 

### Issue request type

    [  ] Question
    [  ] Enhancement
    [x] Bug

0xc0170, 2019-08-14:
@felipeLeast is there a code snippet to reproduce the issue? Without seeing more in detail we can just guess what goes on in the app.

@ARMmbed/mbed-os-ipcore Any suggestions?

teetak01, 2019-08-14:
Which board is this?

Would this be related to #9845

felipeLeast, 2019-08-14:
> Which board is this?
> 
> Would this be related to #9845

It seems to be this problem exaclty, I've notice that the problem usually occur write after a PAN notification (when the device tries to write on S2LP SPI) which is a different SPI from the on the Flash.

But I'm using nrf52840, and when I saw the fix from #9845 I don't know how to apply it on NRF52840, because any pin could be anything, so que get_peripheral_name won't work, and it's not needed, or at least shouldn't be.

Any tips on how to debug and fix this?

felipeLeast, 2019-08-14:
> @felipeLeast is there a code snippet to reproduce the issue? Without seeing more in detail we can just guess what goes on in the app.
> 
> @ARMmbed/mbed-os-ipcore Any suggestions?

I'm sorry for not being specific, but it's very strange error, it doesn't seem to follow a specific sequence or timeline, it's sporadic and happens on both LittleFS and FatFS, with the difference that LittleFS throws a assert error from time to time.

Seems to be related to the SPI, I just couldn't thing of a way to debug it yet. I've just updated my mbed-os to 5.13.3 to see if there's any fix, but still get errors eventually.

When I comment the Wi-SUN initialization routine the FS works for hours without errors. That's why I think they are related. My Wi-SUN config uses the S2LP in a differente SPI than my SPIFBlockDevice, on the same SPI as SPIF I have, RTC and a Energy Meter, but I don't believe that they are the cause of the problems.

I don't have a code snippet because is a custom board, so you wouldn't be able to test it. But It's would be simple to reproduce, just start [nanostack border router](https://github.com/ARMmbed/nanostack-border-router) and place by the end a thread start which appends 50 bytes into a file every 150ms, when this file reaches 1024 bytes of more start a new file and keep the old one. Eventually when you list you directory entry some files will be gone or with a different name, or missing something.

I'l try to reproduce on some demo board and past the code here.

kjbracey-arm, 2019-08-15:
> But I'm using nrf52840, and when I saw the fix from #9845 I don't know how to apply it on NRF52840, because any pin could be anything, so que get_peripheral_name won't work, and it's not needed, or at least shouldn't be.

The core SPI core uses mutexes to serialise SPI accesses to make sure only one SPI peripheral is using any SPI bus at once.

If the HAL does not provide the spi_get_peripheral_name, then it will assume worst-case, and that your Wi-Sun and SPIF are potentially on the same bus or peripheral, and mutex them against each other.

This could be fatal for Wi-Sun - critical radio timings broken by waiting for slow file accesses. I'm less clear why the file accesses should be bothered by contending for a mutex with the radio.

I believe you can implement #9845 for NRF52 by just having it return the result `pin_instance_spi`, as used in NRF52's `spi_init`.

That would stop those stop the two SPI buses from mutexing against each other.

TuomoHautamaki, 2019-09-12:
Not sure which radio you are using but there's at least one PIN conflict issue on X-Nucleo-S2868A1.
Default SPI SCLK pin configuration is D13. On X-Nucleo-S2868A1, resistor R6 must be soldered instead of R11. For more information, see user manual UM2405. https://www.st.com/resource/en/user_manual/dm00498153.pdf

VeijoPesonen, 2019-12-17:
@kjbracey-arm are this and #11732 related?

kjbracey-arm, 2019-12-17:
It could be related. That SPIFBlockDevice bug would offer be a mechanism for the radio to mess it up.

But if that is the issue, that means the WiSUN device and the storage are on the same bus. That's never going to work well. When the storage is in use, it will be hogging the bus, and the radio won't get access to it fast enough for timeslicing things.

kjbracey-arm, 2019-12-17:
Ah, a comment above says that the radio isn't on the same bus, but two other devices are. It's possible the drivers for the other 2 devices could be breaking storage because of 11732.

felipeLeast, 2019-12-18:
@VeijoPesonen and @kjbracey-arm I´ve seen that issue a couple of days ago, I'll try to apply that fix and come back here. If it fixes I'll close this thread.

closed 2020-04-01 by felipeLeast