ARMmbed/mbed-os #11219: NRF52840 softdevice and cordio

felipeLeast, 2019-08-14:
### Description

- What target does this relate to?
This issue is related to NRF52840 cordio and softdevice

- What toolchain are you using?
I'm using GCC_ARM as toolchain

- What tools are you using?
Mbed-os 5.13.2, mbed compiler and visual studio code as IDE.

- What is the SHA of Mbed OS?
b81aeff1a3e171c6421984faa2cc18d0e35746c0

- Steps to reproduce

I saw in another question that on the latest mbed os version the support for softdevice is beign dropped, I would like to know why is that, I saw that cordio is the new default version on 
`targets.json`. The thing is I need to use NRF52840 with BLE advertising, scaning and GATT services. And I haven't seen how to do it with cordio.

If anyone could help me I would appreciate it.

### Issue request type

    [x] Question
    [  ] Enhancement
    [  ] Bug

0xc0170, 2019-08-14:
cc @ARMmbed/mbed-os-pan

paul-szczepanek-arm, 2019-08-14:
You can do all that with Cordio - look at examples. If you use the user API it doesn't matter what implementation is being used. The Cordio implementation is being used because there is no Softdevice implementation for the new API.

felipeLeast, 2019-08-14:
Hmm, I see, I thougth that cordio was a ble mesh wrapper, but if it has normal BLE functions should work fine for me.

The examples you are refering to are these https://github.com/ARMmbed/mbed-os-example-ble ?

paul-szczepanek-arm, 2019-08-15:
Yes. Cordio in mbed-os refers to the host stack (and possibly also link layer), neither of which are directly called by users who should use the BLE API exposed in `/features/FEATURE_BLE/ble`.

0xc0170, 2019-08-23:
I believe the question was answered, I'll close this issue.

closed 2019-08-23 by 0xc0170