ARMmbed/mbed-os #11228: ARMC6: Link fail's with a  space in project name

chrissnow, 2019-08-14:
### Description

mbed-cli compile with ARMC6 fails if there is a space in the project name.
I have tried on 5.11 and master so likely been around for a while.
to reproduce simply replace a - to a space in "mbed-os-example-blinky"

```
C:\Code\test\mbed-os-example blinky>mbed compile
[mbed] Working path "C:\Code\test\mbed-os-example blinky" (program)
Building project mbed-os-example blinky (LPC1768, ARMC6)
Scan: mbed-os-example blinky
Link: mbed-os-example blinky
[Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
Warning: L3912W: Option 'legacyalign' is deprecated.
Fatal error: L6002U: Could not open file blinky.elf: No such file or directory
Not enough information to list the image map.
Finished: 1 information, 1 warning, 0 error and 1 fatal error messages.
[ERROR] Warning: L3912W: Option 'legacyalign' is deprecated.
Fatal error: L6002U: Could not open file blinky.elf: No such file or directory
Not enough information to list the image map.
Finished: 1 information, 1 warning, 0 error and 1 fatal error messages.

[mbed] ERROR: "C:\Python27\python.exe" returned error.
       Code: 1
       Path: "C:\Code\test\mbed-os-example blinky"
       Command: "C:\Python27\python.exe -u C:\Code\test\mbed-os-example blinky\mbed-os\tools\make.py -t arm -m lpc1768 --source . --build .\BUILD\LPC1768\ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```
.link_options.txt has the following in it, I guess it needs quotes around it. 
`-o ./BUILD/LPC1768/ARM/mbed-os-example blinky.elf
`

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-08-19:
Interesting, have you tested any other exporter if this problem is generic or just ARMC6 related?

0xc0170, 2019-08-19:
cc @ARMmbed/mbed-os-tools

chrissnow, 2019-08-19:
gcc_arm is the same
```
C:\Code\test\mbed-os-example blinky>mbed compile -t gcc_arm
[mbed] Working path "C:\Code\test\mbed-os-example blinky" (program)
Building project mbed-os-example blinky (LPC1768, GCC_ARM)
Scan: mbed-os-example blinky
Link: mbed-os-example blinky
arm-none-eabi-gcc: error: blinky.elf: No such file or directory
arm-none-eabi-gcc: error: blinky.map: No such file or directory
[ERROR] arm-none-eabi-gcc: error: blinky.elf: No such file or directory
arm-none-eabi-gcc: error: blinky.map: No such file or directory

[mbed] ERROR: "C:\Python27\python.exe" returned error.
       Code: 1
       Path: "C:\Code\test\mbed-os-example blinky"
       Command: "C:\Python27\python.exe -u C:\Code\test\mbed-os-example blinky\mbed-os\tools\make.py -t gcc_arm -m lpc1768 --source . --build .\BUILD\LPC1768\GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

mark-edgeworth, 2019-09-11:
This issue is not related to exporters, but to building. I have submitted a PR to fix this now.

closed 2019-09-24 by mark-edgeworth