ARMmbed/mbed-os #11230: Is QSPI's _acquire() necessary?

matthew-macovsky-cypress, 2019-08-14:
### Description

The QSPI driver has an `_owner` member and a private [`_acquire()`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/source/QSPI.cpp#L238) function that sets `_owner` to `this`.  This is similar to the SPI driver, which also has an `_owner` and [`_acquire()`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/source/SPI.cpp#L158).  However, the SPI driver uses these because it has [a static array of `spi_peripheral_s`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/SPI.h#L375), each containing their own `spi_t`, and so it makes sense for the `SPI` object to need to take ownership of a particular `spi_peripheral_s`.  The QSPI driver uses no such mechanism and stores a `qspi_t` directly, so is it still necessary to have a concept of ownership or acquisition in that driver?

Moreover, the first time `_acquire()` is called in the QSPI driver, `_initialize()` will be called, in turn calling `qspi_init()` for the second time (as it was already called in the constructor).  That `qspi_init() ` needs to be repeatedly callable on the same `qspi_t` is not documented in the [QSPI porting documentation](https://os.mbed.com/docs/mbed-os/v5.9/reference/quadspi-port.html), unlike the [Watchdog](https://os.mbed.com/docs/mbed-os/v5.9/reference/watchdog-port.html) or [RTC](https://os.mbed.com/docs/mbed-os/v5.9/reference/rtc-port.html) documentation that explicitly notes this requirement.  Is this behavior intentional?  There is [a comment in the QSPI driver](https://github.com/ARMmbed/mbed-os/blob/master/drivers/source/QSPI.cpp#L79):
> //Otherwise we may have to change mode as well, so call _acquire

but as far as I can tell, `_mode` is only set in the constructor, so there is never a need to update it via `qspi_init()`.

As an aside, there is a protected function [`acquire()`](https://github.com/ARMmbed/mbed-os/blob/master/drivers/QSPI.h#L204) declared in QSPI.h which seems to be neither defined nor used anywhere.

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

kjbracey-arm, 2019-08-15:
I've noted this oddity myself, but I'm not familiar enough with QSPI to know which way to resolve it. 

SPI has all that complication to present the API as one `SPI` object per slave, but internally dealing with multiple slaves sharing one bus and/or SPI peripheral, and handing over ownership. (At least for synchronous transfers - the asynchronous API does not cope with shared peripherals).

Either QSPI should support shared buses/peripherals like SPI does, or if it's not going to, it should be simplified to remove the remaining fragments of that shared peripheral support that were copied from SPI.

0xc0170, 2019-08-19:
> Either QSPI should support shared buses/peripherals like SPI does, or if it's not going to, it should be simplified to remove the remaining fragments of that shared peripheral support that were copied from SPI.

cc @ARMmbed/mbed-os-hal

closed 2019-10-25 by ciarmcom