ARMmbed/mbed-os #11232: Target RTL8195 is not currently enabled

MarceloSalazar, 2019-08-15:
### Description

The target realtek_rtl8195am for platform https://os.mbed.com/platforms/Realtek-RTL8195AM/
Is currently disabled in Mbed OS 5.13.2.
We're getting the message "Target does not support mbed OS 5" when using Mbed CLI

Need to check that the target can be compiled with Arm Compiler 6, GCC, and IAR (https://github.com/ARMmbed/mbed-os/issues/9206) and then re-enable target in Mbed OS once tests pass ok.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

adbridge, 2019-08-15:
@ARMmbed/team-realtek  please investigate

tung7970, 2019-08-15:
RTL8195's release_versions has been emptied out since 5.12 due to lack of IAR8 support. I can send in patches to reenable GCC and ARMCC6 support, and fix a few glitches.

MarceloSalazar, 2019-08-15:
@tung7970 thank you! that'd be appreciated!

tung7970, 2019-08-26:
Pull request #11330

MarceloSalazar, 2019-12-05:
@tung7970 please be aware this platform is not working with Arm Compiler 6 and therefore not supported with the Online Compiler.
It remains as non-supported in the latest version of Mbed OS.

MarceloSalazar, 2020-04-14:
Closing as target won't be supported in Mbed 6 - https://github.com/ARMmbed/mbed-os/pull/12775

closed 2020-04-14 by MarceloSalazar