ARMmbed/mbed-os #11233: Maxim targets are not currently enabled

MarceloSalazar, 2019-08-15:
### Description

The Maxim targets are not currently enabled in Mbed OS 5.13.2+
- MAXWSNENV (MAX32610) - https://os.mbed.com/platforms/MAXWSNENV/
- MAX32620HSP (MAX32620) - https://os.mbed.com/platforms/MAX32620HSP/

We're getting the message "Target does not support mbed OS 5" when using Mbed CLI

Need to check that the target can be compiled with Arm Compiler 6, GCC, and IAR 8 (https://github.com/ARMmbed/mbed-os/issues/9206) and then re-enable target in Mbed OS once tests pass ok.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

adbridge, 2019-08-15:
@ARMmbed/team-maximintegrated please investigate

maclobdell, 2019-09-25:
I see that in targets.json, those two boards have no release version listed.  The MAXWSNENV is listed as end-of-life, so that might be intentional.  

```
        "release_versions": []
```
@jessexm - Can you confirm - are both the MAXWSNENV & MAX32620HSP at end-of-life?  
If so, I propose 1) adding a warning on the platform page indicating end-of-life & which versions of Mbed OS are supported, 2) making the platform page unlisted (I can help with that), 3) keeping support for those targets for ~6 months after EOL notification, then stripping out support.

MarceloSalazar, 2020-05-18:
Closed as targets removed - https://github.com/ARMmbed/mbed-os/pull/12984

closed 2020-05-18 by MarceloSalazar