ARMmbed/mbed-os #11234: Minimal-printf should not bypass the retargetting code

evedon, 2019-08-15:
### Description

This issue was raised as a comment by @kjbracey-arm , see https://github.com/ARMmbed/mbed-os/pull/11051#issuecomment-520315848

minimal-printf should not be trying to work around the retargetting - it should just use fputc.

minimal-printf currently fails with this progam:
```
int main()
{
    puts("puts");
    printf("printf no newline");
}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

hugueskamba, 2019-08-15:
What is the expected result of the sample code and what is the actual outcome?

kjbracey-arm, 2019-08-15:
The puts goes through the C library, hence through line-buffering in stdout, and potentially through buffering in UARTSerial. Or maybe even goes to a totally different device to the serial port.

The printf goes straight to serial_putc, with no buffering. So it potentially gets printed before the puts, or comes out on totally the wrong device.

It can work, but anyone turning on "minimal printf" is forced to figure out how to make sure it's going to the same place as "putc" or "puts" or any other stdout-based output (platform dependent) and to make sure that stdout is non-buffered (toolchain dependent).

A lot of complexity, caused by printf jumping API layers.

Cutting out full retargetting complexity is worthwhile, but just doing it in printf isn't the right way to approach it. It should be a full-system thing, keeping all C output consistent.

PR #11235 (2019-08-29): minimal-printf should not bypass the retargetting code

### Description

minimal-printf should not be trying to work around the retargetting - it should just use fputc.
Fixes #11234

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

@kjbracey-arm 

### Release Notes
closed 2019-08-29 by 0xc0170