ARMmbed/mbed-os #11238: CY8CPROTO-062-4343W - QSPIF

nick-atmosphereiot, 2019-08-16:
### Description

Using latest on master, it seems like QSPI functionality was removed from all of the Cypress kits. Why is this? I'm working with the CY8CPROTO-062-4343W kit, and I would like to use the onboard SPI Flash with LittleFS. 

I've tried making a branch and re-enabling QSPIF and QSPI on this board, but then I am left with compile errors. After fixing those, I end up getting hard faults trying to initialize the QSPIF device. 

Looking at the history, it seems like this may have worked in the past. Why was it removed, and when will it be added again? Without QSPI, I don't think there is any other way to store data persistently. 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

linlingao, 2019-08-16:
@ARMmbed/team-cypress

maclobdell, 2019-09-03:
https://github.com/ARMmbed/mbed-os/pull/11367 should fix this.  It was merged late last week.

nick-atmosphereiot, 2019-09-03:
@maclobdell Yep, works with the latest master. Thanks!

closed 2019-09-03 by nick-atmosphereiot