ARMmbed/mbed-os #11251: nRF52 BLE OTA DFUs in mbed OS >=5.13

BracketJohn, 2019-08-16:
### Description

Target: Nordic NRF52840
Mbed OS: Version 5.13
Related PR: https://github.com/ARMmbed/mbed-os/pull/10709

I would like to perform BLE OTA direct firmware upgrades (DFU) using the nRF52840. Until now this was doable via the Nordic S140 softdevice, for example [by following this guide](https://os.mbed.com/users/kord123/notebook/fota-mbed-app-to-nrf52_dk-with-nordic-dfu-boot-loa/). #10709 changed the  default Bluetooth stack to Cordio. Does that mean, that the S140 can no longer be used with the current version and DFUs are not possible any longer?

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-08-19:
cc @ARMmbed/mbed-os-pan

jrobeson, 2019-10-06:
So am i wrong in thinking that DFU on Nordic devices went from not being supported at all since mbed os 2, then not working out of the box, to not working at all?

BracketJohn, 2019-10-16:
I think you are right! Or at least, the last part is the main question of this thread: Does it not work anymore? If so, is it because of the new BLE stack? If so, will this be fixed/changed?

0xc0170, 2019-10-21:
@donatieng Please review

closed 2019-10-25 by ciarmcom

BracketJohn, 2019-10-27:
Sad to see this moved, but I've re-opened on the forum, as requested: https://forums.mbed.com/t/nrf52-ble-ota-dfus-in-mbed-os-5-13/5801

jrobeson, 2019-10-27:
I'd like this reopened as a bug, but it probably needs to be rephrased.

BracketJohn, 2019-10-27:
Thanks, it's good to see that other people are interested in a resolution of this. I will reopen this as a bug later tonight.

0xc0170, 2019-10-28:
Please do not reopen this as bug reported,

> https://forums.mbed.com/t/nrf52-ble-ota-dfus-in-mbed-os-5-13/5801

This should be sufficient to keep this going. Thanks!

jrobeson, 2019-10-28:
@0xc0170 : this is a pretty huge regression though.

0xc0170, 2019-10-28:
Understood, the stack update contains sometimes missed features from some target specific features (it was not part of the BLE itself as supported). This was the case. 
I don't have an answer to the question if it this support comes to Mbed OS. 
@donatieng Please review

Might not be ideal, you can stay with older version that contains softdevice. I am not certain how much work would be to keep the older stack in and update OS version. It should be doable to leave BLE + HAL component (if needed).

jrobeson, 2019-10-28:
@0xc0170 : didn't it work mostly out of the box in mbed OS 2, while requiring custom work in earlier versions of mbed os 5 before not being supported at all now?

pan-, 2019-10-28:
@jrobeson Not really, it never worked portably across all targets. It was a proprietary extension that made its way through mbed-os. We try to avoid that kind of scenario because if the vendor is not able to maintain it's proprietary extension, we don't have the bandwidth to do it. 
It is exactly what happened in that instance and one of the reason why the softdevice has been removed. 
If BLE DFU makes a comeback, that will certainly be as a component supported by all BLE targets.

cederom, 2019-10-31:
+1 for CORDIO BLE DFU and UART two must haves! :-)

CodingGhost, 2020-02-11:
@0xc0170 any ETA on when this will get Implemented? our Product really needs OTA, and we are stuck on MBED pre cordio until it is possible, I think there are many others that have the same problem.