ARMmbed/mbed-os #11252: ESP8266 RSSI returns '0'

star297, 2019-08-18:
### Description
Nucleo-L476RG, 
Online compiler,
mbed-os-example-wifi
https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-wifi/

Issue:
Returns zero for RSSI after several seconds.
RSSI should be available within 1 - 2 ms of the connected AP 

Other fields, MAC, IP, etc report correctly.

Also the example could use updating with more hardware pin connecting information.
 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x ] Bug

0xc0170, 2019-08-19:
cc @ARMmbed/mbed-os-ipcore

MarceloSalazar, 2019-08-19:
@star297 have you updated the firmware of ESP8266 as described here?
https://github.com/ARMmbed/mbed-os/blob/master/components/wifi/esp8266-driver/README.md

I don't see a reason why this issue would be specifically for the L476. Are you seeing the same problem with other targets?

FYI @ARMmbed/team-st-mcd

star297, 2019-08-20:
My ESP8266 Firmware......

AT version:1.7.0.0(Aug 16 2018 00:57:04)
SDK version:3.0.0(d49923c)
compile time:Aug 23 2018 16:58:12
Bin version(Wroom 02):v1.7.0

I have tried two known working ESP8266 both have had a 'AT+RESTORE' just to make sure.

Three targets gives the same terminal output result.
The RSSI:0 indication takes around 15 seconds after the Gateway information.

Terminal Output......

WiFi example
Mbed OS version 5.13.1

Scan:
Network: Molino secured: None BSSID: 2C:5D:93:9:86:8 RSSI: -87 Ch: 4
Network: Molino secured: None BSSID: B8:8:D7:f8:f5:c4 RSSI: -46 Ch: 5
2 networks available.

Connecting to Molino...
Success

MAC: bc:dd:c2:26:45:95
IP: 192.168.1.16
Netmask: 255.255.255.0
Gateway: 192.168.1.1
RSSI: 0

Done

My mbed_app.json file is......

{
    "config": {
        "wifi-ssid": {
            "help": "WiFi SSID",
            "value": "\"Molino\""
        },
        "wifi-password": {
            "help": "WiFi Password",
            "value": "\"mypassword\""
        }
    },
    "target_overrides": {
        "*": {
            "platform.stdio-convert-newlines": true
        },
        "NUCLEO_L476RG": {
            "target.network-default-interface-type" : "WIFI",
            "esp8266.tx"                        : "D8",
            "esp8266.rx"                        : "D2",
            "esp8266.provide-default"           : true,
            "drivers.uart-serial-rxbuf-size"    : 1024,
            "drivers.uart-serial-txbuf-size"    : 1024
          },
        "NUCLEO_F401RE": {
            "target.network-default-interface-type" : "WIFI",
            "esp8266.tx"                        : "D8",
            "esp8266.rx"                        : "D2",
            "esp8266.provide-default"           : true,
            "drivers.uart-serial-rxbuf-size"    : 1024,
            "drivers.uart-serial-txbuf-size"    : 1024
          },
        "NUCLEO_F767ZI": {
            "target.network-default-interface-type" : "WIFI",
            "esp8266.tx"                        : "D1",
            "esp8266.rx"                        : "D0",
            "esp8266.provide-default"           : true,
            "drivers.uart-serial-rxbuf-size"    : 1024,
            "drivers.uart-serial-txbuf-size"    : 1024
          }
    }      
}

surajdagar, 2019-08-21:
Hi, 

I am also seeing the same issue with nRF52840 platform as well. ESP8266 is my wifi module.

My logs are something like this.

WiFi example
Mbed OS version 99.99.99

Scaning.............:
Network: ClickShare-1872218623 secured: WPA2 BSSID: 28:24:FF:bb:16:39 RSSI: -73 Ch: 1
Network: MOOGGUEST secured: WPA2 BSSID: 0:1A:1E:5e:80:81 RSSI: -90 Ch: 1
Network: !/ secured: Unknown BSSID: 0:0:0:dc:10:0 RSSI: 1 Ch: 0
Network: MOOGGUEST secured: WPA2 BSSID: 0:1A:1E:5e:7f:81 RSSI: -87 Ch: 1
Network: MOOG1951 secured: Unknown BSSID: 0:1A:1E:5e:7f:84 RSSI: -94 Ch: 1
Network: ClickShare-1871835073 secured: WPA2 BSSID: D8:61:62:68:4e:ed RSSI: -74 Ch: 1
Network: *vpvlan* secured: None BSSID: 0:1A:1E:5e:80:60 RSSI: -64 Ch: 6
Network: MOOGGUEST secured: WPA2 BSSID: 0:1A:1E:5e:80:61 RSSI: -61 Ch: 6
Network: mwnetbc secured: WPA2 BSSID: 0:1A:1E:5e:80:63 RSSI: -60 Ch: 6
Network: MOOG1951 secured: Unknown BSSID: 0:1A:1E:5e:80:64 RSSI: -60 Ch: 6
Network: MOOGGUEST secured: WPA2 BSSID: 0:1A:1E:5e:7d:e1 RSSI: -76 Ch: 6
Network: Ü secured: Unknown BSSID: 0:0:0:dc:10:0 RSSI: 1 Ch: 0
12 networks available.

Connecting to mHealth...
Success

MAC: bc:dd:c2:0a:ca:d6
IP: 192.168.43.223
Netmask: 255.255.255.0
Gateway: 192.168.43.56
RSSI: 0

closed 2019-08-29 by 0xc0170