ARMmbed/mbed-os #11256: When TARGET_PSA is enabled in Cortex-A, "SECURE" is redeclared

d-kato, 2019-08-19:
### Description

When TARGET_PSA is enabled on Cortex-A boards such as GR-PEACH and GR-LYCHEE, ``SECURE`` is redeclared.

https://github.com/ARMmbed/mbed-os/blob/1471d2248d49f221675336f5bf4c479a29b88828/cmsis/TARGET_CORTEX_A/core_ca.h#L1833-L1837

https://github.com/ARMmbed/mbed-os/blob/1471d2248d49f221675336f5bf4c479a29b88828/features/frameworks/TARGET_PSA/val.h#L171-L174

Since ``core_ca.h`` is a CMSIS file, changing it will have a big impact. On the other hand, ``val.h`` is a file used only by Mbed. And security_t itself is used in ``mbed-os\components\TARGET_PSA\TESTS``, but the ``SECURE`` string is not used, so I think the effect of the change is relatively small.

Can we change ``SECURE`` in ``val.h`` to another name as follows? 
``` .cpp
typedef enum {
    PSA_NONSECURE = 0x0,
    PSA_SECURE    = 0x1,
} security_t;
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-08-19:
cc @ARMmbed/mbed-os-psa Please review

Patater, 2019-08-20:
@d-kato This needs to be changed in the upstream source repository, and Mbed OS can take in the change from there. Please [raise a PR to ARM-software/psa-arch-tests](https://github.com/ARM-software/psa-arch-tests/pulls) (relevant code is https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/val/common/val.h#L169-L172).

Thanks!

d-kato, 2019-08-20:
@Patater Thank you. I done PR below.
https://github.com/ARM-software/psa-arch-tests/pull/109

d-kato, 2019-08-22:
It will be solved when the following PR is merged with mbed-os.
https://github.com/ARM-software/psa-arch-tests/pull/111

d-kato, 2019-10-01:
Please merge this change into mbed-os. https://github.com/ARM-software/psa-arch-tests/pull/111

toyowata, 2019-10-01:
@Patater bump

Patater, 2019-10-02:
We are working on a PR to bring in these changes.

jainvikas8, 2019-10-22:
PR merged, please close the issue.

closed 2019-10-22 by 0xc0170