ARMmbed/mbed-os #11257: ARM_MUSCA_A1_NS: fails to build with ARMC6 - internal fault

0xc0170, 2019-08-19:
### Description

Latest logs can be found: http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/313/build-ARM/FAIL/

mbed-os-example-wifi example, linker error:

```
Compile [ 99.9%]: rtx_thread.c
Compile [100.0%]: ThisThread.cpp
Link: mbed-os-example-wifi
[Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
Warning: L3912W: Option 'legacyalign' is deprecated.
Internal fault: [0xb3b91b:6110004]
Please contact your supplier.
[ERROR] Warning: L3912W: Option 'legacyalign' is deprecated.
Internal fault: [0xb3b91b:6110004]
Please contact your supplier.

```

I can't find artifacts for the job adding this target in #9221 . @ARMmbed/mbed-os-psa Would you be able to help to reproduce this and report to the ARMC6 team if this is valid internal fault (small chance this could be our setup fault).

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-08-20:
cc @gaborkertesz

0xc0170, 2019-08-21:
I can't reproduce locally.

```mbed compile -m ARM_MUSCA_A1_NS -t ARM -v

[DEBUG] Errors: Finished: 0 information, 1 warning and 0 error messages.
Elf2Bin: mbed-os-example-wifi
FromELF: C:\Programs\ARMCompiler6.11\bin\fromelf --bin -o .\BUILD\ARM_MUSCA_A1_NS\ARM\mbed-os-example-wifi.bin .\BUILD\ARM_MUSCA_A1_NS\ARM\mbed-os-example-wifi.elf
[DEBUG] Return: 0
Post-Build: mbed-os-example-wifi
[DEBUG] Secure image file found: .\mbed-os\targets\TARGET_ARM_SSG\TARGET_MUSCA_A1\TARGET_MUSCA_A1_NS\prebuilt\tfm.bin.
**[INFO]** Image version number set to 1.0.0+0
**[INFO]** Writing load address from the macro in flash_layout.h to the image header.. 0x10020000 (dec. 268566528)
| Module              |     .text |   .data |      .bss |
|---------------------|-----------|---------|-----------|
| [lib]\c_w.l         | 10977(+0) |  16(+0) |   348(+0) |
| [lib]\fz_ws.l       |     8(+0) |   0(+0) |     0(+0) |
| [lib]\libcppabi_w.l |    44(+0) |   0(+0) |     0(+0) |
| [lib]\m_ws.l        |    40(+0) |   0(+0) |     0(+0) |
| anon$$obj.o         |    82(+0) |   0(+0) | 52768(+0) |
| main.o              |  1075(+0) |   0(+0) |     4(+0) |
| mbed-os\components  |   122(+0) |   0(+0) |    33(+0) |
| mbed-os\drivers     |   670(+0) |   0(+0) |     0(+0) |
| mbed-os\features    |   197(+0) |   0(+0) |     0(+0) |
| mbed-os\hal         |  3352(+0) |   8(+0) |   248(+0) |
| mbed-os\platform    |  4308(+0) |  64(+0) |   256(+0) |
| mbed-os\rtos        |  9266(+0) | 168(+0) |  6626(+0) |
| mbed-os\targets     |  3132(+0) |  40(+0) |    77(+0) |
| Subtotals           | 33273(+0) | 296(+0) | 60360(+0) |
Total Static RAM memory (data + bss): 60656(+0) bytes
Total Flash memory (text + data): 33569(+0) bytes

Image: .\BUILD\ARM_MUSCA_A1_NS\ARM\mbed-os-example-wifi.bin
```

0xc0170, 2019-08-21:
cc @mbed-os-test - Can you help us to reproduce this issue?

My version of ARMCC is : 6.11

kimlep01, 2019-08-21:
I can reproduce the same error with Ubuntu 16 but in Windows the example compiles without issues. Toolchain version in Ubuntu:
```
armclang --version
Product: ARM Compiler 6.11 Professional
Component: ARM Compiler 6.11
Tool: armclang [5d3b4200]
```

0xc0170, 2019-08-22:
@kimlep01 Thanks ! that would explain why I can't - I am on windows machine currently. 

@ARMmbed/mbed-os-test Can we share this failure with details to the compiler team ? Please do.

devran01, 2019-08-22:
@0xc0170 @kimlep01 It looks like the issue is fixed in 6.12. I just built mbed-os-example-wifi using `mbed compile -m ARM_MUSCA_A1_NS -t ARM` and it builds successfully.

OS: Ubuntu 16
Product: ARM Compiler 6.12 Ultimate
Component: ARM Compiler 6.12
Tool: armclang [5d624a00]

0xc0170, 2019-08-23:
Great, we will update to 6.12 in the upcoming days. Hopefully we can close this one with an update!

0xc0170, 2019-11-21:
I don't see this failure in the recent nightlies, the update for ARM was done.

Closing as resolved

closed 2019-11-21 by 0xc0170