ARMmbed/mbed-os #11263: exporters: profiles being ignored

0xc0170, 2019-08-20:
### Description

We identified exporters have hard coded optimizations and also c++ standard, not following what we have in profiles. There are functions in exporters for checking valid flags, see https://github.com/ARMmbed/mbed-os/pull/11212#issuecomment-522530614

One of  them in uvision.py implementation:

```
        def valid_flag(x):
            return (
                x not in in_template and
                not x.startswith("-O") and
                not x.startswith("-std") and
                not x.startswith("-D")
            )
```

There was an intention there. I don't recall the reason (could be related to the online enviroment?). I don't think this is correct behavior. If profile is not followed, a project via exporter and built via cli are different - it can lead to unexpected results in an application.

Ideally, the exporter project would include profile settings.

cc @ARMmbed/mbed-os-tools @kjbracey-arm 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

mark-edgeworth, 2019-08-20:
This one is actually in tools/export/uvision/__init__.py. Sadly we are aware of lots of issues like this.

0xc0170, 2019-08-20:
> tools/export/uvision/init.py

Correct the code snippet taken from there. Similar code can be found in others (I checked IAR and uvision and gnuarmeclipse) but not in cmake for instance (I suspect cmake and make are taking profiles?).

adbridge, 2020-10-15:
As this applies to the Mbed 5 tools which are no longer supported and Mbed 6 tools will address exporters in a completely different way, I am closing this. Once the Mbed 6 tools are being used please then check if the same issue still exists.

closed 2020-10-15 by adbridge