ARMmbed/mbed-os #11277: FPGA shield uart test

wajahat-ublox, 2019-08-21:
### Description
The FPGA uart test case "9600, 8O1, FC on" enables the odd parity and then checks if data sent by tester is received correctly by DUT uart. This test passes for K64F however it is failing for Ublox C030 targets.
The C030 boards use STM32F4 as MCU and according to the reference manual of this MCU, when parity is enabled, the MSB of value read from Data Register includes the parity bit. So in case of this MCU, the test case should mask the value read with 0xFF before doing the comparison.

Please consider this example:
Tester wrote 0x04 on uart.
Value read from DR is 0x04, the comparison succeeds because in this case the parity bit is set to 0 as parity of data byte is already odd.

Now the written value is incremented:
Tester writes 0x05.
Value read from DR is 0x105, the test asserts here as 0x05 != 0x105 however this is happening because the value in DR includes the parity bit which is now set to 1 as parity of data byte is not odd.

I would also like to know if this behavior is observed for any other targets with ST MCU. I tried on Nucleo-F429ZI but it seems the Arduino header does not have CTS/RTS pins so this test passes with following message:
[1566299117.84][CONN][RXD] Could not find pins for UART testing 

@RobMeades @MarceloSalazar @fahim-ublox @c1728p9

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

MarceloSalazar, 2019-08-21:
FYI @jamesbeyond

0xc0170, 2019-08-21:
cc @ARMmbed/mbed-os-hal

fkjagodzinski, 2019-08-21:
@wajahat-ublox Thanks for a detailed description. :+1:

Looking at the ST HAL and [the Table 146. of STM32F42xxx RM](https://www.st.com/content/ccc/resource/technical/document/reference_manual/3d/6d/5a/66/b4/99/40/d4/DM00031020.pdf/files/DM00031020.pdf/jcr:content/translations/en.DM00031020.pdf#%5B%7B%22num%22%3A1621%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C67%2C754%2Cnull%5D), I can see that the influence of the *PCE bit* on the actual number of data bits has been taken into consideration in `serial_format()` implementation:

https://github.com/ARMmbed/mbed-os/blob/590ce2c958dfcf25759d4eebe2405ce824c2cf2f/targets/TARGET_STM/serial_api.c#L426-L455

but `serial_getc()` seems to be missing this logic:

https://github.com/ARMmbed/mbed-os/blob/590ce2c958dfcf25759d4eebe2405ce824c2cf2f/targets/TARGET_STM/TARGET_STM32F4/serial_device.c#L271-L275

@LMESTM, @jeromecoutant, it seems we should update the `serial_getc()` implementation to make use of `obj_s->databits` for the bitmask calculation. Could you comment on that, please?

wajahat-ublox, 2019-08-22:
@fkjagodzinski Yes i think both `obj_s->databits` and `obj_s->parity` need to be used to get the bitmask. `obj_s->databits` alone will only give total bit length (7, 8 or 9) and without knowing if parity is enabled or not we will not get the correct mask.

0xc0170, 2019-08-22:
> @LMESTM, @jeromecoutant, it seems we should update the serial_getc() implementation to make use of obj_s->databits for the bitmask calculation. Could you comment on that, please?

cc @ARMmbed/team-st-mcd

jeromecoutant, 2020-03-10:
Please check #12611 
Thx

PR #12611 (2020-03-13): STM32F4 UART issue when parity enabled
### Summary of changes 

Fixes #11277

See STM32F429 Reference Manual:
Bits 8:0 DR[8:0]: Data value
When receiving with the parity enabled, the value read in the MSB bit is the received parity
bit.

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type 

    [x] Patch update (Bug fix / Target update / Docs update / Test update / Refactor)
    [] Feature update (N
closed 2020-03-13 by adbridge