ARMmbed/mbed-os #11285: Compiler error -230: FRDM k64F  with STBC-AGM01

NiranjanRavi1993, 2019-08-21:
Hi,

FRDM with  STBC-AGM01:

I was working with FRDM k64F board and when I tried to use ecompass example using FRDM-STBC-AGM01 I am getting a compiler error -230. 
The error repeats if try to import different program related to ecompass . Whereas I am able to run other programs in my compiler such as blinky led and gpio programs.  How to fix this issue?
   
    Bug

0xc0170, 2019-08-22:
Can you provide references to  the programs /libraries used - to be able to reproduce this issue?

NiranjanRavi1993, 2019-08-22:
Sure. 
"FXOS8700Q.h", "FXAS21000.h" are libraries used and mbed.h (latest library version). I also noticed if I revert mbed version to old revisions from 2017, code compiles.  
Attached the file as well for your reference.

Thank you
Thanks and Regards
Niranjan
[FRDM-STBC-AGM01.zip](https://github.com/ARMmbed/mbed-os/files/3530696/FRDM-STBC-AGM01.zip)

0xc0170, 2019-08-23:
I imported your zip, used KL25Z as testing target and also K64F. All compiles. With library revision 94  and also the latest one - no problem. I can't reproduce the issue.

0xc0170, 2019-08-23:
One thing I noticed in the zip - there's mbed folder containing k64f target files and also mbed lib file , why?

0xc0170, 2019-10-22:
@NiranjanRavi1993 Please reopen with an update. I'll close this as there has not been any update for 2 months

closed 2019-10-22 by 0xc0170