ARMmbed/mbed-os #11286: [Cellular]: Multiple UART connections to module

trowbridgec, 2019-08-21:
### Description

We are working with a board that has 2 UART connections to a cellular module. Using 1 UART is relatively straightforward, but as we've discovered, in order to send AT commands to the module while in data mode, you need a separate UART for communication (see this issues https://github.com/ARMmbed/mbed-os/issues/8666#issuecomment-486565413).

From my understanding, it looks like you can just use the static `AT_CellularDevice::get_default_instance()` function to get a reference to the `AT_CellularDevice` object then and use the `get_at_handler()` function which accepts a `FileHandle` (i.e. the 2nd UART). Is this the preferred method for working with 2 UARTs? Are there any examples available of someone using this approach?

Also, I assume that it'd be a good idea to release the 2nd `ATHandler` when not in use.

Thanks!

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

loverdeg-ep, 2019-08-21:
@ARMmbed/mbed-os-wan

loverdeg-ep, 2019-08-22:
@maclobdell

AriParkkila, 2019-08-23:
@trowbridgec Could call `ATHandler::set_file_handle` to assign the second UART after PPP has been connected.

closed 2019-10-25 by ciarmcom