ARMmbed/mbed-os #11287: Can we support QSPIFBlockDevice if memory does not support SFDP  ?

LMESTM, 2019-08-22:
### Description

The QSPIFBlockDevice driver relies on SFDP parameters to adapt its commands to the connected QSPI Flash. 

But we have a memory Macronix MX25LM51245G that does not have programmed SFDP parameters so returns 0xFF ... Is there a way to support this ? Maybe hard coding default parameters ? 

cc @ARMmbed/mbed-os-storage    
 

### Issue request type

    [ x] Question
    [ ] Enhancement
    [ ] Bug

jeromecoutant, 2019-08-22:
@MarceloSalazar

DBS06, 2019-08-22:
@LMESTM the datasheet you provided mention in chapter 11

> The Serial Flash Discoverable Parameter (SFDP) standard provides a consistent method of describing the functional
and feature capabilities of serial flash devices in a standard set of internal parameter tables. These parameter tables
can be interrogated by host system software to enable adjustments needed to accommodate divergent features
from multiple vendors. The concept is similar to the one found in the Introduction of JEDEC Standard, JESD68 on
CFI.
The sequence of issuing RDSFDP instruction in SPI is CS# goes low→send RDSFDP instruction (5Ah)→send 3
address bytes on SI pin→ send 8 dummy cycles → read SFDP code on SO→to end RDSFDP operation can use
CS# to high at any time during data out.
SFDP in SPI is a JEDEC standard, JESD216.
The sequcn of issuing RDSFDP instruction in OPI/DOPI mode:
CS# low → send RDSFDP instruction (5Ah/A5h) → send 4 address bytes on SIO pin→ send 20 dummy cycles →
read SFDP code on SIO[7:0] → to end RDSFDP operation can use CS# to high at any time during data out.

According to datasheet your flash is a SFDP-Flash, but if you read "chapter 1/General/ Protocol Support" you will see that your flash only support SPI and Octa-SPI, but does not support Quad-SPI.

LMESTM, 2019-08-22:
@DBS06 
we asked Macronix and they said that this part MX25LM51245G does not support SFDP yet. 
they said they'll be adding support on it in the future. 

Yes it supports only single IO / Octal IO mode. I'm working on adding QSPI support for this part in Single IO fallback mode.

jeromecoutant, 2019-08-30:
@ARMmbed/mbed-os-storage

LMESTM, 2019-09-09:
@ARMmbed/mbed-os-storage @MarceloSalazar  - we would appreciate some feedback on this question - thanks

MarceloSalazar, 2019-09-10:
@SeppoTakalo can you please review?

SeppoTakalo, 2019-09-16:
At this time, due to our prioritisation of work on customer projects, we are unable to devote the time to fix this issue. As an open source project, we welcome fixes to Mbed OS, so if would like to contribute a fix yourself then we would very much welcome that, please see our contribution guidelines here - https://os.mbed.com/docs/mbed-os/v5.13/contributing/workflow.html#contributions

closed 2019-10-25 by ciarmcom