ARMmbed/mbed-os #11295: QSPIFBlockDevice does not differentiate between mode (alt) and dummy cycles

kyle-cypress, 2019-08-23:
### Description

When sending commands to the QSPI hal, QSPIFBlockDevice treats mode (alt) cycles as though they were dummy cycles. This causes problems on the PSOC6 targets where the data pins are tri-stated during the dummy cycles. This means that if the last value sent prior to the dummy cycles happens to also be a valid mode value for the memory in use, the memory part can wind up in an unexpected state.
For example, in QSPIFBlockDevice::_sfdp_detect_best_bus_read_mode:
```C++
_dummy_and_mode_cycles = (basic_param_table_ptr[QSPIF_BASIC_PARAM_TABLE_444_READ_INST_BYTE - 1] >> 5) 
      + (basic_param_table_ptr[QSPIF_BASIC_PARAM_TABLE_444_READ_INST_BYTE - 1] & 0x1F);
```

QSPIFBlockDevice should retain the mode cycle and dummy cycles as separate values and pass them separately to the QSPI hal driver (which already supports mode/alt bits as a separate entity).

The problem was reproduced using features-storage-tests-kvstore-static_tests built with GCC_ARM for CY8CPROTO_062_4343W, while attempting to test the addition of QSPI support for PSOC6 targets.

I will submit a pull request shortly with a proposed fix.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-08-23:
+1 for the issue and fix

VeijoPesonen, 2019-12-17:
@kyle-cypress This issue should be closed I assume.

kyle-cypress, 2019-12-17:
Yes, this is addressed by #11602. Thanks.

closed 2020-03-02 by SeppoTakalo