ARMmbed/mbed-os #11305: Porting NUCLEO-F767ZI make errors

bytesontherocks, 2019-08-23:
### Description

### Issue request type

0xc0170, 2019-08-23:
cc @ARMmbed/team-st-mcd Please review

bytesontherocks, 2019-08-28:
Any chance to have a look at the issue? Any clue how to tackle it?
Thanks!

0xc0170, 2019-08-28:
> cc @ARMmbed/team-st-mcd Please review

Can you review this one?

bytesontherocks, 2019-08-28:
Hi thanks for your reply, 

And yes, sorry, I did try to remove the unfound path (the doubt was if it was really necessary).

However, you can see that there are compilation errors, it seems related to the target macros...

Any suggestions?

[image]

bytesontherocks, 2019-08-28:
Hi, I think I'm reformulating my question: I'm working with STM32IDE and STM32F767ZI Nucleo board. What is the best way to get mbed up and running without having to start fighting the system configuration, etc?

Because when I try to export one example from mbed for Sw4STM32 it doesn't work.

Thanks.

LMESTM, 2019-08-28:
Hi,

Not sure how to help. can you share a snapshot of how the structure project looks like in the IDE ? 

If the IDE tries to compile "mbed-os-example-blinky/usr/src/mbed-sdk", then it must be some path defined in your project. As it does not exist, please try to remove this path from your project structure ?

LMESTM, 2019-08-28:
Unfortunately there is no support as of now for exporting a mbed example directly to STM32CubeIDE. You'd have to go thru the process you've started and adapt the generated project settings. 
You may use a  project example from STM32CubeF7 firmware package to check the reference project settings.

closed 2019-08-28 by bytesontherocks