ARMmbed/mbed-os #11310: NUMAKER_PFM_M2351 doesn't compile with GCC, ARMC5, IAR

MarceloSalazar, 2019-08-23:
### Description

I've tested the [mbed-os-example-blinky](https://github.com/ARMmbed/mbed-os-example-blinky) example application with latest Mbed OS (5.13.3 at this time).

The mbed tools show:
-  Target NUMAKER_PFM_M2351 is not supported by toolchain GCC_ARM
-  Target NUMAKER_PFM_M2351 is not supported by toolchain IAR
-  Target NUMAKER_PFM_M2351 is not supported by toolchain IAR

However, these toolchains have to be supported to be compliant with Mbed Enabled

Related to this:
https://github.com/ARMmbed/mbed-os/issues/10898

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-08-23:
I believe this is the only target with an exception.

@ARMmbed/team-nuvoton When IAR and GCC ARM are going to be supported?

ccli8, 2019-08-28:
Toolchain support on M2351:

- ARMC5 doesn't support ARMv8M, so it is excluded. Support ARMC6 instead.
- GCC support has been on-going. It is not enabled because there's something wrong in GCC-compiled secure function code.
- IAR support is not scheduled currently. Maybe after #10959 lands so that related changes can be clear.

MarceloSalazar, 2020-05-15:
Closing as the m2351 is not enabled in master for Mbed OS 6.0.
We'll retest when reintroduced. It will have to work with gcc 9 and arm compiler 6.

closed 2020-05-15 by MarceloSalazar