ARMmbed/mbed-os #11312: MTB_MXCHIP_EMW3166 module doesn't compile with Arm Compiler 5

MarceloSalazar, 2019-08-23:
### Description

I've tested the [mbed-os-example-blinky](https://github.com/ARMmbed/mbed-os-example-blinky) example application with latest Mbed OS (5.13.3 at this time).
The application doesn't compile with Arm Compiler 5.

```
Link: mbed-os-example-blinky
Error: L6218E: Undefined symbol WICED_EMAC::get_instance() (referred from BUILD/MTB_MXCHIP_EMW3166/ARMC5/mbed-os/targets/TARGET_WICED/wiced_interface/default_wifi_interface.o).
Error: L6218E: Undefined symbol WicedInterface::WicedInterface(EMAC&, OnboardNetworkStack&) (referred from BUILD/MTB_MXCHIP_EMW3166/ARMC5/mbed-os/targets/TARGET_WICED/wiced_interface/default_wifi_interface.o).
Error: L6218E: Undefined symbol VTT for WicedInterface (referred from BUILD/MTB_MXCHIP_EMW3166/ARMC5/mbed-os/targets/TARGET_WICED/wiced_interface/default_wifi_interface.o).
Finished: 0 information, 0 warning and 3 error messages.
[ERROR] Error: L6218E: Undefined symbol WICED_EMAC::get_instance() (referred from BUILD/MTB_MXCHIP_EMW3166/ARMC5/mbed-os/targets/TARGET_WICED/wiced_interface/default_wifi_interface.o).
Error: L6218E: Undefined symbol WicedInterface::WicedInterface(EMAC&, OnboardNetworkStack&) (referred from BUILD/MTB_MXCHIP_EMW3166/ARMC5/mbed-os/targets/TARGET_WICED/wiced_interface/default_wifi_interface.o).
Error: L6218E: Undefined symbol VTT for WicedInterface (referred from BUILD/MTB_MXCHIP_EMW3166/ARMC5/mbed-os/targets/TARGET_WICED/wiced_interface/default_wifi_interface.o).
Finished: 0 information, 0 warning and 3 error messages.
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-08-23:
cc @ARMmbed/mbed-os-ipcore

0xc0170, 2019-08-23:
Looking at WICED libraries, there is no ARMC5 prebuilt binary, only ARMC6. Isn't this intentional?

See https://os.mbed.com/forum/bugs-suggestions/topic/35215/?page=1#comment-61959 .

AnttiKauppila, 2019-08-26:
@SeppoTakalo Can you comment on this?

MarceloSalazar, 2020-04-14:
Arm Compiler 5 is not supported - https://os.mbed.com/blog/entry/Removal-of-ARM-Compiler-5-and-uARM-toolc/

closed 2020-04-14 by MarceloSalazar