ARMmbed/mbed-os #11313: Move RAM partition with ARM compiler

VVESTM, 2019-08-23:
### Description

**I would like to move the RAM partition on NUCLEO_H743ZI board. For now, I am able to do it with IAR and GCC_ARM toolchains but not with ARM one.**

After my developments for ethernet on STM32H7 (see here PR #11274), I faced some issue with those tests : 
> tests-mbed_platform-crash_reporting
> tests-mbedmicro-rtos-mbed-heap_and_stack

For those tests, it seems I need to keep the vector table and crash report ram to 0x20000000.
Regarding that, I decided to move only the RAM from 0x20000000 to 0x24000000.

I can do it for IAR and GCC_ARM, this is temporary available in my GitHub account : [branch](https://github.com/VVESTM/mbed-os/tree/vve_eth_h7_tmp) and [commit](https://github.com/ARMmbed/mbed-os/commit/f5256a8168560b7961d3f0609919382a9080625b).

For ARM, I cannot succeed to perform the same modification. I have tried this in stm32h743xI.sct :
>#define MBED_RAM_START              0x24000000
#define MBED_RAM_SIZE               0x80000
#define MBED_VECTTABLE_RAM_START    0x20000000
#define MBED_VECTTABLE_RAM_SIZE     0x298
#define MBED_CRASH_REPORT_RAM_START (MBED_VECTTABLE_RAM_START + MBED_VECTTABLE_RAM_SIZE)
#define MBED_CRASH_REPORT_RAM_SIZE  0x100
#define MBED_RAM0_START             (MBED_RAM_START)
#define MBED_RAM0_SIZE              (MBED_RAM_SIZE) 

But it seems there is no effect and the MSP stay in 0x20000000 area... (tests-mbedmicro-rtos-mbed-heap_and_stack test failed because msp is 0x2001ff70) :
    mbed test -t GCC_ARM -m NUCLEO_H743ZI -v -n tests-mbedmicro-rtos-mbed-heap_and_stack

**Do you know if there is an other place to modify in order to move RAM with ARM toolchain ? 
Is there some hardcoded values that forbid user to move RAM ?** 

For information, I also tried to exchange IRAM1 and IRAM2 sections in https://github.com/ARMmbed/mbed-os/blob/master/tools/arm_pack_manager/index.json file with no success.

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

jeromecoutant, 2019-08-23:
@kjbracey-arm

kjbracey-arm, 2019-08-23:
Initial stack pointer is specified here - doesn't come from the linker map:

https://github.com/ARMmbed/mbed-os/blob/a1540c5f77322b319c94fc70d00398f92a0d2c33/targets/TARGET_STM/TARGET_STM32H7/TARGET_STM32H743xI/device/TOOLCHAIN_ARM_STD/startup_stm32h743xx.S#L27

VVESTM, 2019-08-23:
Thanks @kjbracey-arm. Seems to be better if I set __initial_sp to 0x24080000.
No way to calculate this value from .sct file ?

I will perform a full test campaign to be sure nothing else is broken. I will then update this ticket.

kjbracey-arm, 2019-08-23:
Sure, I believe

                IMPORT ||Image$$ARM_LIB_STACK$$ZI$$Limit||

    _initial_sp EQU ||Image$$ARM_LIB_STACK$$ZI$$Limit||

should do it.

VVESTM, 2019-08-29:
Thanks @kjbracey-arm. As discussed with you, the exact solution to add in .S file is : 
            IMPORT |Image$$ARM_LIB_STACK$$ZI$$Limit|
__Vectors       DCD     |Image$$ARM_LIB_STACK$$ZI$$Limit|  ; Top of Stack

I have still some issue with IAR toolchain now. I will push a new PR as soon as I have the 3 toolchains working (ARM, GCC_ARM and IAR)

closed 2019-08-29 by VVESTM