ARMmbed/mbed-os #11329: How to create variable send/receive communication of I2CSlave

matsujirushi, 2019-08-25:
Hi,
I want to create variable send/receive communication of I2CSlave.

I tried, but it didn't work.
Could you tell me how to or sample code?

Thanks.

**I2C Slave (MbedOS5 + LPC1768)**
```cpp
#include <mbed.h>
 
#define DLM "\r\n"
 
#define I2C_SLAVE_ADDRESS   (0x40)
 
static I2CSlave _I2c(MBED_I2C0);
 
static void I2cSlaveDoWork()
{
    int addressed = _I2c.receive();
    if (addressed == I2CSlave::NoData) return;
 
    printf("Enter I2C Slave" DLM);
 
    switch (addressed)
    {
    case I2CSlave::WriteAddressed:
        printf("WriteAddressed" DLM);
        for (int i = 0; ; i++)
        {
            int data = _I2c.read();
            printf("%d:%#02x" DLM, i, data);
 
            if (data < 0) break;
        }
        break;
    case I2CSlave::ReadAddressed:
        printf("ReadAddressed" DLM);
        for (int i = 0; ; i++)
        {
            printf("%d:%#02x" DLM, i, i);
            if (_I2c.write(i) != 1) break;
        }
        break;
    }
 
    printf("Leave I2C Slave" DLM);
}
 
int main()
{
    _I2c.address(I2C_SLAVE_ADDRESS);
    while (true)
    {
        I2cSlaveDoWork();
    }
}
```

**I2C Master (Arduino Uno)**
```cpp
#include <Wire.h>
 
void setup()
{
  Wire.begin();

  delay(1000);

  Serial.println("1st write.");
  Wire.beginTransmission(0x20);
  Wire.write(0x11);
  Wire.endTransmission();
  
  delay(1000);
 
  Serial.println("2nd write.");
  Wire.beginTransmission(0x20);
  Wire.write(0x22);
  Wire.write(0x33);
  Wire.endTransmission();

  delay(1000);

  Serial.println("1st read.");
  Wire.requestFrom(0x20, 1);
  while (Wire.available())
  {
    Serial.print(Wire.read(), HEX);
  }

  delay(1000);

  Serial.println("2st read.");
  Wire.requestFrom(0x20, 2);
  while (Wire.available())
  {
    Serial.print(Wire.read(), HEX);
  }
}
 
void loop()
{
}
```

**I2C Slave log**
```
Enter I2C Slave
WriteAddressed
0:0x11
1:0x22
2:0x22
3:0x22
4:0x22
5:0x22
6:0x22
7:0x22
...
```

**Logic Analyzer**
[image]

desmond-blue, 2019-08-26:
Hi @matsujirushi, 

`_I2c.read()` returns the byte read so the condition `if (data < 0) break;` will always be true and it causes a infinite loop, please refer to our example and domentation of [SPISlave here](https://os.mbed.com/docs/mbed-os/v5.13/apis/i2cslave.html), and for this kind of usage question, we suggest you to ask on our [Questions page](https://os.mbed.com/questions/).

matsujirushi, 2019-08-27:
@desmond-blue Thanks for your response.

I cannot find how to get condition of stopped on I2C.
https://github.com/ARMmbed/mbed-os/blob/master/drivers/I2CSlave.h

Can we create variable send/receive communication?

desmond-blue, 2019-08-27:
@matsujirushi You can use `int read(char* data, int length)` to set max read length, it will read while reach max length or receive stop bit, please refer [here](https://os.mbed.com/docs/mbed-os/v5.13/mbed-os-api-doxy/classmbed_1_1_i2_c_slave.html#a161ea43dffd2c0f39a36a1fee6ba336d). Or you have to define the length of address/command in slave spec.

matsujirushi, 2019-08-27:
>  it will read while reach max length or receive stop bit,
How to get the received length?

desmond-blue, 2019-08-27:
This API is not designed for this kind of usage, you probably need to use lower level i2c driver to achieve what you need.

matsujirushi, 2019-08-27:
Okay. I got it.
Thanks for your support.

closed 2019-08-27 by matsujirushi