ARMmbed/mbed-os #11346: [DISCO_F746NG] Cannot Access SD Card

Biouche, 2019-08-27:
### Description
Hello everyone,
I am struggling accessing SD card on a DISCO_F746NG with a very simple code : 

```
/* mbed Microcontroller Library
 * Copyright (c) 2018 ARM Limited
 * SPDX-License-Identifier: Apache-2.0
 */

#include "mbed.h"
#include "stats_report.h"
#if COMPONENT_SD
#include "SDBlockDevice.h"
#endif 

SDBlockDevice bd(
        MBED_CONF_SD_SPI_MOSI,
        MBED_CONF_SD_SPI_MISO,
        MBED_CONF_SD_SPI_CLK,
        MBED_CONF_SD_SPI_CS);

// main() runs in its own thread in the OS
int main()
{
    #if COMPONENT_SD
    printf("Component SD is enabled\n");
    #endif
    // Initialize the block device
    printf("bd.init()\n");
    int err = bd.init();
    printf("bd.init -> %d\n", err);

    printf("OPENING FILE\r\n");
    FILE *fp = fopen("/SD/myfile.txt", "w+");
    if (fp!=NULL)
    {
        printf("WRITTING TO FILE\r\n");
        fputs("Hello World!\n", fp);
        printf("CLOSING FILE\r\n");
        fclose(fp);
    }
    return 0;
}
```

Here is the output I get : 

```
Component SD is enabled
bd.init()
bd.init -> -5005
OPENING FILE
```

Error -5005 means sd card is missing according to docs.

I am developing on Mbed Studio 0.5.3
Project configuration : 
- mbed-os version : 5.12.0
- Toolchain :  ARMC6
- SD Card :  Samsung MicroSD HC 8Go
- SD Card format :  MS-DOS (FAT32)

My mbed_app.json is the following : 

```
{
    "target_overrides": {
        "*": {
            "platform.stack-stats-enabled": true,
            "platform.heap-stats-enabled": true,
            "platform.cpu-stats-enabled": true,
            "platform.thread-stats-enabled": true,
            "platform.sys-stats-enabled": true,
            "platform.stdio-convert-newlines": true
        },
        "DISCO_F746NG": {
            "target.features_add": ["STORAGE"],
            "target.components_add": ["SD"]
        }
    }
}
```

Thank you for your help.

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

LMESTM, 2019-08-27:
@Biouche 
only a SD card over SPI is supported for now in mbed, not the native SDIO interface of the MCU. SDIO support in mbed is under discussion.

This means you have to use a SD card shield to be connected to the arduino connector for instance.
do you confirm that you are using such a shiled ?

Biouche, 2019-08-27:
Thank you for your reply @LMESTM 
Here is a picture of my board with sd inside.

[https://imgur.com/a/jlJrYO5](url)

Is this what you call an "sd card shield" ?

LMESTM, 2019-08-27:
hmm not sure - it looks like the SD card is inserted directly in the DISCO board SD slot. 
I'd think about something like this
https://www.seeedstudio.com/category/Arduino-c-1001/SD-Card-Shield-V4-p-1381.html
not having any preference on a specific shield - I just took the first picture I found ...

jeromecoutant, 2019-08-27:
@MarceloSalazar 
We need to support SDIO :-)
Thx

closed 2019-10-25 by ciarmcom