ARMmbed/mbed-os #11350: Make echo server configurable in cellular socket test

talorion, 2019-08-27:
### Description

is it possible to make the ECHO_SERVER_NAME and ECHO_SERVER_UDP_PORT configurable in the test 

features\cellular\TESTS\socket\udp\main.cpp

replace line 41, 42 
#define ECHO_SERVER_NAME "echo.mbedcloudtesting.com"
#define ECHO_SERVER_UDP_PORT  8877

with 
#ifndef MBED_CONF_APP_ECHO_SERVER_ADDR
#define ECHO_SERVER_ADDR "echo.mbedcloudtesting.com"
#else
#define ECHO_SERVER_NAME MBED_CONF_APP_ECHO_SERVER_ADDR
#endif

#ifndef MBED_CONF_APP_ECHO_SERVER_PORT
#define ECHO_SERVER_UDP_PORT  8877
#else
#define ECHO_SERVER_UDP_PORT  MBED_CONF_APP_ECHO_SERVER_PORT
#endif

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

talorion, 2019-08-28:
thank you for the hint, but i would like to run all tests as part of our CI toolchain. since .mbedignore does not work with tests i have to find a way to either delete this particular test or get it to run. this one is the only one i can not get to run since i can not configure the servers using the CellularInterface.json you mentioned.

AriParkkila, 2019-08-28:
@talorion Mbed OS socket tests are recommended nowadays. You can find them at `TESTS/netsocket/udp` and run with `tools/test_configs/CellularInterface.json`.

closed 2019-10-25 by ciarmcom