ARMmbed/mbed-os #11371: network_wifi test build break after mbed-os 5.12 release with GCC_ARM

aqib-ublox, 2019-08-29:
### Description
If you compile mbed-os-tests-network-wifi with any release prior to mbed-os-5.12.0, it compiles successfully with GCC_ARM tool-chain but from this release build breaks.
Possible cause is update in build script.

**Test build logs:**

[GCC_compile_error_logs.txt](https://github.com/ARMmbed/mbed-os/files/3554300/GCC_compile_error_logs.txt)

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-08-30:
ccc @ARMmbed/mbed-os-ipcore

michalpasztamobica, 2019-08-30:
Hi @aqib-ublox ,

I confirm the error, but it is not related to IPCore code, but rather to the compiler acting weird on Windows machines:
```
        Compile [  5.0%]: get_interface.cpp
        [removed lots and lots of compilation flags]
        Compile: arm-none-eabi-g++ 
        [DEBUG] Return: 1
        [DEBUG] Output: arm-none-eabi-g++: error: CreateProcess: No such file or directory
```

I checked that Linux machines compile the most recent code without issues.
On Windows I can see the same error no matter which test I compile (I tried `mbed test --compile -t GCC_ARM -m UBLOX_EVK_ODIN_W2 -n features-storage-tests-blockdevice-general_block_device -v` with the same result).
I am using the recommended `6 2017-q2-update` version of the GNU ARM toolchain.

@ARMmbed/mbed-os-tools , can you give us a hand here?

aqib-ublox, 2019-08-30:
@michalpasztamobica can u just checkout mbed-os-5.11_x any version you would be able to compile without any issue?
I guess script is modified to create folder in somewhat differently where this problem popped up of long paths?

michalpasztamobica, 2019-08-30:
I tried the very most recent  `8 2019-q3-update` GNU ARM toolchain and I got the exact same error.
I confirm, earlier versions compile fine. It might be some change in the build scripts.

0xc0170, 2019-09-04:
> I tried the very most recent 8 2019-q3-update GNU ARM toolchain and I got the exact same error.
I confirm, earlier versions compile fine. It might be some change in the build scripts.

@michalpasztamobica Will you look at this issue?

michalpasztamobica, 2019-09-04:
I am looking into this and I also asked @mark-edgeworth for help. I admit I couldn't get the GCC to work, even following official guidelines on a clean set up, even though this worked for Mark. I will keep trying and return here whenever I get something to work.

michalpasztamobica, 2019-09-16:
On my machine I managed to compile after reverting [this commit](https://github.com/ARMmbed/mbed-os/commit/841607d0290674bb3e2043789c380460bb95e56a). It was enough to remove the files added in it.
However on another machine I could not reproduce this success with the same steps. I also think the files themselves are innocent and simply trigger some glitch in the environment setup. 
@aqib-ublox , can you try if this would work in your setup?
@mark-edgeworth , could there be a limit of files that get compiles? I am trying to find an explanation of why a few files being removed made a difference for me...

mark-edgeworth, 2019-09-16:
Could this be another case of 'command line too long'? The compile commands that are generated behind the scenes are enormous, and can cause apparently random failures. If this sounds possible then you might try putting the build sources nearer to the root (perhaps initially directly under c:).

michalpasztamobica, 2019-09-16:
I have tried this already but it didn't work.
However, I assume that since May we might have added a lot of headers and removing just this particular one made a difference...
BTW. is there any deeper "behind the scene" than the one we see with `-v` option?
When I enabled it I saw quite some lenghty commands indeed, but they rarely exceeded 2000 characters, while Windows GCC would only start complaining at 32k characters and more, right?
So is there some more commands that we are not seeing with `-v` parameter ?

mark-edgeworth, 2019-09-16:
I don't know, this was only a theory....

mark-edgeworth, 2019-09-16:
If this header is a specific problem, does it compile in a minimal gcc example? Just a thought...

michalpasztamobica, 2019-09-17:
Indeed, I can compile a normal `main.cpp` program with `mbed compile`.
The problem only occurs for me when I run `mbed test --compile ...`.

aqib-ublox, 2019-09-17:
> On my machine I managed to compile after reverting [this commit](https://github.com/ARMmbed/mbed-os/commit/841607d0290674bb3e2043789c380460bb95e56a). It was enough to remove the files added in it. [...]

are you pointing t right commit?

michalpasztamobica, 2019-09-17:
Yes. I performed a git bisection from mbed-os-5.12 to HEAD and this commit made a difference on my setup - commits before it compiled and commits after it didn't. I totally cannot explain it.

MarceloSalazar, 2019-10-16:
@michalpasztamobica do you have an update on this?
is there anything we can do to help to find the root cause of this problem.

michalpasztamobica, 2019-11-18:
@mark-edgeworth , @MarceloSalazar , @aqib-ublox I finally got a commercial VMWare license and managed to reproduce the scenario on a clean setup of VM with Windows 10. Here are the exact steps I took:

1) Install VMWare Workstation 15.5 Pro
2) Get Windows 10 VMWare image from https://developer.microsoft.com/en-us/windows/downloads/virtual-machines (I used: `Windows 10 1903 and the Windows 10 SDK, version 1903 (10.0.18362.1`))
3) Unzip the image
4) Open virtual machine from the ovl file.
5) Power on the machine
6) Follow https://os.mbed.com/docs/mbed-os/v5.14/quick-start/offline-with-mbed-cli.html 
7) Download and install mbed-cli v4.10: https://github.com/ARMmbed/mbed-cli-windows-installer/releases/tag/v0.4.10
8) Open a the basic windows cmd prompt:
8) `mbed new test`
9) `cd test`
10) `mbed add mbed-os`
11) `mbed deploy` (only updated the library to commit `2e96145b7607de430235dd795ab5350c1d4d64d7`)
12) `mbed test --compile -t GCC_ARM -m UBLOX_EVK_ODIN_W2 -n mbed-os-tests-network-wifi -v`

The result for me is:
`[DEBUG] Output: arm-none-eabi-g++: error: CreateProcess: No such file or directory`

This only happens on Windows. This also happens with other tests, as mentioned above.

I find these to be valid reproduction steps and I think we can now consider the issue report valid?

mark-edgeworth, 2019-11-18:
This looks like a problem in the ```mbed test``` subcommand; @jamesbeyond can you take a look?
Does this build without the ```test``` command? ie ```mbed compile -t GCC_ARM -m UBLOX_EVK_ODIN_W2 -n mbed-os-tests-network-wifi -v?

michalpasztamobica, 2019-11-18:
> Indeed, I can compile a normal `main.cpp` program with `mbed compile`.
> The problem only occurs for me when I run `mbed test --compile ...`.

mark-edgeworth, 2019-11-18:
ok, thanks for trying. I'm pretty convinced then that the problem is in ```mbed test```

MarceloSalazar, 2019-11-18:
I don't think it's a problem with Mbed CLI, but rather a problem with the GCC toolchain in Windows - see https://github.com/ARMmbed/mbed-os/issues/7129

You should be able to get the test working well with Arm Compiler 6 or Arm Compiler 5. Can you try please?
If you really need to use GCC, you can use Linux/Mac or the WSL (https://docs.microsoft.com/en-us/windows/wsl/install-win10)

SeppoTakalo, 2019-11-18:
https://stackoverflow.com/questions/12598933/ndk-build-createprocess-make-e-87-the-parameter-is-incorrect

Windows problem.. Command cannot be longer that 32768 characters.

michalpasztamobica, 2019-11-19:
@SeppoTakalo , @MarceloSalazar  with the '-v' option I did not see any command longer than 2000 characters. See my [comment above](https://github.com/ARMmbed/mbed-os/issues/11371#issuecomment-531794660). The path to project location in my case is: "C:/Users/User". It can't get much shorter than this.

If it really is a Windows issue, can we get to the exact command which exceeds the length and document it so that we are able to indicate any further reports to an explicit official information that GCC on Windows is not supported?

Also - should the build system catch this error (preferably, by measuring the real commands length?) and give a reasonable message, rather than `CreateProcess was not found`?

michalpasztamobica, 2020-01-21:
@aqib-ublox ,
I got information that upgrading to GCC-9 resolved this issue (thanks for the hint, @maciejbocianski and @fkjagodzinski ).
Could you try it out and let us know if this solved your issue?

0xc0170, 2020-02-19:
@aqib-ublox Can you update this issue? Is it still relevant or shall we close?

aqib-ublox, 2020-02-19:
@0xc0170 Its working for us network test is building fine after GCC toolchain update.
@michalpasztamobica thanks

MarceloSalazar, 2020-03-25:
Disussed with @aqib-ublox and agreed that Ublox will discuss internally and confirm that's actually resolved.

MarceloSalazar, 2020-04-08:
Closing as we can't reproduce this with 5.15

closed 2020-04-08 by MarceloSalazar