ARMmbed/mbed-os #11386: Atomically read correct millisecond timestamp from RTC

maxgerhardt, 2019-08-31:
### Description

* mbed-os 5.12.2 
* target: NUCLEO_L476RG
* toolchain GCC, mbed-cli  1.10.1

I have a need for getting a timestamp from a very-low-drift and low-power source with still reasonable accuracy. That is why I cannot use the `Timer` (based on a microsecond resolution timer after whose activation the MCU jumps up to a constant high current draw) or the [`LowPowerTimer`](https://os.mbed.com/docs/mbed-os/v5.12/apis/lowpowertimer.html) which as the docs say is only good "for measuring small times (between microseconds and seconds).", but my range would be in 30 minutes to a few hours..

This is why I intended to use my STM32L476RG's RTC, which does have the register for subsecond register with a resolution determined by the dividers of the synchronous and asynchronous dividers. 

Now my problems lies within being able to atomically read a millisecond accurate and consistent time and date from the CPU. Since commit https://github.com/ARMmbed/mbed-os/pull/7365/commits/10529932361900c9472b262241f955f433c5278b you have changed the code to disable the shadow registers of the RTC "to avoid waiting for the synchronization duration". Doesn't this however destroy the capability of atomatically reading a consistent time and date? The shadow register's usage after all is, when the subsecond, time or date registers are read out, it will freeze all the other upper registers at the time of the read, so that a consistent snapshot can be read out. The RTC keeps ticking in the background and once the frozen registers are read out, the RTC fill the registers in with their actual current time value. Without the shadow registers, we're reading from a permanently updating register value. 

Now as [STM's document](https://www.st.com/content/ccc/resource/training/technical/product_training/group0/44/60/e0/b2/ee/f4/47/ff/STM32F7_WDG_TIMERS_RTC/files/STM32F7_WDG_TIMERS_RTC.pdf/jcr:content/translations/en.STM32F7_WDG_TIMERS_RTC.pdf) says on page 9 & 10

> When the Bypass Shadow Registers control bit is set, the actual calendar registers are read directly. The advantage of this mode is that there is no need to wait for the synchronization delay. The disadvantage is that the read values can be false or not consistent due to synchronization issues, so they must be read twice and compared with previous read values to ensure they are correct and coherent.

Which also what the code in the commit above is doing in the `while()` loop. I have imitated this behaviour in the following function below, which is based on [`rtc_read()`](https://github.com/ARMmbed/mbed-os/blob/bdd6cb8dee1e6beb8f67f3b474bdddfb824e641c/targets/TARGET_STM/rtc_api.c#L154-L203). It returns a 64-bit integer which is a unix timestamp in milliseconds 

```cpp
uint64_t RTC_GetTimestampMillis() {
#define RTC_SSR_RESERVED_MASK 0x0000ffffU
    struct tm timeinfo {};
    /* Since the shadow registers are bypassed we have to read the time twice and compare them until both times are the same */
    uint32_t Read_subsec = RTC->SSR & RTC_SSR_RESERVED_MASK;
    uint32_t Read_time = RTC->TR & RTC_TR_RESERVED_MASK;
    uint32_t Read_date = RTC->DR & RTC_DR_RESERVED_MASK;

    while ((Read_time != (RTC->TR & RTC_TR_RESERVED_MASK)) ||
           (Read_date != (RTC->DR & RTC_DR_RESERVED_MASK)) ||
           (Read_subsec != (RTC->SSR & RTC_SSR_RESERVED_MASK))) {
        Read_time = RTC->TR & RTC_TR_RESERVED_MASK;
        Read_date = RTC->DR & RTC_DR_RESERVED_MASK;
        Read_subsec = RTC->SSR & RTC_SSR_RESERVED_MASK;
    }

    timeinfo.tm_mday = RTC_Bcd2ToByte((uint8_t)(Read_date & (RTC_DR_DT | RTC_DR_DU)));
    timeinfo.tm_mon  = RTC_Bcd2ToByte((uint8_t)((Read_date & (RTC_DR_MT | RTC_DR_MU))  >> 8)) - 1;
    timeinfo.tm_year = RTC_Bcd2ToByte((uint8_t)((Read_date & (RTC_DR_YT | RTC_DR_YU))  >> 16)) + 68;
    timeinfo.tm_hour = RTC_Bcd2ToByte((uint8_t)((Read_time & (RTC_TR_HT  | RTC_TR_HU))  >> 16));
    timeinfo.tm_min  = RTC_Bcd2ToByte((uint8_t)((Read_time & (RTC_TR_MNT | RTC_TR_MNU)) >> 8));
    timeinfo.tm_sec  = RTC_Bcd2ToByte((uint8_t)((Read_time & (RTC_TR_ST  | RTC_TR_SU))  >> 0));

    // Convert to timestamp
    time_t t;
    if (_rtc_maketime(&timeinfo, &t, RTC_4_YEAR_LEAP_YEAR_SUPPORT) == false) {
        //error
        //wtf? attempt to read normal time (lesser resolution)
        time(&t);
        return t * 1000ULL;
    }
    //see STM32L4x6 reference manual chapter 38.5.10 RTC_SSR
    uint32_t subsecs = (PREDIV_S_VALUE - Read_subsec) * 1000UL / (PREDIV_S_VALUE + 1);
    if(Read_subsec > PREDIV_S_VALUE) {
        //subtract one second. we are already in a linear time space form the unix timestamp conversion above, so we can easily subtract one second now.
        t--;
    }
    uint64_t time_milliseconds = t * 1000ULL + subsecs;
    return time_milliseconds;
}
```

For register definitions see [STM32L4x6 reference manual](https://www.st.com/content/ccc/resource/technical/document/reference_manual/02/35/09/0c/4f/f7/40/03/DM00083560.pdf/files/DM00083560.pdf/jcr:content/translations/en.DM00083560.pdf) page 1250.

The expected resolution of this is 1000 / (PREDIV_S_VALUE + 1) = 1 / 256 = ~3.9 milliseconds. Printing the time in a loop gives the expected results (starting timestamp is arbitrary)

```
2020-07-08 02:28:27.992
2020-07-08 02:28:27.996
2020-07-08 02:28:27.996
2020-07-08 02:28:28.000
2020-07-08 02:28:28.000
2020-07-08 02:28:28.003
2020-07-08 02:28:28.007
2020-07-08 02:28:28.007
2020-07-08 02:28:28.011
```

My concrete questions is: Is this code reasonable, i.e., can it reliably provide a consistent RTC time with the shadow registers being disabled but the register being read-out twice? My main worry is an edge case in which e.g. after the subsecond register is read, some time (microseconds..) passes, in which the RTC of course advances, and then e.g. the seconds in the time register increment to a new value; E.g., a timestamp like `00:00:01.999` (`hh:mm:ss.millis`) although it's actually `00:00:00.999` but the seconds are `1` because it was read out slightly later.

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

40Grit, 2019-09-01:
@kjbracey-arm

0xc0170, 2019-09-02:
cc @ARMmbed/team-st-mcd

kjbracey-arm, 2019-09-02:
> the LowPowerTimer which as the docs say is only good "for measuring small times (between microseconds and seconds).", 

Sounds like a documentation error. That would conceivably apply to `Timer`, but even then is a bit out-of-date, because the timer is now 64-bit (extended in software).

When it was 32-bit, wrap range would have been an issue. I would say `LowPowerTimer` should be fine, and on this hardware, maybe it maps to what you're reading anyway?

Another way to get a monotonic millisecond time is to use the `Kernel::get_ms_count` call, but that won't be RTC-referenced. You could combine the results of an initial or periodic RTC read to give yourself something with millisecond precision, and one-second real-time accuracy.

(That count read is "free" in an RTOS build, as it's just reading a variable that is incremented anyway. In a non-RTOS build, it just maps to a `LowPowerTimer`).

One important point is that RTC reads might be slow, and on some hardware it only has second resolution.

Really, the `gettimeofday` implementation in `mbed_rtc_time.cpp` should be combining HAL RTC and the millisecond count to produce something nicer. I'd say you could file that as an issue - the 1-second granularity of that implementation.

> Is this code reasonable, i.e., can it reliably provide a consistent RTC time with the shadow registers being disabled but the register being read-out twice?

I'm not familiar with the hardware, but in the simplest case of 2 registers that need to be read, this code is solid:

     uint32_t old_hi = TIME_HI;
     uint32_t lo = TIME_LO;
     uint32_t new_hi = TIME_HI;
     uint64_t t;
     if (old_hi == new_hi) {
          // Didn't wrap while reading; t = "(hi,lo)".
          t = ((uint64_t) new_hi << 32) | lo;
     } else {
          // Did wrap while reading; t = "(hi,0)".
          t = (uint64_t) new_hi << 32;
     }

Same method would extend to multiple registers - in the above "TIME_LO" could have been concocted the same way via 2 subregisters.

maxgerhardt, 2019-09-03:
> Sounds like a documentation error. That would conceivably apply to Timer, but even then is a bit out-of-date, because the timer is now 64-bit (extended in software).
> When it was 32-bit, wrap range [...]

Indeed `LowPowerTimer` is just a `Timer` object with the callbacks from `get_lp_ticker_data()`, which in my configuration with `MBED_CONF_TARGET_LPTICKER_LPTIM` enabled will use `LPTIM1`, the low power timer of the STM32L476RG. 

As https://os.mbed.com/docs/mbed-os/v5.12/apis/timer.html#warnings-and-notes says

> Timers are based on 64-bit unsigned microsecond counters, but for backward compatibility, the read_ms() and read_us() methods only return 32-bit signed integers. This limits their range before wrapping to 49 days and 35 minutes respectively. Use read_high_resolution_us() to access the full range of over 500,000 years.

So to not get an overflow `read_high_resolution_us()` must be used. I will still confirm this independently, because the function will invoke `ticker_read_us()` which will get `LPTIM1->CNT`; not sure yet where the overflow handling logic is.

Now I'm however not entirely sold on the presented piece of code. Given that `TIME_HI` resepresents the high (most significant, slow changing) bits and `TIME_LO` represents the lower (least significant, fast changing) part, when `old_hi != new_hi`, why is it assumed `lo = 0` now?

kjbracey-arm, 2019-09-03:
> So to not get an overflow `read_high_resolution_us()` must be used. 

Yes. 

>I will still confirm this independently, because the function will invoke `ticker_read_us()` which will get `LPTIM1->CNT`; not sure yet where the overflow handling logic is.

An interrupt handler is set up to trigger before the hardware timer wraps, and that increments a high order counter in the common timer code.

> Now I'm however not entirely sold on the presented piece of code. Given that `TIME_HI` resepresents the high (most significant, slow changing) bits and `TIME_LO` represents the lower (least significant, fast changing) part, when `old_hi != new_hi`, why is it assumed `lo = 0` now?

By the time you reach the `if`, the time in the past - the time sampling happened over the period of the 3 prior hardware reads. That read wasn't instantaneous, but when the hi word doesn't change, what you sampled was the precise instant of the middle low read.

When high does change, you know that low wrapped, so the time `new_hi:0000` definitely happened some time during the interval. So it is a valid time corresponding to the read interval, even though it may not line up precisely with the middle read like the non-wrapping reads.

Now, if that lack of alignment with middle read matters to you (you'd have to be pretty precise!), you could heuristically look at the low number, and, assuming `hi_new == hi_old + 1`, deduce whether what you read was `hi_old::low` or `hi_new:low`, depending on whether `low` looks "close to wrap" or "just wrapped". But there's a danger that heuristic goes wrong, if somehow a lot of time passed during the read interval. You could return `hi_new:2000` when the time is actually `hi_new:1000`, and the next read could return a lower number.

By choosing `hi_new:0000`, you guarantee monoticity - you're returning a time that definitely falls within that window and has now passed.

The monoticity concern is small if the timers are wide enough, and you're doing something like disabling interrupts during the read (so avoiding thread switching or suspending), but in the general case I would tend to go for the simple, monotonic `new_hi:0`.

If you don't like a tendency to return `0` in the low word (eg random number generator?), another approach is to just try again:

     old_hi = TIME_HI;
     do {
          old_hi = new_hi;
          lo = TIME_LO;
          new_hi = TIME_HI;
    } while (new_hi != old_hi);

That's also reasonable, if it doesn't bother you to wait for a "better" time near wrap, but relies on you being faster than `TIME_HI` increments to make progress. Will still mess up the even distribution of the low word a bit though - it will tend to have a clump just above 0.

maxgerhardt, 2019-09-05:
Thanks for the explanations, I think this is satisfactory.

closed 2019-09-05 by maxgerhardt

AndreiCherniaev, 2021-11-02:
see also https://community.st.com/s/question/0D53W000013JV5nSAG/32f417-rtc-errata-sheet-if-reading-subseconds-is-this-code-ok