ARMmbed/mbed-os #11387: Atmel ATSAMD51J19A Target Support Possibility

rsispal, 2019-09-01:
### Description

Hi there,

Hope it's okay to ask this here. I see that there are ongoing efforts to add support to the [SAMD21 range](https://github.com/ARMmbed/mbed-os/issues/9016), and was curious to know if anything is happening with the Atmel SAMD51 Cortex-M4 range.

I use the Adafruit ItsyBitsy M4 Express for prototyping which has the [Atmel ATSAMD51J19](https://www.microchip.com/wwwproducts/en/ATSAMD51J19A) which has 512KB flash and 192KB RAM. Comparing that to some currently supported targets such as the EFM32 Pearl Gecko - if I looked at the specs correctly the ATSAMD51J19 part is equally (if not more) capable?

Hoping there might at least be a starting point with an existing supported board with comparable specification that I can take a look at and assess how complex it might be to port.

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

40Grit, 2019-09-01:
Microchip has shown zero interest in Mbed. They've even gone as far as telling me they see no demand for Mbed. Mchip has always been out of touch though...

Therefore the only SAM parts with mbed support are those supported by companies using them in existing designs. 

Personally, I have completely given up on the SAM line.

Some nice features...
but too many ways to interpret their reference materials and register descriptions.

Weeks, dare I say months, lost fixing bugs I could never quite determine a root cause for. And too many drivers I have needed to fix personally.

40Grit, 2019-09-01:
Sorry, the above was no help.

0xc0170, 2019-09-02:
I am not aware of any other activity around SAM family besides the one referenced above.

closed 2019-10-25 by ciarmcom

0xc0170, 2019-10-28:
Community contributions are welcome. Note, the current Atmel targets still need some update to be compatible with Mbed OS 5 (they are enabled only for mbed 2).