ARMmbed/mbed-os #11389: Unable to build attestation test with ARMC5

dgreen-arm, 2019-09-02:
### Description

On master [(940d3fd)](https://github.com/ARMmbed/mbed-os/commit/940d3fdf603bf7590468055ab62df25bca23de03), I'm unable to build the compliance attestation test with ARMC5.

I'm running `python -m mbed test -m K64F -t ARMC5 --build-data BUILD/tests/build_data.json -n components-target_psa-tests-compliance_attestation-test_a001 -v -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 -DMBED_ALL_STATS_ENABLED -DSKIP_TIME_DRIFT_TESTS=1`

and getting the following error:
`[DEBUG] Return: 1
[DEBUG] Output: "./events/EventQueue.h", line 171: Error:  #442: too few arguments for class template "events::UserAllocatedEvent" 
[DEBUG] Output: "./events/EventQueue.h", line 217: Error:  #442: too few arguments for class template "events::UserAllocatedEvent" 
[DEBUG] Output: ./components/wifi/esp8266-driver/ESP8266Interface.cpp: 0 warnings, 2 errors`

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-09-02:
https://github.com/ARMmbed/mbed-os/pull/11342 - this one brought in UserAllocatedEven

cc @maciejbocianski @kjbracey-arm

kjbracey-arm, 2019-09-02:
https://github.com/ARMmbed/mbed-os/blob/940d3fdf603bf7590468055ab62df25bca23de03/events/EventQueue.h#L170-L171

Hmm, that should be

    template<typename F, typename A> 
    bool cancel(UserAllocatedEvent<F, A> *event)

maciejbocianski, 2019-09-02:
fix https://github.com/ARMmbed/mbed-os/pull/11391

0xc0170, 2019-09-03:
cc @ARMmbed/mbed-os-test ARMC5 job in nightly 💯

Patater, 2019-09-03:
#11391 has been merged. This issue should be closed as fixed.

closed 2019-09-04 by 0xc0170