ARMmbed/mbed-os #11397: Fail to scan on 1M and coded PHY simultaneously - only the first scan takes effect

lichangche, 2019-09-03:
### Description

Master branch as well 5.13.x cannot start scan on 1M PHY and coded PHY simultaneously - only the first scan takes effect.

Reproduced using the following example:
https://github.com/lichangche/mbed-os-example-ble/tree/master/dual_phy_scan

On NRF52840_DK
compiler - updated MBED_CLI 
SHA of Mbed - 940d3fdf603bf7590468055ab62df25bca23de03 
(verified and it happens on latest master branch and  5.13.4 as well)

Run the above example on 2 nrf52840_DKs, both will advertise coded PHY and try to scan on coded as well as 1M PHY.  Expect to see from both consoles:

found peer device on LE coded:LE coded 
as well as
found device with manuf data on LE 1M

Real output:

Depending on the order scan - 
if start 1M phy scan first, only see reports on the 1M PHY;
if starts coded phy scan first, can only see reports on coded PHY.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ X] Bug

pan-, 2020-10-05:
@lichangche You should not start scanning twice. You should put in your scan parameters that you want to scan on the 1M and Coded PHY. You can do that by using the function [`ScanParameter::setPhys`](https://github.com/ARMmbed/mbed-os/blob/ad40b1b2672d59b058f7c104fdc612d0dddf3ac3/connectivity/FEATURE_BLE/include/ble/gap/ScanParameters.h#L195)

paul-szczepanek-arm, 2021-03-18:
Scanning on multiple phys works fine.

Two things, as Vincent already pointed out you must use `setCodedPhyConfiguration` to enable the coded scanning together with 1m. You can only start scanning once.
```
scan_params.set1mPhyConfiguration(ble::scan_interval_t(200), scan_window_t(200), false);
scan_params.setCodedPhyConfiguration(ble::scan_interval_t(200), scan_window_t(200), false);
```

Second, to advertise on coded phy you cannot use legacy PDU:
```
advertising_params.setUseLegacyPDU(false);
```

closed 2021-03-18 by paul-szczepanek-arm