ARMmbed/mbed-os #11405: Empty program still links a lot of unused code making a rather heavy .hex

ladislas, 2019-09-03:
### Description

*This is a duplicated of https://github.com/ARMmbed/mbed-cli/issues/915*

Hi there!

Coming from the AVR world, I've recently started playing with ARM and mbed to my great pleasure. My way of exploring this new paradigm is to start from the examples, run them, tear them down until it stops working and build again one brick at a time to understand how things work together.

To start with, I've tried compiling an empty program `int main(void) { return 0; }`, expecting the resulting .hex to be close to empty.

But this is what I got:

```console
$ mbed compile --stats-depth=10 --color -j8 --profile release

[mbed] Working path "/Users/ladislas/dev/leka/LKAlphaOS-mbed" (program)
[Warning] @,: Compiler version mismatch: Have 8.2.1; expected version >= 6.0.0 and < 7.0.0
Building project LKAlphaOS-mbed (NRF52_DK, GCC_ARM)
Scan: LKAlphaOS-mbed
Link: LKAlphaOS-mbed
Elf2Bin: LKAlphaOS-mbed
| Module                                                                                                                  |     .text |    .data |    .bss |
|-------------------------------------------------------------------------------------------------------------------------|-----------|----------|---------|
| [fill]                                                                                                                  |    30(+0) |    4(+0) |  26(+0) |
| [lib]/c.a/lib_a-callocr.o                                                                                               |    96(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-ctype_.o                                                                                                |     0(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-dtoa.o                                                                                                  |  3952(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-errno.o                                                                                                 |    12(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-freer.o                                                                                                 |   652(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-impure.o                                                                                                |     0(+0) | 1068(+0) |   0(+0) |
| [lib]/c.a/lib_a-init.o                                                                                                  |    72(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-locale.o                                                                                                |     0(+0) |  364(+0) |   0(+0) |
| [lib]/c.a/lib_a-localeconv.o                                                                                            |     8(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-malloc.o                                                                                                |    16(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-mallocr.o                                                                                               |  1392(+0) | 1040(+0) |  52(+0) |
| [lib]/c.a/lib_a-mbtowc_r.o                                                                                              |    44(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-memchr.o                                                                                                |   160(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-memcpy.o                                                                                                |   308(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-memmove.o                                                                                               |   208(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-memset.o                                                                                                |   160(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-mprec.o                                                                                                 |  1548(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-reallocr.o                                                                                              |   876(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-reent.o                                                                                                 |     0(+0) |    0(+0) |   4(+0) |
| [lib]/c.a/lib_a-s_frexp.o                                                                                               |   112(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-sbrkr.o                                                                                                 |    36(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-strlen.o                                                                                                |   220(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-strncpy.o                                                                                               |   116(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-svfiprintf.o                                                                                            |   248(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-svfprintf.o                                                                                             |  6724(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-vsnprintf.o                                                                                             |   136(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-wctomb_r.o                                                                                              |    28(+0) |    0(+0) |   0(+0) |
| [lib]/gcc.a/_aeabi_uldivmod.o                                                                                           |    48(+0) |    0(+0) |   0(+0) |
| [lib]/gcc.a/_arm_addsubdf3.o                                                                                            |   888(+0) |    0(+0) |   0(+0) |
| [lib]/gcc.a/_arm_cmpdf2.o                                                                                               |   272(+0) |    0(+0) |   0(+0) |
| [lib]/gcc.a/_arm_fixdfsi.o                                                                                              |    80(+0) |    0(+0) |   0(+0) |
| [lib]/gcc.a/_arm_muldivdf3.o                                                                                            |  1060(+0) |    0(+0) |   0(+0) |
| [lib]/gcc.a/_arm_unorddf2.o                                                                                             |    44(+0) |    0(+0) |   0(+0) |
| [lib]/gcc.a/_dvmd_tls.o                                                                                                 |     4(+0) |    0(+0) |   0(+0) |
| [lib]/gcc.a/_udivmoddi4.o                                                                                               |   720(+0) |    0(+0) |   0(+0) |
| [lib]/misc/crt0.o                                                                                                       |   116(+0) |    0(+0) |   0(+0) |
| [lib]/misc/crtbegin.o                                                                                                   |    64(+0) |    4(+0) |  28(+0) |
| [lib]/misc/crtend.o                                                                                                     |     0(+0) |    0(+0) |   0(+0) |
| [lib]/misc/crti.o                                                                                                       |     0(+0) |    0(+0) |   0(+0) |
| [lib]/misc/crtn.o                                                                                                       |     0(+0) |    0(+0) |   0(+0) |
| mbed-os/drivers/Serial.o                                                                                                |    54(+0) |    0(+0) |   0(+0) |
| mbed-os/drivers/UARTSerial.o                                                                                            |    20(+0) |    0(+0) |   0(+0) |
| mbed-os/hal/mbed_gpio.o                                                                                                 |    88(+0) |    0(+0) |   0(+0) |
| mbed-os/hal/mbed_ticker_api.o                                                                                           |   950(+0) |    0(+0) |   0(+0) |
| mbed-os/hal/mbed_us_ticker_api.o                                                                                        |    64(+0) |    4(+0) |  65(+0) |
[... cut ...]
This seems like a lot for an empty program... And I'm also surprised things like `lib_a-dtoa.o` are actually linked even if I'm only blinking a led or doing nothing at all.

As I said, I'm new to this, so maybe I'm doing something wrong, or maybe I missed the part in the documentation explaining why it's this way.

Is this the expected behavior?

Thanks a lot for your time! :)
Ladislas

- target: NRF52_DK with and without bare-metal
- toolchain: GCC_ARM
- mbed-cli: 1.10.0
- mbed-os: 808c45062f

### Issue request type

    [x] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-10-24:
There has been effort to remove floating point, you can use minimal printf (see documentation) - it should help.

What are the other parts that should not be there? What optimizations have you used?

lto is another one that could also help here, will happen once we update to newer GCC version

ladislas, 2019-10-24:
I haven't had time to try with minimal printf, but as soon as I do I'll let you know.

That being said, I'm surprised that when the program is empty, a lot is still linked to the final .hex. But I might be missing something.

ladislas, 2019-10-27:
@0xc0170 minimal printf does indeed make the final .hex lighter: 47572 bytes vs.  33644(-13928) bytes.

I've tried adding the `-flto` flag to the release profile using arm gcc 8.2.1 but I have a linker error:

```
Link: mbed-os-example-blinky
/usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: /var/folders/5c/gvcsvggs555__n19mcc8c0780000gn/T//ccPNhOD7.ltrans0.ltrans.o: in function `mbed_error_printf':
<artificial>:(.text+0x410c): undefined reference to `__wrap_vsnprintf'
/usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: /var/folders/5c/gvcsvggs555__n19mcc8c0780000gn/T//ccPNhOD7.ltrans0.ltrans.o: in function `__wrap_main':
<artificial>:(.text+0x43e4): undefined reference to `main'
```

I'll wait till it's officially supported.

0xc0170, 2019-10-28:
Is there anything else for this ticket? Shall we close?

We will do updates also for baremetal . `lto` will come one day hopefully as well!

ladislas, 2019-10-28:
We can close for now and wait for updates.

closed 2019-10-28 by ladislas