ARMmbed/mbed-os #11406: NUCLEO-H743ZI2 TIM17 problem

vhsabino, 2019-09-03:
### Description
Hello, we are trying to instance TIM17 to use it as PWMout, however it crashes the code. We removed everything from the code, except this pin (PB_9 or PF_7) and the code crashes instantly. 
Thanks in advance.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

LMESTM, 2019-09-04:
@vhsabino  - would be great to share the sample code and the console logs when reporting such a problem.  

You're not supposed to explicitly select a given Timer instance in MBED. If using PB_9, the default Timer instance that will be selected is TIM4, See pin map table here:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32H7/TARGET_STM32H743xI/TARGET_NUCLEO_H743ZI2/PeripheralPins.c#L206

In case you want to use TIM17, you have the possibility to use PB9_ALT0 pin. Which is the same pin with an alternative ALT configuration than the default one. See here:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32H7/TARGET_STM32H743xI/TARGET_NUCLEO_H743ZI2/PeripheralPins.c#L207

Hope this helps

0xc0170, 2019-09-04:
@vhsabino As noted above, without description how to reproduce, we can't much help.

vhsabino, 2019-09-04:
Thank you for your fast answer!

We are aware about the pin map table and we tried PB9_ALT0 as well before come here.

Here is the simplest code I can provide you.

`#include <mbed.h>
Serial pc(USBTX, USBRX, 115200);
PwmOut LED(PF_7);
int main() {
  // It does not print anything with the PWM at the PF_7, if I change the interface to DigitalOut it prints.
  pc.printf("Hello World!\n"); 
  LED.write(0.5);
}
`

I appreciate your help!

LMESTM, 2019-09-04:
Ok thanks - that is clearer now. 
We may be missing a few lines with TIM17 indeed then, like here:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32H7/pwmout_device.c#L73

@jeromecoutant

jeromecoutant, 2020-01-21:
Please review #12295

PR #12295 (2020-01-23): STM32H7 correct PWMOUT instances

### Summary of changes 

This fixes #11406

PWM with TIM17 was not possible

#### Impact of changes 

#### Migration actions required 

### Documentation 

------------------------------------------------------------------------------------------
closed 2020-01-23 by adbridge