ARMmbed/mbed-os #11411: QSPIFBlockDevice::read dont work in QSPI mode with W25Q32

squli, 2019-09-04:
### Description
QSPIFBlockDevice doesn't work correctly with W25Q32 flash memory. I try to read data after erase and read only 0x88(0b1000). I think it means, that MCU waits data on 4 lines, but memory send it only by one line.

[INFO][QSPIF]: _reset_flash_mem:
[INFO][QSPIF]: Initialize flash memory OK
[INFO][QSPIF]: Init - verified SFDP Signature and version Successfully
[INFO][QSPIF]: Erase Type(A) 1 - Inst: 0xffh, Size: 4096
[INFO][QSPIF]: Erase Type 1 - Inst: 0x20h, Size: 4096
[INFO][QSPIF]: Erase Type(A) 2 - Inst: 0xffh, Size: 32768
[INFO][QSPIF]: Erase Type 2 - Inst: 0x52h, Size: 32768
[INFO][QSPIF]: Erase Type(A) 3 - Inst: 0xffh, Size: 65536
[INFO][QSPIF]: Erase Type 3 - Inst: 0xd8h, Size: 65536
[INFO][QSPIF]: Erase Type(A) 4 - Inst: 0xffh, Size: 1
[INFO][QSPIF]: Erase Type 4 - Inst: 0xffh, Size: 1
[INFO][QSPIF]: Init - Setting Quad Enable
 

To reproduce 
    qspi->init();
    qspi->erase(0x00, 0x2000);    
    qspi->read(rbuf, 0x00, 128);
    for (int i = 0; i < 128; i++)
      printf("%x ", rbuf[i]);    

[INFO][QSPIF]: Read Inst: 0xebh
 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 
88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88
 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 8
8 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 
88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88 88
 88 88 88 88 88 ...

To use memory with mbed i commented line 701 of QSPIFBlockDevice.
    //_sfdp_detect_best_bus_read_mode(param_table, basic_table_size, shouldSetQuadEnable, is_qpi_mode, _read_instruction);

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-09-05:
> terget stm32F446ze on nucleo board

@ARMmbed/team-st-mcd Have you seen similar issue or it's new?

jeromecoutant, 2019-09-05:
Issue is not the target.
Issue is the QSPI memory, and yes, we have already seen some issue: #11287

LMESTM, 2019-09-05:
Has W25Q32 flash memory been tested with any other target ? 
What are the results of QSPI (not QSPIF) test cases ?

0xc0170, 2019-09-05:
> What are the results of QSPI (not QSPIF) test cases ?

@squli Can you run qspi driver tests? using `mbed test` command and share the results here.

We don't have W25Q32FVSSIG  in our tree, do we?

squli, 2019-09-06:
> @squli Can you run qspi driver tests? using mbed test command and share the results here.
How can i do it? 
I try to use mbed test -n mbed-os-tests-mbed_hal-qspi - mbedgt: test name 'mbed-os-tests-mbed_hal-qspi' not found in '.\BUILD\tests\NUCLEO_F446ZE\GCC_ARM\test_spec.json' (specified with --test-spec option)

LMESTM, 2019-09-09:
You would have to write a new a flash configuration "W25Q32FVSSIG_config.h" in: 
mbed/TESTS/mbed_hal/qspi/flash_configs
You should derive this config from another existing config and use the memory datasheet for getting the memory details. 

Then you need to add a line in mbed/TESTS/mbed_hal/qspi/flash_configs/flash_configs.h like

#elif defined(TARGET_NUCLEO_F446ZE)
#include "W25Q32FVSSIG_config.h"

LMESTM, 2019-09-09:
and another question, why have you done the below change ?

> To use memory with mbed i commented line 701 of QSPIFBlockDevice.
//_sfdp_detect_best_bus_read_mode(param_table, basic_table_size, shouldSetQuadEnable, is_qpi_mode, _read_instruction);

@0xc0170 shouldn't this been discussed by storage team ?

0xc0170, 2019-09-09:
cc @ARMmbed/mbed-os-storage

squli, 2019-09-09:
i made W25Q32FVSSIG_config.h and modify flash_configs.h. 

Build failures:
        BUILD/tests/NUCLEO_F446ZE/GCC_ARM/mbed-os/TESTS/mbed_hal/qspi/mbed-os/TESTS/mbed_hal/qspi/main.o: In function `main': .\mbed-os\TESTS\mbed_hal\qspi/main.cpp:601: multiple definition of `main'   BUILD/tests/NUCLEO_F446ZE/GCC_ARM/main.o:c:\work\test_consumption\nucleo_consuption/.\main.cpp:9: first defined here         collect2.exe: error: ld returned 1 exit status

> and another question, why have you done the below change ?
I want to use single wire-SPI: 1-1-1 and want to disable another read commands in driver

LMESTM, 2019-09-10:
> c:\work\test_consumption\nucleo_consuption/.\main.cpp

It seems you have a local main.cpp for your own development in your tree:

> I want to use single wire-SPI: 1-1-1 and want to disable another read commands in driver

There may be a better way for doing so - need guidance from @ARMmbed/mbed-os-storage

squli, 2019-09-10:
> It seems you have a local main.cpp for your own development in your tree:

Thank you, i didn't remove my main() function.

I checked my hardware and reproduce problem from first post. After i made command mbed test -n mbed-os-tests-mbed_hal-qspi. Here is output:

> mbedgt: test case summary event not found
>         no test case report present, assuming test suite to be a single test case!
>         test suite: mbed-os-tests-mbed_hal-qspi
>         test case: mbed-os-tests-mbed_hal-qspi
> mbedgt: test on hardware with target id: 07780221033361293C77F76C
> mbedgt: test suite 'mbed-os-tests-mbed_hal-qspi' ..................................................... SYNC_FAILED in 37.62 sec
>         test case: 'mbed-os-tests-mbed_hal-qspi' ..................................................... ERROR in 37.62 sec
> mbedgt: test case summary: 0 passes, 1 failure
> mbedgt: all tests finished!
> mbedgt: shuffle seed: 0.9379687156
> mbedgt: test suite report:
> | target                | platform_name | test suite                  | result      | elapsed_time (sec) | copy_method |
> |-----------------------|---------------|-----------------------------|-------------|--------------------|-------------|
> | NUCLEO_F446ZE-GCC_ARM | NUCLEO_F446ZE | mbed-os-tests-mbed_hal-qspi | SYNC_FAILED | 37.62              | default     |
> mbedgt: test suite results: 1 SYNC_FAILED
> mbedgt: test case report:
> | target                | platform_name | test suite                  | test case                   | passed | failed | result | elapsed_time (sec) |
> |-----------------------|---------------|-----------------------------|-----------------------------|--------|--------|--------|--------------------|
> | NUCLEO_F446ZE-GCC_ARM | NUCLEO_F446ZE | mbed-os-tests-mbed_hal-qspi | mbed-os-tests-
> 
> mbed_hal-qspi | 0      | 1      | ERROR  | 37.62              |
> mbedgt: test case results: 1 ERROR
> mbedgt: completed in 40.02 sec
> mbedgt: exited with code 1

LMESTM, 2019-09-10:
Apparently the target does not start at all, no test is launched because your target and PC can't sync at all 
> mbedgt: test suite 'mbed-os-tests-mbed_hal-qspi' ..................................................... SYNC_FAILED in 37.62 sec

Please make sure that at least one other basic test is running fine

squli, 2019-09-16:
```
Build successes:
  * NUCLEO_F446ZE::GCC_ARM::MBED-BUILD
  * NUCLEO_F446ZE::GCC_ARM::MBED-OS-TESTS-MBED_PLATFORM-CIRCULARBUFFER
mbedgt: greentea test automation tool ver. 1.5.0
mbedgt: test specification file '.\BUILD\tests\NUCLEO_F446ZE\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using '.\BUILD\tests\NUCLEO_F446ZE\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
mbedgt: processing target 'NUCLEO_F446ZE' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: test case filter (specified with -n option)
        mbed-os-tests-mbed_platform-circularbuffer
        test filtered in 'mbed-os-tests-mbed_platform-circularbuffer'
mbedgt: running 1 test for platform 'NUCLEO_F446ZE' and toolchain 'GCC_ARM'
mbedgt: mbed-host-test-runner: started
mbedgt: retry mbedhtrun 1/1
mbedgt: ['mbedhtrun', '-m', u'NUCLEO_F446ZE', '-p', u'COM9:115200', '-f', u'"BUILD/tests/NUCLEO_F446ZE/GCC_ARM/mbed-os/TESTS/mbed_platform/CircularBuffer/CircularBuffer.bin"', '-e', u'"mbed-os\\TESTS\\host_tests"', '-d', 'D:', '-c', 'default', '-t', u'07780221033361293C77F76C', '-r', 'default', '-C', '4', '--sync', '5', '-P', '60'] failed after 1 count
mbedgt: checking for GCOV data...
mbedgt: test case summary event not found
        no test case report present, assuming test suite to be a single test case!
        test suite: mbed-os-tests-mbed_platform-circularbuffer
        test case: mbed-os-tests-mbed_platform-circularbuffer
mbedgt: test on hardware with target id: 07780221033361293C77F76C
mbedgt: test suite 'mbed-os-tests-mbed_platform-circularbuffer' ...................................... SYNC_FAILED in 37.44 sec
        test case: 'mbed-os-tests-mbed_platform-circularbuffer' ...................................... ERROR in 37.44 sec
mbedgt: test case summary: 0 passes, 1 failure
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.0767498477
mbedgt: test suite report:
| target                | platform_name | test suite                                 | result      | elapsed_time (sec) | copy_method |
|-----------------------|---------------|--------------------------------------------|-------------|--------------------|-------------|
| NUCLEO_F446ZE-GCC_ARM | NUCLEO_F446ZE | mbed-os-tests-mbed_platform-circularbuffer | SYNC_FAILED | 37.44              | default     |
mbedgt: test suite results: 1 SYNC_FAILED
mbedgt: test case report:
| target                | platform_name | test suite                                 | test case                                  | passed | failed | result | elapsed_time (sec) |
|-----------------------|---------------|--------------------------------------------|--------------------------------------------|--------|--------|--------|--------------------|
| NUCLEO_F446ZE-GCC_ARM | NUCLEO_F446ZE | mbed-os-tests-mbed_platform-circularbuffer | mbed-os-tests-mbed_platform-circularbuffer | 0      | 1      | ERROR  | 37.44              |
mbedgt: test case results: 1 ERROR
mbedgt: completed in 40.07 sec
mbedgt: exited with code 1
[mbed] ERROR: "mbedgt" returned error.
       Code: 1
       Path: "c:\work\test_consumption\nucleo_consuption"
       Command: "mbedgt --test-spec .\BUILD\tests\NUCLEO_F446ZE\GCC_ARM\test_spec.json -n mbed-os-tests-mbed_platform-circularbuffer"
       Tip: You could retry the last command with "-v" flag for verbose output
---

What means SYNC_FAILED ? My nucleo looks fine and works good
```

0xc0170, 2019-09-17:
@squli What firmware is on your STLINK ? I found few older issues where the firmware update fixed the sync problem. See for instance: https://github.com/ARMmbed/mbed-stress-test/issues/4

LMESTM, 2019-09-17:
@squli also you can add -v option to the mbed test command to get more debug information.

I'm surprised that the test runs at 115200 bauds, usually they run at 9600.
`mbedgt: ['mbedhtrun', '-m', u'NUCLEO_F446ZE', '-p', u'COM9:115200',
`

You may try to run the test directly from mbed-os directory instead of your application root directory. 
`mbed test -m  NUCLEO_F446ZE -t GCC_ARM -n tests-mbed_platform-circularbuffer -v`

squli, 2019-09-17:
`@squli What firmware is on your STLINK ?`
It has V2.J33.M25 and doesnt need to update.

`I'm surprised that the test runs at 115200 bauds, usually they run at 9600.`
I cant find where i can set this value. May be you can tell me?

`@squli also you can add -v option to the mbed test command to get more debug information.`
-v show my error:
```
Build successes:
  * NUCLEO_F446ZE::GCC_ARM::MBED-BUILD
  * NUCLEO_F446ZE::GCC_ARM::MBED-OS-TESTS-MBED_PLATFORM-CIRCULARBUFFER
[mbed] Exec "mbedgt --test-spec .\BUILD\tests\NUCLEO_F446ZE\GCC_ARM\test_spec.json -n mbed-os-tests-mbed_platform-circularbuffer -V" in "c:\work\test_consumption\nucleo_consuption"
mbedgt: greentea test automation tool ver. 1.5.0
mbedgt: test specification file '.\BUILD\tests\NUCLEO_F446ZE\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using '.\BUILD\tests\NUCLEO_F446ZE\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        |---------------|----------------------|-------------|-------------|--------------------------|
        | NUCLEO_F446ZE | NUCLEO_F446ZE[0]     | COM9        | D:          | 07780221033361293C77F76C |
mbedgt: processing target 'NUCLEO_F446ZE' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        |---------------|----------------------|-------------|-------------|--------------------------|
        | NUCLEO_F446ZE | NUCLEO_F446ZE[0]     | COM9:115200 | D:          | 07780221033361293C77F76C |
mbedgt: test case filter (specified with -n option)
        mbed-os-tests-mbed_platform-circularbuffer
        test filtered in 'mbed-os-tests-mbed_platform-circularbuffer'
mbedgt: running 1 test for platform 'NUCLEO_F446ZE' and toolchain 'GCC_ARM'
        use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        found 'host_tests' directory in: 'mbed-os\TESTS\host_tests'
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m NUCLEO_F446ZE -p COM9:115200 -f "BUILD/tests/NUCLEO_F446ZE/GCC_ARM/mbed-os/TESTS/mbed_platform/CircularBuffer/CircularBuffer.bin" -e "mbed-os\TESTS\host_tests" -d D: -c default -t 07780221033361293C77F76C -r default -C 4 --sync 5 -P 60
mbedgt: mbed-host-test-runner: started
[1568720594.38][HTST][INF] host test executor ver. 0.0.1
[1568720594.38][HTST][INF] copy image onto target...
[1568720594.38][COPY][INF] Waiting up to 60 sec for '07780221033361293C77F76C' mount point (current is 'D:')...
Exception in thread Thread-3:
Traceback (most recent call last):
  File "C:\Python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\Python27\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "C:\Python27\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 503, in run_test_thread
    verbose=verbose)
  File "C:\Python27\lib\site-packages\mbed_greentea\mbed_test_api.py", line 326, in run_host_test
    returncode, htrun_output = run_htrun(cmd, verbose)
  File "C:\Python27\lib\site-packages\mbed_greentea\mbed_test_api.py", line 144, in run_htrun
    sys.stdout.write(decoded_line.rstrip() + '\n')
  File "C:\Python27\lib\site-packages\colorama\ansitowin32.py", line 40, in write
    self.__convertor.write(text)
  File "C:\Python27\lib\site-packages\colorama\ansitowin32.py", line 141, in write
    self.write_and_convert(text)
  File "C:\Python27\lib\site-packages\colorama\ansitowin32.py", line 169, in write_and_convert
    self.write_plain_text(text, cursor, len(text))
  File "C:\Python27\lib\site-packages\colorama\ansitowin32.py", line 174, in write_plain_text
    self.wrapped.write(text[start:end])
  File "C:\Python27\lib\site-packages\colorama\ansitowin32.py", line 40, in write
    self.__convertor.write(text)
  File "C:\Python27\lib\site-packages\colorama\ansitowin32.py", line 141, in write
    self.write_and_convert(text)
  File "C:\Python27\lib\site-packages\colorama\ansitowin32.py", line 169, in write_and_convert
    self.write_plain_text(text, cursor, len(text))
  File "C:\Python27\lib\site-packages\colorama\ansitowin32.py", line 174, in write_plain_text
    self.wrapped.write(text[start:end])
  File "C:\Python27\lib\encodings\cp866.py", line 12, in encode
    return codecs.charmap_encode(input,errors,encoding_map)
UnicodeEncodeError: 'charmap' codec can't encode character u'\u0ba2' in position 0: character maps to <undefined>

mbedgt: could not generate test report
mbedgt: completed in 7.35 sec
mbedgt: exited with code -1000
[mbed] ERROR: "mbedgt" returned error.
       Code: -1000
       Path: "c:\work\test_consumption\nucleo_consuption"
       Command: "mbedgt --test-spec .\BUILD\tests\NUCLEO_F446ZE\GCC_ARM\test_spec.json -n mbed-os-tests-mbed_platform-circularbuffer -V"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

LMESTM, 2019-09-17:
@0xc0170 any idea ?

0xc0170, 2019-09-18:
The only related issue I could find was : https://github.com/ARMmbed/mbed-os/issues/4318

This one comes from: 
>   File "C:\Python27\lib\site-packages\mbed_greentea\mbed_test_api.py", line 144, in run_htrun
    sys.stdout.write(decoded_line.rstrip() + '\n')

@squli What is your greentea version? If you can update to the latest version and we should take it from there

squli, 2019-11-05:
Hello! I try to fix this issue.

Look at my debug log:
> 
> [DBG ][QSPIF]: basic_param_table_ptr[QSPIF_BASIC_PARAM_TABLE_QPI_READ_SUPPORT_BYTE] = 254
> [DBG ][QSPIF]: Read Bus Mode set to 4-4-4, Instruction: 0xebh
> [DBG ][QSPIF]: Read Bus Mode set to 1-4-4, Instruction: 0xebh
> **[DBG ][QSPIF]: Reading Config Register Success: value = 0x60
> [DBG ][QSPIF]: fast mode enable - Writing Config Register Success: value = 0x62
> [DBG ][QSPIF]: Verifying Config Register Success: value = 0x60**
> [DBG ][QSPIF]: Init - Setting Quad Enable
> [DBG ][QSPIF]: Device Does not Have a QE Bit, continue based on Read Inst 0xeb
                                                                   
I marked bold strange peace of log. Error was here - W25Q32 has QE bit in status register 2, and to write and read this register uses 0x31 and 0x35 commands. 
I change _enable_fast_mdoe function to correctly set this bit and memory began work.

new log:

> [DBG ][QSPIF]: Read Bus Mode set to 4-4-4, Instruction: 0xebh
> [DBG ][QSPIF]: Read Bus Mode set to 1-4-4, Instruction: 0xebh
> [DBG ][QSPIF]: Reading Config Register Success: value = 0x0
> [DBG ][QSPIF]: fast mode enable - Writing Config Register Success: valu
> e = 0x2
> [DBG ][QSPIF]: Verifying Config Register Success: value = 0x2
> [DBG ][QSPIF]: Init - Setting Quad Enable
> [DBG ][QSPIF]: Device Does not Have a QE Bit, continue based on Read In....

my changed function:
```
int QSPIFBlockDevice::_enable_fast_mdoe()
{
    char status_reg[QSPI_MAX_STATUS_REGISTER_SIZE] = {0};
    unsigned int read_conf_register2_inst = 0x35;
    unsigned int write_conf_register2_inst = 0x31;
    unsigned int read_conf_register_inst_size = 1;
    char old_status_reg2 = 0x00;
    char new_status_reg2 = 0x00;
    char status_reg_qer_setup[QSPI_MAX_STATUS_REGISTER_SIZE] = {0};

    status_reg_qer_setup[2] = 0x2; // Bit 1 of config Reg 2

    // Configure  BUS Mode to 1_1_1 for all commands other than Read
    _qspi_configure_format(QSPI_CFG_BUS_SINGLE, QSPI_CFG_BUS_SINGLE, QSPI_CFG_ADDR_SIZE_24, QSPI_CFG_BUS_SINGLE,
                           QSPI_CFG_ALT_SIZE_8, QSPI_CFG_BUS_SINGLE, 0);

    // Read Status Register
    if (QSPI_STATUS_OK == _qspi_send_general_command(read_conf_register2_inst, QSPI_NO_ADDRESS_COMMAND, NULL, 0,
                                                     &old_status_reg2,
                                                     read_conf_register_inst_size)) {  // store received values in status_value
        tr_debug("Reading Config Register Success: value = 0x%x", (int)old_status_reg2);
    } else {
        tr_error("Reading Config Register failed");
        return -1;
    }

    // Set Bits for Quad Enable
    //for (int i = 0; i < QSPI_MAX_STATUS_REGISTER_SIZE; i++) {
    //    status_reg[i] |= status_reg_qer_setup[i];
    //}
    status_reg[1] = old_status_reg2 | 0x02;

    // Write new Status Register Setup
    if (_set_write_enable() != 0) {
        tr_error("Write Enabe failed");
        return -1;
    }

    if (QSPI_STATUS_OK == _qspi_send_general_command(write_conf_register2_inst, QSPI_NO_ADDRESS_COMMAND, &status_reg[1],
                                                     read_conf_register_inst_size, NULL,
                                                     0)) {   // Write Fast mode bit to status_register
        tr_debug("fast mode enable - Writing Config Register Success: value = 0x%x",
                 (int)status_reg[1]);
    } else {
        tr_error("fast mode enable - Writing Config Register failed");
        return -1;
    }

    if (false == _is_mem_ready()) {
        tr_error("Device not ready after write, failed");
        return -1;
    }

    // For Debug
    memset(status_reg, 0, QSPI_MAX_STATUS_REGISTER_SIZE);
    if (QSPI_STATUS_OK == _qspi_send_general_command(read_conf_register2_inst, QSPI_NO_ADDRESS_COMMAND, NULL, 0,
                                                     &new_status_reg2,
                                                     read_conf_register_inst_size)) {  // store received values in status_value
        tr_debug("Verifying Config Register Success: value = 0x%x", (int)new_status_reg2);
    } else {
        tr_error("Verifying Config Register failed");
        return -1;
    }
    
    if (new_status_reg2 != (old_status_reg2 | 0x02)) {
        tr_error("Verifying Config Register failed");
        return -1;
    }

    return 0;
}
```

0xc0170, 2019-11-05:
@squli You want to send PR fixing this ?

There's another opened PR addressing some QSPI block device as well: https://github.com/ARMmbed/mbed-os/pull/11531 - might be fixed there as well

adbridge, 2020-10-15:
As there has been no update or response on this issue for a year, I am closing it. If the issue is still relevant, please raise an updated issue.

closed 2020-10-15 by adbridge