ARMmbed/mbed-os #11416: STM32F303RE nightly CI watchdog restart failing

Tharazi97, 2019-09-05:
### Description

STM32F303RE is recently failing CI nightly on the tests:

- tests-mbed_hal-watchdog_reset
- tests-mbed_hal-reset_reason
- tests-mbed_drivers-watchdog_reset
- tests-mbed_drivers-reset_reason

All of them by the same fault ``:FAIL: Watchdog did not reset the device as expected.`` 
I have tried to reproduce the problem on my board, but I couldn't. Looking in the implementation of STM watchdogs I found that there is no difference between this target and other STM targets, but on other boards these tests pass.

Maybe it's problem with hardware on this specific board used on CI.
### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jeromecoutant, 2019-09-05:
Hi

Test on my side:

| target              | platform_name | test suite                        | result | elapsed_time (sec) | copy_method |
|---------------------|---------------|-----------------------------------|--------|--------------------|-------------|
| NUCLEO_F303RE-ARMC6 | NUCLEO_F303RE | tests-mbed_drivers-watchdog       | OK     | 19.85              | default     |
| NUCLEO_F303RE-ARMC6 | NUCLEO_F303RE | tests-mbed_drivers-watchdog_reset | OK     | 26.72              | default     |
| NUCLEO_F303RE-ARMC6 | NUCLEO_F303RE | tests-mbed_hal-watchdog           | OK     | 26.96              | default     |
| NUCLEO_F303RE-ARMC6 | NUCLEO_F303RE | tests-mbed_hal-watchdog_reset     | OK     | 22.83              | default     |
| NUCLEO_F303RE-ARMC6 | NUCLEO_F303RE | tests-mbed_hal-watchdog_timing    | OK     | 37.35              | default     |

> Maybe it's problem with hardware on this specific board used on CI.

Yes....

jamesbeyond, 2019-09-05:
Thanks @jeromecoutant we'll check the target in our CI

jeromecoutant, 2019-09-10:
Hi
Any update?
Regards,

0xc0170, 2019-09-10:
I assume we still use the same boards as the latest nightly contains there failures.

 @jeromecoutant What is your test command (compilation flags) ? to check if they match our tests

jeromecoutant, 2019-09-10:
> @jeromecoutant What is your test command (compilation flags) ? to check if they match our tests

mbed test -m NUCLEO_F303RE -t ARM -n *watchdog*

0xc0170, 2019-09-10:
Try this one (might be missing a flag or two, but I keep this one to test locally): 

`mbed test --compile -m NUCLEO_F303RE  -t ARM -n *watchdog -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 -DMBED_ALL_STATS_ENABLED -DMBED_CPU_STATS_ENABLED=1`

Then run each watchdog test via `mbedgt -n tests-........ -v -V` - I use verbose to get all details

0xc0170, 2020-02-19:
I don't see this failure in nightly anymore. @jamesbeyond can this be closed?

jamesbeyond, 2020-02-19:
belive fixed. closing

closed 2020-02-19 by jamesbeyond