ARMmbed/mbed-os #11422: STM32H743 : tests-mbed_platform-crash_reporting test failing

VVESTM, 2019-09-05:
### Description

Following latest patches done on STM32H743 for ethernet (PR #11274), the test **"tests-mbed_platform-crash_reporting"** is now failing. This seems to be due to RAM relocation from 0x20000000 to 0x24000000.

In order to make this test a success, it seems that we need to have crash data ram partition in 0x20000000 area.
So, I decided to use this memory layout : 
- 0x20000000 => int vect (as before ethernet patches)
- 0x20000298 => crash data ram (as before ethernet patches)
- 0x24000000 => ram (mandatory for ethernet)

Source code is in my local repository : [vve_h7_ram branch](https://github.com/ARMmbed/mbed-os/compare/master...VVESTM:vve_h7_ram)

**This is working fine for ARM and GCC_ARM toolchains.**
**For IAR toolchain, I cannot have it working**. This command is always returning an error : 
```
mbed test -t IAR -m NUCLEO_H743ZI -v -n tests-mbed_platform-crash_reporting
(note that in debug mode, with --profile=tools/profiles/debug.json, the test is passed)
```

I have investigated on this issue but cannot conclude for now. Any help is welcome...
Here is my first analysis : 

When rebooting the board, in [mbed_error_initialize ](https://github.com/ARMmbed/mbed-os/blob/master/platform/source/mbed_error.c#L214) function, we calculate the crc of report_error_ctx and check if it is similar to the one stored in crash data ram partition. In our case, it is different. The crc stored in crash data ram partition has been corrupted. (in fact, the last bytes is removed).
It seems I have a corruption in the crc when rebooting the board. I checked the value of this crc at several places : 
- in __NVIC_SystemReset() => called during "shutdown", data is the right one
- in SystemInit() => called at init, data is corrupted

I don't know where to look between those 2 functions...

I have also tried to add dummy data in [_mbed_error_ctx structure](https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_error.h#L833). If I add "char dummy[5];" at the end of the structure, the test is passed. It is the end of the structure that is corrupted.

Any help is welcome...
(I have trouble to use debugger as the test itself is making a reboot...)

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-09-05:
@ARMmbed/team-st-mcd Please review

VVESTM, 2019-09-10:
After some more tests, I can find a workaround... If I write 2 times the crc in the crash data ram partition, the test is passed. See [commit](https://github.com/ARMmbed/mbed-os/commit/93f9848cb65130b83580a1e21097c1f6076bd263) in my [branch](https://github.com/ARMmbed/mbed-os/compare/master...VVESTM:vve_h7_ram). 

I don't understand why yet. It seems that the last write to this partition is failing. (I have tried to add DMB, cache...)
@kjbracey-arm, this is following the question I asked you by mail.

Also, strange thing is that it appears only with IAR in release mode. (IAR debug mode is OK, ARM and GCC_ARM are OK too).

VVESTM, 2019-09-18:
I exchanged on this subject internally and this is linked to the ECC cache mechanism. Actually, there is no way to ensure that the last byte is written before a system reset occurs.

If we want to avoid this situation, we need to write an extra data before the reset. 
The same discussion has already occurs on [ST Community](https://community.st.com/s/question/0D50X0000AupcL4SQI/axi-sram-losing-last-few-writes-across-reset).

So, I will push a PR in order to modify H7 linker scripts to move crash data ram into DTCMRAM and add a dummy in mbed_error_ctx structure.

SeppoTakalo, 2019-09-18:
I think this belongs to @ARMmbed/mbed-os-core not ip-core

VVESTM, 2020-10-05:
Seen with @jeromecoutant, I close this issue.

NUCLEO_H743ZI is deprecated.
Issue only with IAR.

closed 2020-10-05 by VVESTM