ARMmbed/mbed-os #11426: Problems compiling mbed on Nordic nRF51-DK

Tharazi97, 2019-09-05:
### Description

Compiling any program on newest Master branch produces this error:
```
++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: handler == (IRQHandler_t)NVIC_GetVector(irq)
Location: 0x1FD61
File: .\rtos\source\TARGET_CORTEX\mbed_rtx_idle.cpp+67
Error Value: 0x0
Current Thread: IL$hB
                                                KB
                                                  KJ`hIB
KGJ`h
IBIBHh!
C`[hGF  Id: 0x0 Entry: 0xE5 Sta...

For more info, visit: https://mbed.com/s/error?error=0x80FF0144&osver=999999&core=0x410CC200&comp=2&ver=60300&tgt=NRF51_DK
 -- MbedOS Error Info --
```
I am investigating whats causing this problem. Everything works ok on Mbed-os-5.13.4

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-09-05:
This looks like a runtime error rather than compiling. What compilers have been tested?

Tharazi97, 2019-09-05:
Yes, sorry I meant, a runtime error, compiling works good. I have testes it only on GCC_ARM, because I don't have acces to other compilers. Currently I am bisecting commits to look where the problem occured for the first time.

Tharazi97, 2019-09-06:
It started failing since this commit #10104
@kjbracey-arm

maciejbocianski, 2019-09-09:
@jamesbeyond

jamesbeyond, 2019-09-10:
Could you make a comment on this @kjbracey-arm

maciejbocianski, 2019-09-11:
I'm looking in to it...

maciejbocianski, 2019-09-13:
Fix: https://github.com/ARMmbed/mbed-os/pull/11479

0xc0170, 2019-09-17:
The fix was integrated, closing as resolved

closed 2019-09-17 by 0xc0170