ARMmbed/mbed-os #11428: NRF52480: baremetal build fails due to cryptocell dependency on mbedtls

0xc0170, 2019-09-06:
### Description

Reference: https://github.com/ARMmbed/mbed-os/pull/11418. As we fixed config issue for some NRF52 targets, we identified another issue causing baremetal build to fail. mbedtls is excluded from the build as its not available in the baremetal build. However, some NRF52 targets have feature cryptocell included in their target configuration.

Cryptocell feature depends on mbedtls. Therefore it should not be available in the baremetal build. A question is : how to disable it for baremetal build? There are two concepts: requires and features. We can't just remove the feature from targets as it might be used, so how to make it available only for non-baremetal build?

This will be known issue for 5.14.0 most likely.

@ARMmbed/mbed-os-tools Please review

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-09-06:
cc @RonEld

Patater, 2019-10-11:
What changes would be necessary (perhaps to Mbed OS build tools) in order to make the baremetal build capable of excluding cryptocell?

RonEld, 2019-10-15:
There isn't a real dependency in Cryptocell to Mbed TLS.
The dependency is in the porting layer between Mbed TLS API.
One can still call the Cryptocell API directly, without the need for Mbed TLS API.
However, this needs to be further investigated, to understand what the build failures are, and whether there is a simpler wayto keep same API.

Excluding Cryptocell basically means removing the `CRYPTOCELL310` from the target feature

Patater, 2019-10-15:
We'd like to exclude CRYPTOCELL310 from the target features, but just for baremetal.

RonEld, 2019-10-15:
I think this would be good as a temporary patch, but if Cryptocell wouldn't increase much RAM, wouldn't it be an advantage to have it?

Patater, 2019-10-15:
I think including it would be against the aims of the baremetal profile.

RonEld, 2019-10-15:
Wouldn't we want cryptography in baremetal, if possible?

Anyway, in baremetal profile, one should remove the `CRYPTOCELL310`, ( and possibly also `TRNG` and `MBEDTLS_CONFIG_HW_SUPPORT`), similar to what is done for [`MTB_LAIRD_BL654`](https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L7753)

loverdeg-ep, 2019-10-15:
We would like Cryptocell to be available in bare-metal if possible.
@trowbridgec @AGlass0fMilk @farrenv

Patater, 2019-10-16:
> Wouldn't we want cryptography in baremetal, if possible?

I'd like to know the reasons why Mbed TLS isn't in the baremetal profile. Maybe the same reasoning would apply to why not CryptoCell.

If we add Mbed TLS to baremetal, this target would would fine with `CRYPTOCELL310`.

How much flash does enabling `CRYPTOCELL310` require?

0xc0170, 2019-10-21:
> I'd like to know the reasons why Mbed TLS isn't in the baremetal profile. Maybe the same reasoning would apply to why not CryptoCell.

I would not expect having TLS in baremetal but definitely should work with it. 

> We would like Cryptocell to be available in bare-metal if possible.

@loverdeg-ep You would vote then for having mbedtls as well in baremetal.

cc @bulislaw

loverdeg-ep, 2019-10-21:
@0xc0170 I would just like for it to work with bare-metal. 
Cryptocell does not need to be the default.

RonEld, 2019-10-22:
@0xc0170 
>I would not expect having TLS in baremetal but definitely should work with it.

I agree that TLS should probably not be by default in baremetal, but I would say that the crypto part should be in baremetal

bulislaw, 2019-11-06:
I don't know why TLS wouldn't be available in baremetal. From what I remember it was initially excluded because of work/testing needed and the scope of work. In practise I can easily find a use cases to have non-RTOS build with TLS. So I think we should work on getting TLS to work with baremetal. Another question, is presence of cryptocell inflating the image even if it's not used?

amitchone, 2019-11-26:
Just chiming in. FWIW I would also find a use-case for TLS in baremetal. At the moment I'm using the full RTOS just for periodic wakes/TX due to the ease of use with LoRa/TLS. It's not a huge issue of course, just a nice-to-have.

PR #12166 (2020-02-14): NRF52840_DK: Fix baremetal linker error

### Summary of changes 

Fixes #11428 
Compile in the inclusion of cryptocell310 only if the library is included
in the build

#### Impact of changes 

It is now possible to build MCU_NRF52840 targets with the baremetal profile.
i.e It is now possible to build `mbed-os-example-blinky-baremetal` with `$ mbed compile -t <TOOLCHAIN> -m NRF52840_DK`
closed 2020-02-14 by 0xc0170