ARMmbed/mbed-os #11432: STM32F3** compile warning

pilotak, 2019-09-06:
### Description
When compiling STM32F3 series there is a warning
```
[Warning] stm32f3xx_ll_tim.h@3526,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f3xx_ll_tim.h@3545,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
```
the **`tmpreg`** is not used
https://github.com/ARMmbed/mbed-os/blob/87cdef9258960e1269b1cf2ba4da27557e2fb788/targets/TARGET_STM/TARGET_STM32F3/device/stm32f3xx_ll_tim.h#L3518-L3546
I belive the fix is

```diff
__STATIC_INLINE void LL_TIM_EnableBRK(TIM_TypeDef *TIMx)
{
-  __IO uint32_t tmpreg; 
-
  SET_BIT(TIMx->BDTR, TIM_BDTR_BKE);
-
-  /* Note: Any write operation to this bit takes a delay of 1 APB clock cycle to become effective. */
-  tmpreg = READ_REG(TIMx->BDTR);
-  (void)(tmpreg);
}
```

```diff
__STATIC_INLINE void LL_TIM_DisableBRK(TIM_TypeDef *TIMx)
{
-  __IO uint32_t tmpreg;
-
  CLEAR_BIT(TIMx->BDTR, TIM_BDTR_BKE);
-
-  /* Note: Any write operation to this bit takes a delay of 1 APB clock cycle to become effective. */
-  tmpreg = READ_REG(TIMx->BDTR);
-  (void)(tmpreg);
}
```

### Issue request type

    [ ] Question
    [*] Enhancement
    [] Bug

0xc0170, 2019-09-06:
cc @ARMmbed/team-st-mcd Please review

LMESTM, 2019-09-09:
@pilotak considering the note above, I wouldn't use your fix right away.
`/* Note: Any write operation to this bit takes a delay of 1 APB clock cycle to become effective. */`

Also the (void) line is supposedly here to avoid unused warning but is implying another one ... we will report it internally to get the proper fix

pilotak, 2019-09-09:
ok, i compared it with STM32F4 series and all that stuff is not there. It actually reads some setting which is not used afterwards, but it might me some workaround. Anyway i let you fix that.

The reason i would like it to be fixed it because when compiling this warning is after every compiled file which is anoying

kjbracey-arm, 2019-09-09:
The register readback is clearly desired - the code is weird and overdoing it though.

Assuming a working compiler, you don't need to actually put the read result anywhere - `tmpreg` doesn't need to exist, let along be `volatile`/`__IO`. Just

    READ_REG(TIMx->BDTR); 

would do. You could stick a `(void)` in front of that, but doesn't achieve much - the compiler is required to do the read, because `BDTR` is marked volatile, so reads have side-effects. If you're the sort of person who puts `(void)` in front of function calls where you don't do the results, it would match that.

I don't know if the code is like it is to work around some sort of past compiler - trying to do a "double volatile" to make it "really, really work". The compiler is just pointing out that it's nonsense.

If these are "interrupt enable/disable" functions, then technically a little bit more care is needed with barriers, to stop surrounding stuff getting reordered. You presumably want the read to happen before some following stuff?

See `NVIC_EnableIRQ` and `NVIC_DisableIRQ` in CMSIS, and discussion here, where  `NVIC_EnableIRQ` was recently revised:

https://github.com/ARM-software/CMSIS_5/issues/493

jeromecoutant, 2019-09-09:
ST_INTERNAL_REF 71851

jeromecoutant, 2019-09-09:
Hi @pilotak 
I couldn't reproduce yet your compilation warning...
I tried IAR, GCC and ARM...
Regards,

pilotak, 2019-09-09:
latest master https://github.com/ARMmbed/mbed-os/commit/372682854b7d87e32cb5da1b656133baaee1821
 ARM GCC 6-2017-q2-update

```cpp
#include "mbed.h"

int main() {
    printf("test\n");
}
```

```sh
mbed compile -m NUCLEO_F303RE
```

```
Compile [ 15.5%]: DeviceKey.cpp
[Warning] stm32f3xx_ll_tim.h@3526,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f3xx_ll_tim.h@3545,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 15.6%]: mbed_trace.c
Compile [ 15.7%]: ns_list.c
Compile [ 15.8%]: common_functions.c
Compile [ 15.9%]: stoip4.c
Compile [ 16.0%]: ip4tos.c
Compile [ 16.2%]: stoip6.c
Compile [ 16.3%]: greentea_test_env.cpp
[Warning] stm32f3xx_ll_tim.h@3526,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f3xx_ll_tim.h@3545,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 16.4%]: ip6tos.c
Compile [ 16.5%]: ns_nvm_helper.c
Compile [ 16.6%]: nsdynmemLIB.c
Compile [ 16.7%]: unity.c
Compile [ 16.8%]: utest_stack_trace.cpp
Compile [ 16.9%]: mbed-utest-shim.cpp
[Warning] stm32f3xx_ll_tim.h@3526,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f3xx_ll_tim.h@3545,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 17.1%]: unity_handler.cpp
[Warning] stm32f3xx_ll_tim.h@3526,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f3xx_ll_tim.h@3545,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 17.2%]: utest_case.cpp
[Warning] stm32f3xx_ll_tim.h@3526,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f3xx_ll_tim.h@3545,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 17.3%]: utest_default_handlers.cpp
[Warning] stm32f3xx_ll_tim.h@3526,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f3xx_ll_tim.h@3545,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 17.4%]: utest_greentea_handlers.cpp
[Warning] stm32f3xx_ll_tim.h@3526,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f3xx_ll_tim.h@3545,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 17.5%]: utest_shim.cpp
[Warning] stm32f3xx_ll_tim.h@3526,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f3xx_ll_tim.h@3545,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 17.6%]: utest_harness.cpp
[Warning] stm32f3xx_ll_tim.h@3526,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f3xx_ll_tim.h@3545,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 17.7%]: utest_types.cpp
[Warning] stm32f3xx_ll_tim.h@3526,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
[Warning] stm32f3xx_ll_tim.h@3545,10: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
Compile [ 17.9%]: LoRaMacChannelPlan.cpp
Compile [ 18.0%]: LoRaMacCrypto.cpp
Compile [ 18.1%]: LoRaMacCommand.cpp
Compile [ 18.2%]: LoRaPHYAS923.cpp
Compile [ 18.3%]: LoRaMac.cpp
[Warning] LoRaMac.h@691,26: 'LoRaMac::_ongoing_tx_msg' will be initialized after [-Wreorder]
[Warning] LoRaMac.h@689,28:   'loramac_mlme_confirm_t LoRaMac::_mlme_confirmation' [-Wreorder]
[Warning] LoRaMac.cpp@68,1:   when initialized here [-Wreorder]
Compile [ 18.4%]: LoRaPHY.cpp
Compile [ 18.5%]: LoRaPHYCN470.cpp
Compile [ 18.6%]: LoRaWANInterface.cpp
```

jeromecoutant, 2019-09-09:
OK, I am using GCC 7.3.1

pilotak, 2019-09-09:
Is it better for me to upgrade? this would solve the issue on my side but doen't fix this

LMESTM, 2019-09-17:
I would suggest to upgrade the compiler version for now.
stm32f3xx_ll_tim.h is primarily developed for C (not C++) framework and this warning only appears when compiling a C++ source code file that includes the header. Considering the warning doesn't exist when compiling C source file and the next version of compiler doesn't show this warning, I'm afraid that the team in charge will not have a closer look at the warning. 
Nevertheless I shared previous comment from @kjbracey-arm to see if code can be improved regardless of the warning.

pilotak, 2019-09-18:
fair enough, thanks for looking into it

closed 2019-09-18 by pilotak