ARMmbed/mbed-os #11436: Error running make twice in makefile project

matthew-macovsky-cypress, 2019-09-06:
### Description

Running the following sequence of commands:

> mbed import mbed-os-example-blinky
> cd mbed-os-example-blinky/mbed-os
> git checkout master
> cd ..
> mbed export -m CY8CPROTO_062_4343W -i eclipse_gcc_arm
> make
> make

results in the output:

>  g++     .././mbed-os/platform/cxxsupport/mstd_mutex.cpp   -o .././mbed-os/platform/cxxsupport/mstd_mutex
> .././mbed-os/platform/cxxsupport/mstd_mutex.cpp:18:10: fatal error: mstd_mutex: No such file or directory
>  #include <mstd_mutex>
>           ^~~~~~~~~~~~
> compilation terminated.
> make[1]: *** [<builtin>: .././mbed-os/platform/cxxsupport/mstd_mutex] Error 1
> make: *** [Makefile:26: all] Error 2

I believe the issue is the result of the following:

1. On the first run of make, dependency (.d) files are created, listing mstd_mutex as a dependency of mstd_mutex.cpp
2. On the second run of make, the dependency files are included
3. Make is unable to match mstd_mutex to a rule, and so
4. Make uses a default rule to attempt to compile mstd_mutex as an object file

I believe that a possible solution is to add the line `%: ;` to the end of the Makefile template.  This would provide a default do-nothing rule for files that do not match other rules (such as mstd_mutex) and should not affect other targets.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-09-09:
@matthew-macovsky-cypress Can you create PR with a fix?

cc @ARMmbed/mbed-os-tools

matthew-macovsky-cypress, 2019-09-09:
PR: #11445