ARMmbed/mbed-os #11442: UDPSocket::sendto does not work with ipv6 and local-link interface addresss

slavaDev503, 2019-09-09:
### Description
UDP socket can`t send data to ipv6 destination address. The project with mbed os 5.13.0 work properly. But if I update mbed os to 5.13.1 or higher I have an error.

I research this problem and get code which work wrong (commit b272c5f14909705b19397263ec274e45b9e36507):
``` C++
 struct netif *netif_ = netif_get_by_index(s->conn->pcb.ip->netif_idx);
    if (!netif_) {
        netif_ = &default_interface->netif;
    }
    if (netif_) {
        if ((addr.version == NSAPI_IPv4 && !get_ipv4_addr(netif_)) ||
                (addr.version == NSAPI_IPv6 && !get_ipv6_addr(netif_))) {
            return NSAPI_ERROR_PARAMETER;
        }
    }
```
Function `get_ipv6_addr` iterate over all available ipv6 addresses and skip local-link address which is set during network initialization (`EMACInterface::connect()->LWIP::Interface::bringup() -> netif_create_ip6_linklocal_address()`). Because there is no one other ipv6 address in interface  function return NULL. After that `return NSAPI_ERROR_PARAMETER` occur and socket sendto return error -3003. 

target STM32F446ZE on nucleo board
toolchain GCC_ARM 6 2017-q2-update
mbed cli

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-09-12:
@tymoteuszblochmobica Please review 

cc @ARMmbed/mbed-os-ipcore

tymoteuszblochmobica, 2019-09-30:
If adress is link local LWIP returns NULL.
const ip_addr_t *LWIP::get_ipv6_addr(const struct netif *netif)
{
#if LWIP_IPV6
    if (!netif_is_up(netif)) {
        return NULL;
    }

    for (int i = 0; i < LWIP_IPV6_NUM_ADDRESSES; i++) {
        if (ip6_addr_isvalid(netif_ip6_addr_state(netif, i)) &&
                !ip6_addr_islinklocal(netif_ip6_addr(netif, i))) {
            return netif_ip_addr6(netif, i);
        }
    }
#endif
    return NULL;
}

@kjbracey-arm    Does  ip6_addr_islinklocal  have special purpose in this function or can it be removed?

michalpasztamobica, 2019-11-20:
@slavaDev503 , the fix is merged, please check if this works fine for you?

0xc0170, 2019-11-20:
As we believe this was fixed, we will closed. In case there is still an issue, please reopen or create a new bug report if are seeing different issue.

closed 2019-11-20 by 0xc0170