ARMmbed/mbed-os #11451: File System Integration test fails on CY8CKIT-062-WIFI-BT

maclobdell, 2019-09-10:
### Description

Target: CY8CKIT-062-WIFI-BT (most likely other Cypress targets too)
Mbed OS branch:  mbed-os-5.14
Toolchain:  ARM compiler 6

Tests Command:
>mbed test -n tests-integration-fs-single* -DINTEGRATION_TESTS -t ARM

Suspected Problem: The heap doesn't seem to be allocated correctly.  Looking at the map file it is not filling the available memory like it should.

Result:

```
| target                    | platform_name       | test suite                      | result  | elapsed_time (sec) | copy_method |
|---------------------------|---------------------|---------------------------------|---------|--------------------|-------------|
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-fs-single     | TIMEOUT | 335.87             | default     |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-fs-threaded   | TIMEOUT | 635.28             | default     |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-net-single    | OK      | 337.32             | default     |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-net-threaded  | OK      | 122.74             | default     |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-stress-net-fs | TIMEOUT | 788.71             | default     |
mbedgt: test suite results: 3 TIMEOUT / 2 OK
mbedgt: test case report:
| target                    | platform_name       | test suite                      | test case                               | passed | failed | result  | elapsed_time (sec) |
|---------------------------|---------------------|---------------------------------|-----------------------------------------|--------|--------|---------|--------------------|
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-fs-single     | QSPIF+LFS 1 file, buff     1            | 1      | 0      | OK      | 4.46               |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-fs-single     | QSPIF+LFS 1 file, buff     4            | 0      | 0      | ERROR   | 0.0                |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-fs-single     | QSPIF+LFS 1 file, buff    16            | 0      | 0      | SKIPPED | 0.0                |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-fs-single     | QSPIF+LFS 1 file, buff    64            | 0      | 0      | SKIPPED | 0.0                |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-fs-single     | QSPIF+LFS 1 file, buff   256            | 0      | 0      | SKIPPED | 0.0                |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-fs-single     | QSPIF+LFS 1 file, buff  1024            | 0      | 0      | SKIPPED | 0.0                |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-fs-single     | QSPIF+LFS 1 file, buff  4096            | 0      | 0      | SKIPPED | 0.0                |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-fs-single     | QSPIF+LFS 1 file, buff 16384            | 0      | 0      | SKIPPED | 0.0                |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-fs-single     | QSPIF+LFS format                        | 1      | 0      | OK      | 1.42               |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-fs-threaded   | QSPIF+LFS 2 files, buff 4b/256b         | 0      | 0      | ERROR   | 0.0                |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-fs-threaded   | QSPIF+LFS 3 files, buff 256b/1kb/4kb    | 0      | 0      | SKIPPED | 0.0                |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-fs-threaded   | QSPIF+LFS format                        | 1      | 0      | OK      | 1.43               |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-net-single    | WiFi   128 buffer                       | 1      | 0      | OK      | 97.68              |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-net-single    | WiFi   256 buffer                       | 1      | 0      | OK      | 49.75              |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-net-single    | WiFi  1024 buffer                       | 1      | 0      | OK      | 16.9               |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-net-single    | WiFi  4096 buffer                       | 1      | 0      | OK      | 9.84               |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-net-single    | WiFi network setup                      | 1      | 0      | OK      | 4.49               |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-net-threaded  | WiFi 1 thread                           | 1      | 0      | OK      | 16.83              |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-net-threaded  | WiFi 2 threads                          | 1      | 0      | OK      | 33.41              |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-net-threaded  | WiFi network setup                      | 1      | 0      | OK      | 5.98               |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-stress-net-fs | QSPIF+LFS format                        | 1      | 0      | OK      | 1.44               |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-stress-net-fs | QSPIF+LFS+WiFi 1 thread, dl, file seq.  | 1      | 0      | OK      | 49.38              |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-stress-net-fs | QSPIF+LFS+WiFi 2 threads, dl, 1kb       | 0      | 0      | ERROR   | 0.0                |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-stress-net-fs | QSPIF+LFS+WiFi 3 threads, dl, 256b, 1kb | 0      | 0      | SKIPPED | 0.0                |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-stress-net-fs | Test memory allocation of 10 K bytes    | 1      | 0      | OK      | 0.06               |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-stress-net-fs | Test memory allocation of 20 K bytes    | 1      | 0      | OK      | 0.07               |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-stress-net-fs | Test memory allocation of 40 K bytes    | 1      | 0      | OK      | 0.14               |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-stress-net-fs | Test memory allocation of 60 K bytes    | 1      | 0      | OK      | 0.06               |
| CY8CKIT_062_WIFI_BT-ARMC6 | CY8CKIT_062_WIFI_BT | tests-integration-stress-net-fs | WiFi network setup                      | 1      | 0      | OK      | 5.97               |
mbedgt: test case results: 18 OK / 3 ERROR / 8 SKIPPED
```

Details: 
Other toolchains pass all tests.

Workaround / Solution: 

If the following lines are removed from the startup file, then all tests pass.  I suspect that these tests are allocating memory, but there is a new regression introduced in Mbed OS 5.14 for the Cypress targets that is preventing the heap from being allocated properly for Mbed OS.

 \targets\TARGET_Cypress\TARGET_PSOC6\TARGET_CY8CKIT_062_WIFI_BT\device\TOOLCHAIN_ARM\startup_psoc6_01_cm4.S

; <h> Heap Configuration
;   <o>  Heap Size (in Bytes) <0x0-0xFFFFFFFF:8>
; </h>
                IF :DEF:__HEAP_SIZE
Heap_Size       EQU     __HEAP_SIZE
                ELSE
Heap_Size       EQU     0x00000400
                ENDIF
                AREA    HEAP, NOINIT, READWRITE, ALIGN=3
__heap_base
Heap_Mem        SPACE   Heap_Size
__heap_limit

cc @ARMmbed/team-cypress 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

yarbcy, 2019-09-13:
@maclobdell 
We tried comment:
\targets\TARGET_Cypress\TARGET_PSOC6\TARGET_CY8CKIT_062_WIFI_BT\device\TOOLCHAIN_ARM\startup_psoc6_01_cm4.S

; Heap Configuration
; Heap Size (in Bytes) <0x0-0xFFFFFFFF:8>
;
IF :DEF:__HEAP_SIZE
Heap_Size EQU __HEAP_SIZE
ELSE
Heap_Size EQU 0x00000400
ENDIF
AREA HEAP, NOINIT, READWRITE, ALIGN=3
__heap_base
Heap_Mem SPACE Heap_Size
__heap_limit

But still see the same issue that you described.

Could you please double check if this fix is work for you?

yarbcy, 2019-09-13:
Please also give us branch that you use and commit.

maclobdell, 2019-09-17:
Unfortunately, the workaround I  proposed, does not seem to work anymore.  It appears the failing tests are resulting in a hard fault.

vmedcy, 2019-09-24:
Hi @maclobdell @hennadiykytsun @yarbcy 

I executed `mbed test -n tests-integration-fs-single* -DINTEGRATION_TESTS -t ARMC6 -m CY8CKIT_062_WIFI_BT -v` on mbed-os-5.14 branch (b6e5a0a8afa34dec9dae8963778aebce0c82a54b), the tests fail with assertion "could not mount block device" at the first test case (QSPIF+LFS format):

```
calling mbedhtrun: mbedhtrun -m CY8CKIT_062_WIFI_BT -p COM26:9600 -f "BUILD/tests/CY8CKIT_062_WIFI_BT/ARMC6/TESTS/integration/fs-single/fs-single.hex" -e "TESTS\host_tests" -d E: -c default -t 19001301f306191901f3061900000000000000002e127069 -r default -C 10 --sync 5 -P 60

...

[1569343210.00][CONN][RXD] >>> Running case #1: 'QSPIF+LFS format'...
[1569343210.05][CONN][INF] found KV pair in stream: {{__testcase_start;QSPIF+LFS format}}, queued...
[1569343210.09][CONN][RXD] lfs error:494: Corrupted dir pair at 0 1
[1569343210.14][CONN][RXD] lfs error:2216: Invalid superblock at 0 1
mbedgt: :70::FAIL: Expected 0 Was -84. could not mount block device

[1569343210.20][CONN][RXD] :70::FAIL: Expected 0 Was -84. could not mount block device
[1569343210.24][CONN][INF] found KV pair in stream: {{__testcase_finish;QSPIF+LFS format;0;1}}, queued...
[1569343210.32][CONN][RXD] >>> 'QSPIF+LFS format': 0 passed, 1 failed with reason 'Assertion Failed'
[1569343210.32][CONN][RXD]
[1569343210.39][CONN][RXD] >>> Test cases: 0 passed, 1 failed with reason 'Assertion Failed'
[1569343210.41][CONN][RXD] >>> TESTS FAILED!

...

mbedgt: test on hardware with target id: 19001301f306191901f3061900000000000000002e127069
mbedgt: test suite 'tests-integration-fs-single' ..................................................... FAIL in 33.10 sec
	test case: 'QSPIF+LFS 1 file, buff     1' .................................................... SKIPPED in 0.00 sec
	test case: 'QSPIF+LFS 1 file, buff     4' .................................................... SKIPPED in 0.00 sec
	test case: 'QSPIF+LFS 1 file, buff    16' .................................................... SKIPPED in 0.00 sec
	test case: 'QSPIF+LFS 1 file, buff    64' .................................................... SKIPPED in 0.00 sec
	test case: 'QSPIF+LFS 1 file, buff   256' .................................................... SKIPPED in 0.00 sec
	test case: 'QSPIF+LFS 1 file, buff  1024' .................................................... SKIPPED in 0.00 sec
	test case: 'QSPIF+LFS 1 file, buff  4096' .................................................... SKIPPED in 0.00 sec
	test case: 'QSPIF+LFS 1 file, buff 16384' .................................................... SKIPPED in 0.00 sec
	test case: 'QSPIF+LFS format' ................................................................ FAIL in 0.19 sec
mbedgt: test case summary: 0 passes, 1 failure
```
(full log attached: [test_CY8CKIT_062_WIFI_BT_ARMC6.log](https://github.com/ARMmbed/mbed-os/files/3648326/test_CY8CKIT_062_WIFI_BT_ARMC6.log))

I executed the same test with different toolchain (GCC_ARM 6), and with board from different vendor: DISCO_L475VG_IOT01A (also using QSPIFBlockDevice as the underlying block device for LittleFS). In all situations, test failed with the identical error.

I checked the QSPI transactions (set few breakpoints in cy_qspi_api.c), was not able to catch any failures.

The assertions from the LittleFS core functions: `lfs error:494: Corrupted dir pair at 0 1`, `lfs error:2216: Invalid superblock at 0 1`, `FAIL: Expected 0 Was -84. could not mount block device` - doesn't indicate where is the real root cause of the failure. Very likely, the issue is not target- or toolchain- specific.

In the problem description, I see different test case is failing - `QSPIF+LFS 1 file, buff     4`, while `QSPIF+LFS format` and `QSPIF+LFS 1 file, buff     1` are padding. Is it possible that the test result depends on the order of integration tests? I have not tried executing everything in batch.

maclobdell, 2019-12-04:
I will run these tests again and check with our team.

maclobdell, 2019-12-04:
@jamesbeyond - can you confirm that these integration tests are working across other boards?

shuopeng-deng, 2019-12-06:
Filed https://github.com/ARMmbed/mbed-os/pull/12044
For fs integration test failures

0xc0170, 2020-01-03:
> Filed #12044

It was integrated. @maclobdell Is this issue still relevant?

SeppoTakalo, 2020-01-10:
This board is not anymore in Mbed OS, so the ticket should be closed.

vmedcy, 2020-01-10:
@SeppoTakalo: can you clarify your last comment? CY8CKIT_062_WIFI_BT is well supported: https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CY8CKIT_062_WIFI_BT
https://os.mbed.com/platforms/CY8CKIT-062-WiFi-BT/

SeppoTakalo, 2020-01-13:
Oh.. It looks like the tittle was using incorrect name. I tried looking for `CY8CKIT-062-WIFI-BT`instead of `CY8CKIT_062_WIFI_BT`

jamesbeyond, 2020-01-13:
> @jamesbeyond - can you confirm that these integration tests are working across other boards?

Sorry for the late reply, but the integration tests are running just fine with `K64F` and `NUCLEO_F429ZI` target

VeijoPesonen, 2020-03-06:
The issues doesn't occur only with ARM compiler but also with GCC. I tested with couple of other boards for comparison.
```
mbedgt: test case report:
| target                      | platform_name       | test suite                    | test case                            | passed | failed | result | elapsed_time (sec) |
|-----------------------------|---------------------|-------------------------------|--------------------------------------|--------|--------|--------|--------------------|
| CY8CPROTO_062_4343W-ARMC6   | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS 1 file, buff     1         | 1      | 0      | OK     | 4.03               |
| CY8CPROTO_062_4343W-ARMC6   | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS 1 file, buff     4         | 1      | 0      | OK     | 2.15               |
| CY8CPROTO_062_4343W-ARMC6   | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS 1 file, buff    16         | 1      | 0      | OK     | 1.89               |
| CY8CPROTO_062_4343W-ARMC6   | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS 1 file, buff    64         | 1      | 0      | OK     | 1.82               |
| CY8CPROTO_062_4343W-ARMC6   | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS 1 file, buff   256         | 1      | 0      | OK     | 1.88               |
| CY8CPROTO_062_4343W-ARMC6   | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS 1 file, buff  1024         | 1      | 0      | OK     | 1.89               |
| CY8CPROTO_062_4343W-ARMC6   | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS 1 file, buff  4096         | 1      | 0      | OK     | 1.85               |
| CY8CPROTO_062_4343W-ARMC6   | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS 1 file, buff 16384         | 1      | 0      | OK     | 1.94               |
| CY8CPROTO_062_4343W-ARMC6   | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS format                     | 1      | 0      | OK     | 1.21               |
| CY8CPROTO_062_4343W-ARMC6   | CY8CPROTO_062_4343W | tests-integration-fs-threaded | QSPIF+LFS 2 files, buff 4b/256b      | 1      | 0      | OK     | 5.14               |
| CY8CPROTO_062_4343W-ARMC6   | CY8CPROTO_062_4343W | tests-integration-fs-threaded | QSPIF+LFS 3 files, buff 256b/1kb/4kb | 0      | 1      | FAIL   | 1.25               |
| CY8CPROTO_062_4343W-ARMC6   | CY8CPROTO_062_4343W | tests-integration-fs-threaded | QSPIF+LFS format                     | 1      | 0      | OK     | 1.24               |
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS 1 file, buff     1         | 1      | 0      | OK     | 3.35               |
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS 1 file, buff     4         | 1      | 0      | OK     | 2.18               |
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS 1 file, buff    16         | 1      | 0      | OK     | 1.84               |
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS 1 file, buff    64         | 1      | 0      | OK     | 1.85               |
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS 1 file, buff   256         | 1      | 0      | OK     | 1.85               |
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS 1 file, buff  1024         | 1      | 0      | OK     | 1.87               |
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS 1 file, buff  4096         | 1      | 0      | OK     | 1.87               |
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS 1 file, buff 16384         | 1      | 0      | OK     | 1.97               |
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-single   | QSPIF+LFS format                     | 1      | 0      | OK     | 1.21               |
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-threaded | QSPIF+LFS 2 files, buff 4b/256b      | 1      | 0      | OK     | 4.95               |
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-threaded | QSPIF+LFS 3 files, buff 256b/1kb/4kb | 0      | 1      | FAIL   | 1.26               |
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-threaded | QSPIF+LFS format                     | 1      | 0      | OK     | 1.19               |
| DISCO_L475VG_IOT01A-ARMC6   | DISCO_L475VG_IOT01A | tests-integration-fs-single   | QSPIF+LFS 1 file, buff     1         | 1      | 0      | OK     | 9.55               |
| DISCO_L475VG_IOT01A-ARMC6   | DISCO_L475VG_IOT01A | tests-integration-fs-single   | QSPIF+LFS 1 file, buff     4         | 1      | 0      | OK     | 6.71               |
| DISCO_L475VG_IOT01A-ARMC6   | DISCO_L475VG_IOT01A | tests-integration-fs-single   | QSPIF+LFS 1 file, buff    16         | 1      | 0      | OK     | 6.02               |
| DISCO_L475VG_IOT01A-ARMC6   | DISCO_L475VG_IOT01A | tests-integration-fs-single   | QSPIF+LFS 1 file, buff    64         | 1      | 0      | OK     | 5.98               |
[... cut ...]
| NRF52840_DK-GCC_ARM         | NRF52840_DK         | tests-integration-fs-threaded | QSPIF+LFS 2 files, buff 4b/256b      | 1      | 0      | OK     | 11.24              |
| NRF52840_DK-GCC_ARM         | NRF52840_DK         | tests-integration-fs-threaded | QSPIF+LFS 3 files, buff 256b/1kb/4kb | 1      | 0      | OK     | 15.94              |
| NRF52840_DK-GCC_ARM         | NRF52840_DK         | tests-integration-fs-threaded | QSPIF+LFS format                     | 1      | 0      | OK     | 0.23               |
mbedgt: test case results: 82 OK / 2 FAIL
mbedgt: completed in 892.52 sec
```

VeijoPesonen, 2020-03-23:
Current situation after applying PR #12668 

```
mbedgt: test case report:
| target                      | platform_name       | test suite                    | test case                                            | passed | failed | result | elapsed_time (sec) |
|-----------------------------|---------------------|-------------------------------|------------------------------------------------------|--------|--------|--------|--------------------|
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-threaded | QSPIF+LFS 2 files, block size 4B/256B                | 1      | 0      | OK     | 6.01               |
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-threaded | QSPIF+LFS 4 files, block size 4B/256B/1KiB/4KiB      | 1      | 0      | OK     | 11.68              |
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-threaded | QSPIF+LFS 5 files, block size 4B/256B/1KiB/4KiB/4KiB | 0      | 1      | FAIL   | 6.48               |
| NRF52840_DK-GCC_ARM         | NRF52840_DK         | tests-integration-fs-threaded | QSPIF+LFS 2 files, block size 4B/256B                | 1      | 0      | OK     | 11.26              |
| NRF52840_DK-GCC_ARM         | NRF52840_DK         | tests-integration-fs-threaded | QSPIF+LFS 4 files, block size 4B/256B/1KiB/4KiB      | 1      | 0      | OK     | 22.06              |
| NRF52840_DK-GCC_ARM         | NRF52840_DK         | tests-integration-fs-threaded | QSPIF+LFS 5 files, block size 4B/256B/1KiB/4KiB/4KiB | 1      | 0      | OK     | 28.04              |
```

The issue with Cypress boards seems to be that it runs out of space - which it shouldn't.

```
[1584966615.00][CONN][RXD] >>> Running case #3: 'QSPIF+LFS 5 files, block size 4B/256B/1KiB/4KiB/4KiB'...
[1584966615.08][CONN][INF] found KV pair in stream: {{__testcase_start;QSPIF+LFS 5 files, block size 4B/256B/1KiB/4KiB/4KiB}}, queued...
[1584966619.01][CONN][RXD] lfs warn:313: No more free space 8
mbedgt: :52::FAIL: Expected 4096 Was 0. failed to write
[1584966619.06][CONN][RXD] :52::FAIL: Expected 4096 Was 0. failed to write
```

Investigation continues

VeijoPesonen, 2020-03-27:
The issue isnt't reproducible once the tests are cleaned up - requires PR # 12668 - and you have swiched to LittleFS**v2**. At the moment you can get LittleFSv2 with following tricks:

```
features/storage/filesystem$ git clone git@github.com:ARMmbed/mbed-littlefs.git littlefsv2
features/storage/filesystem/littlefsv2$ cd littlefsv2
features/storage/filesystem/littlefsv2$ git fetch origin pull/14/head:littlefsv2 && git checkout littlefsv2
features/storage/filesystem/littlefsv2$ cd ../../../../
```

To get the tests running with LittleFSv2:
```
features/storage/filesystem/littlefsv2$ cd ../../../../
$ git diff
diff --git i/TESTS/integration/fs-threaded/main.cpp w/TESTS/integration/fs-threaded/main.cpp
index 4aaaaddfab..0f9e98f84e 100644
--- i/TESTS/integration/fs-threaded/main.cpp
+++ w/TESTS/integration/fs-threaded/main.cpp
@@ -26,7 +26,7 @@
 #endif
 
 #include "mbed.h"
-#include "LittleFileSystem.h"
+#include "LittleFileSystem2.h"
 #include "utest/utest.h"
 #include "unity/unity.h"
 #include "greentea-client/test_env.h"
@@ -57,7 +57,7 @@ void led_thread()
 
 BlockDevice *bd = BlockDevice::get_default_instance();
 SlicingBlockDevice sbd(bd, 0, MBED_CONF_APP_TESTS_FS_SIZE);
-LittleFileSystem fs("sd");
+LittleFileSystem2 fs("sd");
 
 constexpr char fname_prefix[] = "mbed-file-test-";
 constexpr char fname_postfix[] = ".txt";
```

```
mbedgt: test case report:
| target                      | platform_name       | test suite                    | test case                                            | passed | failed | result | elapsed_time (sec) |
|-----------------------------|---------------------|-------------------------------|------------------------------------------------------|--------|--------|--------|--------------------|
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-threaded | QSPIF+LFS 2 files, block size 4B/256B                | 1      | 0      | OK     | 2.71               |
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-threaded | QSPIF+LFS 4 files, block size 4B/256B/1KiB/4KiB      | 1      | 0      | OK     | 5.27               |
| CY8CPROTO_062_4343W-GCC_ARM | CY8CPROTO_062_4343W | tests-integration-fs-threaded | QSPIF+LFS 5 files, block size 4B/256B/1KiB/4KiB/4KiB | 1      | 0      | OK     | 6.4                |
mbedgt: test case results: 3 OK
mbedgt: completed in 42.63 sec
```

So in the end I can't say is this some nasty issue with the target support or something to do with LittleFSv1.

0xc0170, 2020-06-18:
@maclobdell we have littlefs vs 2 already. IS this still related? Also the priority should be cleared?

evedon, 2020-11-09:
@maclobdell Is this issue still relevant? Can you update?

0xc0170, 2020-11-27:
I am closing this. If reopened, it needs updated information.

closed 2020-11-27 by 0xc0170

ifyall, 2021-01-08:
As documented on the platform pages for the Cypress boards, LFSv1 is not supported on those boards. Please use LFSv2:
https://os.mbed.com/platforms/CY8CKIT-062-WiFi-BT/