ARMmbed/mbed-os #11461: DigitalIn/DigitalOut test issue with ST boards due to compilation optimization

jeromecoutant, 2019-09-11:
### Description

Hi

Digital test with CI shield is quite simple:
- 2 pins are connected thanks to the shield
- Out pin is set to 1
- In pin is verified to be 1

We got issues since 5.12.0 label...
Read pin is not 1.

Test become OK if:
- Out pin is set to 1 2 times
- Compilation option is changed from -Os to -O1

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jeromecoutant, 2019-09-11:
https://github.com/ARMmbed/mbed-os/blob/2cd10fc7867f9b9447f379c09df99a7ca9dec57b/drivers/DigitalOut.h#L81

If I add a DMB or DSB after the gpio_write, test is also OK

0xc0170, 2019-09-12:
What's the compiler exposing this? What are the instructions in gpio_write - what does it exactly do?

kjbracey-arm, 2019-09-12:
What CPU?

If you're doing a loopback thing, you might have to worry about timing here.

If the input and output were different peripherals, the order of the read and write might not be maintained in the CPU. But in this case they are both going via the same peripheral, so order will be maintained. Otherwise you would need a DSB to stop the read happening while the write was still in a buffer. (Or maybe even a readback to flush intermediate bridges).

The Linux kernel is very cautious about this sort of thing - all core I/O macros for accessing peripherals include a `DSB` just in case, to allow for synchronisation of RAM versus device, or two devices. Just using a `volatile uint32_t *` isn't enough in general.

But a DSB is a pretty big instruction in the T32 instruction set, and the M3 and M4 usually don't need it, so people don't generally bother. Only the M7 really needs to worry about it so much. (Eg DSBs between changing memory descriptors and poking the Ethernet controller)

Here though, I think it must be an artifact of your chip. If you look at the assembly, I expect the write and read are basically back-to-back with optimisation on, and the GPIO device is not responding to the changed input level in time for the read - maybe it just needed 1 more I/O clock to register the change.

I think assuming GPIO is totally instantaneous at CPU speed is probably optimistic.

I'd suggest modifying the test to have a tiny delay with `wait_ns`. That's a bit more to the point than the DSB.

LMESTM, 2019-09-12:
> I'd suggest modifying the test to have a tiny delay with wait_ns. That's a bit more to the point than the DSB.

I vote for this. 

I'd prefer NOT to have DSB by default here. I would assume that drivers that really need to make sure that GPIOs are set before next lines of code are executed will add their own barriers for ensuring this.

kjbracey-arm, 2019-09-12:
For inter-device DSBs, I think you either need them on absolutely every HW access in HAL or driver, just in case, or you need to defer them to the level that knows it's accessing two devices and that order between them is important.

That could be inside the HAL, inside an intermediate driver, or at application level.  Here it would be the application knowing it needs to order this DigitalOut and DigitalIn, so the DSB would go there.