ARMmbed/mbed-os #11482: Valgrind: 'Invalid read of size 1' in events-equeue unit test

kimlep01, 2019-09-13:
### Description

While running unit tests with Valgrind locally I noticed that Valgrind reports an error when running events-equeue test. The test gets stuck in `TestEqueue.test_equeue_multithread` but eventually continues and throws the following error:

```
==7825== Memcheck, a memory error detector
==7825== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==7825== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==7825== Command: ./events-equeue -o junit
==7825== 
==7825== Invalid read of size 1
==7825==    at 0x19C922: equeue_unqueue_by_id (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x19CC5C: equeue_cancel (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x19D412: equeue_chain_update (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x19C1E0: equeue_destroy (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x19237C: TestEqueue_test_equeue_chain_Test::TestBody() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1DFAF2: void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1D9C30: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1BA589: testing::Test::Run() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1BAEFD: testing::TestInfo::Run() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1BB57F: testing::TestCase::Run() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1C5877: testing::internal::UnitTestImpl::RunAllTests() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1E0CB8: bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==  Address 0x5dbf4cc is 220 bytes inside a block of size 2,048 free'd
==7825==    at 0x4C30D3B: free (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==7825==    by 0x19C224: equeue_destroy (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x19236D: TestEqueue_test_equeue_chain_Test::TestBody() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1DFAF2: void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1D9C30: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1BA589: testing::Test::Run() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1BAEFD: testing::TestInfo::Run() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1BB57F: testing::TestCase::Run() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1C5877: testing::internal::UnitTestImpl::RunAllTests() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1E0CB8: bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1DAA7E: bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1C42FF: testing::UnitTest::Run() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==  Block was alloc'd at
==7825==    at 0x4C2FB0F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==7825==    by 0x19BF78: equeue_create (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x19165D: TestEqueue_test_equeue_chain_Test::TestBody() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1DFAF2: void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1D9C30: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1BA589: testing::Test::Run() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1BAEFD: testing::TestInfo::Run() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1BB57F: testing::TestCase::Run() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1C5877: testing::internal::UnitTestImpl::RunAllTests() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1E0CB8: bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1DAA7E: bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825==    by 0x1C42FF: testing::UnitTest::Run() (in /builds/ws/unittest-valgrind-fork-tester/mbed-os/BUILD/unittests/events-equeue)
==7825== 
==7825== 
==7825== HEAP SUMMARY:
==7825==     in use at exit: 0 bytes in 0 blocks
==7825==   total heap usage: 556 allocs, 556 frees, 202,235 bytes allocated
==7825== 
==7825== All heap blocks were freed -- no leaks are possible
==7825== 
==7825== For counts of detected and suppressed errors, rerun with: -v
==7825== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)

```

Steps to reproduce:
clone mbed-os master in Linux
build and run unit tests: `mbed test --unittests`
cd to BUILD/unittests
Run event-equeue with Valgrind: `valgrind --track-origins=yes --leak-check=full --show-reachable=yes ./events-equeue`

Tested with:
Ubuntu 16 & Valgrind 3.11
Ubuntu 18 & Valgrind 3.13 (CI)

Related PR to mbed-os-ci to run this test in the future (after this is resolved):
https://github.com/ARMmbed/mbed-os-ci/pull/941

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-09-13:
cc @ARMmbed/mbed-os-core

Tharazi97, 2019-09-16:
The delay on multithread test that you've mentioned is probably caused by using usleep function in this test. It is not related with the problem with chain test. The Valgring error will be fixed by PR that I'll rise soon.

PR #11491 (2019-09-27): Equeue chain, add documentation of using equeue_destroy
### Description

If queues are chained, it is needed to unchain them first, before calling destroy, or call the destroy function on queues in order that chained queues are destroyed first. Destroying queues in wrong order would result in errors or other program malfunctions.
Fixes: #11482

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

@maciejbocianski @jamesbeyond @kimlep01 @geky @kjbracey-arm 
### Relea [... cut]
closed 2019-09-27 by adbridge