ARMmbed/mbed-os #11485: UNITTESTS - error: use of undeclared identifier 'errno'

ladislas, 2019-09-13:
### Description

On macOS Mojave (v10.14.6 - 18G95), from a fresh clone of [ARMmbed/mbed-os](https://github.com/ARMmbed/mbed-os) (f5dd7db8c7929276531008e924bdc28af3a96159), running:

```sh
$ mbed test --unittests
```

gives the following error:

```console
[ 66%] Building CXX object CMakeFiles/features-cellular-framework-AT-athandler.MbedOS.dir/Users/ladislas/dev/mbed/mbed-os/features/cellular/framework/AT/ATHandler.cpp.o
In file included from /Users/ladislas/dev/mbed/mbed-os/features/cellular/framework/AT/ATHandler.cpp:25:
In file included from /Users/ladislas/dev/mbed/mbed-os/UNITTESTS/../rtos/ThisThread.h:26:
In file included from /Users/ladislas/dev/mbed/mbed-os/UNITTESTS/../rtos/mbed_rtos_types.h:21:
/Users/ladislas/dev/mbed/mbed-os/UNITTESTS/target_h/cmsis_os2.h:66:9: warning: empty struct has size 0 in C, size 1 in C++ [-Wextern-c-compat]
typedef struct {
        ^
/Users/ladislas/dev/mbed/mbed-os/features/cellular/framework/AT/ATHandler.cpp:750:5: error: use of undeclared identifier 'errno'
    errno = 0;
    ^
/Users/ladislas/dev/mbed/mbed-os/features/cellular/framework/AT/ATHandler.cpp:753:55: error: use of undeclared identifier 'errno'
    if ((result == LONG_MIN || result == LONG_MAX) && errno == ERANGE) {
                                                      ^
1 warning and 2 errors generated.
gmake[2]: *** [CMakeFiles/features-cellular-framework-AT-athandler.MbedOS.dir/build.make:63: CMakeFiles/features-cellular-framework-AT-athandler.MbedOS.dir/Users/ladislas/dev/mbed/mbed-os/features/cellular/framework/AT/ATHandler.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1878: CMakeFiles/features-cellular-framework-AT-athandler.MbedOS.dir/all] Error 2
gmake: *** [Makefile:141: all] Error 2

Building unit tests failed.
[mbed] ERROR: "/usr/local/opt/python/bin/python3.7" returned error.
       Code: 2
       Path: "/Users/ladislas/dev/github/mbed-os"
       Command: "/usr/local/opt/python/bin/python3.7 /Users/ladislas/dev/mbed/mbed-os/UNITTESTS/mbed_unittest.py --build ./BUILD/unittests"
       Tip: You could retry the last command with "-v" flag for verbose output

```

I also tried the manual approach by running:

```bash
$ cd UNITTESTS && mkdir build && cd build
$ cmake ..
$ gmake
```

and the same issue arises.

On the other hand after installing gcc and make with homebrew and running the following:

```bash
$ cmake .. -DCMAKE_BUILD_TYPE=Debug -DCOVERAGE=True -DCMAKE_MAKE_PROGRAM=/usr/local/bin/gmake -DCMAKE_CXX_COMPILER=/usr/local/bin/g++-9 -DCMAKE_C_COMPILER=/usr/local/bin/gcc-9
$ gmake VERBOSE=1
$ ctest -v
```

works well and all tests pass with no issues.

I think the reason is that `mbed test --unittests` is using the default gcc/g++ on macOS which points to clang and that clang cannot build the tests.

So it is either a bug or we need to improve the documentation for macOS, ask users to install gcc through `brew install gcc` and pass the correct flags to cmake to point to the gcc@9.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-09-16:
cc @ARMmbed/mbed-os-test

0xc0170, 2019-09-16:
@ladislas As the documentation states, you should use GCC (not certain why you mentioned above clang)

https://os.mbed.com/docs/mbed-os/v5.13/tools/unit-testing.html - there's a mention of install gcc for macOS

ladislas, 2019-09-16:
@0xc0170 Yes, it is written but stated as optional.

> 6. (Optional) Install GCC with brew install gcc.

That being said, how do you pass the `-DCMAKE_CXX_COMPILER=/usr/local/bin/g++-9` flag to `mbed test --unittests`? 

The documentation states that tests should be run with the latter but only use CMake for advanced use.

> Use Mbed CLI to build and run unit tests. For advanced use, you can run CMake and a Make program directly.

I don't consider running macOS "advanced use", so I think the documentation should state that on macOS, only the advanced use, with a mandatory gcc install via brew, is possible.

Or we could change mbed-cli to check the OS, check if gcc is installed and output the correct error message if one of those requirements is not fulfilled.

I'm happy to work on a patch for mbed-cli if need be but slightly changing the documentation should do the work for now.

Tharazi97, 2019-09-16:
I had similiar problem on Windows with MinGW.
```
[  2%] Building CXX object CMakeFiles/features-cellular-framework-AT-athandler.MbedOS.dir/C_/Users/int_szyk/Documents/GitHub/mbed-os/features/cellular/framework/AT/ATHandler.cpp.obj
C:\Users\int_szyk\Documents\GitHub\mbed-os\features\cellular\framework\AT\ATHandler.cpp: In member function 'void mbed::ATHandler::write_int(int32_t)':
C:\Users\int_szyk\Documents\GitHub\mbed-os\features\cellular\framework\AT\ATHandler.cpp:1397:48: error: expected ')' before 'PRIi32'
     int32_t result = sprintf(number_string, "%" PRIi32, param);
                             ~                  ^~~~~~~
                                                )
mingw32-make[2]: *** [CMakeFiles\features-cellular-framework-AT-athandler.MbedOS.dir\build.make:63: CMakeFiles/features-cellular-framework-AT-athandler.MbedOS.dir/C_/Users/int_szyk/Documents/GitHub/mbed-os/features/cellular/framework/AT/ATHandler.cpp.obj] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:1980: CMakeFiles/features-cellular-framework-AT-athandler.MbedOS.dir/all] Error 2
mingw32-make: *** [Makefile:140: all] Error 2
```

The quick fix that i made to make it work is to change this line of code 
```int32_t result = sprintf(number_string, "%" PRIi32, param);```
to this:
```int32_t result = sprintf(number_string, "%", WIN32, param);```

0xc0170, 2019-09-18:
> I had similiar problem on Windows with MinGW.

this should work, mingw should include proper types to get that resolved.

> I'm happy to work on a patch for mbed-cli if need be but slightly changing the documentation should do the work for now.

Please do!

ladislas, 2019-09-18:
> Please do!

@0xc0170 which one? ;) The documentation or a patch on mbed-cli?

timovanasten, 2019-10-23:
Also having this error on macOS Catalina. I have gcc installed, but I've not been able to make  ```mbed test --unittests``` use gcc. Any updates on this?

0xc0170, 2019-10-23:
@timovanasten Even setting manual path for GCC as in : https://os.mbed.com/docs/mbed-os/v5.14/tools/manual-installation.html does not help?

timovanasten, 2019-10-23:
@0xc0170 Thanks for the quick reply. You mean similar to GCC_ARM_PATH? I don't see a option for that in the link you mentioned, but maybe I'm understanding you incorrectly.

After running `mbed test --unittests It gives the following information  about the compiler:

```- The C compiler identification is AppleClang 11.0.0.11000033
-- The CXX compiler identification is AppleClang 11.0.0.11000033
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/g++
-- Check for working CXX compiler: /usr/bin/g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done 
```

The build then starts and errors at the same spot as @ladislas.

If I install a older version off gcc using `brew install gcc@8` it will use it instead of Clang but then errors at:

```
/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/strings.h:89:23: error: expected initializer before '__OSX_AVAILABLE_STARTING'
 int  flsll(long long) __OSX_AVAILABLE_STARTING(__MAC_10_9, __IPHONE_7_0);
``` 
similar to [here](https://stackoverflow.com/questions/47247099/mac-os-high-sierra-os-upgrade-fails-make-in-c) and [here](https://github.com/Homebrew/homebrew-core/issues/44579) but the suggested solutions there just lead to the same problem.

ladislas, 2019-10-23:
@timovanasten does using cmake with the following solves the error?

```
$ cmake .. -DCMAKE_BUILD_TYPE=Debug -DCOVERAGE=True -DCMAKE_MAKE_PROGRAM=/usr/local/bin/gmake -DCMAKE_CXX_COMPILER=/usr/local/bin/g++-9 -DCMAKE_C_COMPILER=/usr/local/bin/gcc-9
$ gmake VERBOSE=1
$ ctest -v
```

ladislas, 2019-10-23:
On macOS, if you want to use the brew gcc version, you need to call `gcc-9` and not `gcc`, the later referring to clang's gcc.

timovanasten, 2019-10-23:
@ladislas I was guessing so, but seems that there is no way to control it using the `mbed test --unittests` command which is unfortunate.

timovanasten, 2019-10-23:
@ladislas I tried your command, but unfortunately didn't work, although it did use GNU 9 this time

```
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- Checking whether C compiler has -isysroot
-- Checking whether C compiler has -isysroot - yes
-- Checking whether C compiler supports OSX deployment target flag
-- Checking whether C compiler supports OSX deployment target flag - yes
-- Check for working C compiler: /usr/local/bin/gcc-9
CMake Error: Generator: execution of make failed. Make command was: /usr/local/bin/gmake cmTC_7f04f/fast &&
-- Check for working C compiler: /usr/local/bin/gcc-9 -- broken
CMake Error at /usr/local/Cellar/cmake/3.15.4/share/cmake/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/usr/local/bin/gcc-9"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/Timo/Documents/Hyperloop/Software/dh04-nucleo/UNITTESTS/build/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/local/bin/gmake cmTC_7f04f/fast && No such file or directory
    Generator: execution of make failed. Make command was: /usr/local/bin/gmake cmTC_7f04f/fast &&

  CMake will not be able to correctly generate this project.
```

ladislas, 2019-10-23:
@timovanasten you also need to `brew install gmake` or remove it from the first command

timovanasten, 2019-10-23:
@ladislas I tried that but couldn't get it to work. What did your CMakeLists.txt file in the UNITTESTS folder look like (assuming you had one in there)?

ladislas, 2019-10-23:
I just used the one in the UNITTESTS folder.

https://github.com/ARMmbed/mbed-os/blob/master/UNITTESTS/CMakeLists.txt

and followed the following:

https://github.com/ARMmbed/mbed-os/tree/master/UNITTESTS#build-tests-directly-with-cmake

mirelachirica, 2019-10-31:
The error stated in description:

"/Users/ladislas/dev/mbed/mbed-os/features/cellular/framework/AT/ATHandler.cpp:750:5: error: use of undeclared identifier 'errno'"

was addressed and merged in:

https://github.com/ARMmbed/mbed-os/commit/dc64165f340d4cbf0c4c88e42cc6f08d26bb865f

Can this issue be closed?

closed 2019-10-31 by 0xc0170