ARMmbed/mbed-os #11492: Power consumption with NSAPI_PPP_AVAILABLE 1 (Cellular PPP mode)

JRubis, 2019-09-16:
### Description

Board: STM32L4-discovery
Cell Module: Quectel BG96 connected via STMod+ connector 

I continue to investigate low power operation on this board with a pretty simple use case. Using AT mode (NSAPI_PPP_AVAILABLE set to 0) I have good results of around ~15uA once I enter my long sleep loop. However if I enable PPP mode (NSAPI_PPP_AVAILABLE set to 1) I see much higher power consumption of around ~200uA during my long sleep loop. Using a logic analyzer it looks like something waking up the device every ~100mS for a short period of time. 

It seems something related to PPP isn't being shutdown once I disconnect the network and power down the modem. Maybe I am missing an API call or something? 

Here is my code, which is loosely based on the cellular example.

```
int main() {
    	
	/* Disable input on STDIN to allow MCU to enter deep sleep */
	mbed_file_handle(STDIN_FILENO)->enable_input(false);

   	CellularDevice *device = CellularDevice::get_default_instance();

    iface = CellularContext::get_default_instance();

	if(iface != nullptr)
	{
		// sim pin, apn, credentials and possible plmn are taken automatically from json when using NetworkInterface::set_default_parameters()
		iface->set_default_parameters();
		iface->add_event_listener(status_callback);
		
			while (1) 
			{
				/* Attempt to connect to the cellular network */
				if (do_connect() == NSAPI_ERROR_OK) 
				{
				  	print_function("\n\n Connected to the network.");
					
					/* Network disconnect */
					if (iface->disconnect() == NSAPI_ERROR_OK) 
					{
					  	print_function("\n\n Disonnected from the network.");
					}					
					else
					{
					  	print_function("\n\n disconnect failed.");
					}
					
					/* Power down the cell module */
					device->soft_power_off();
				}
				else
				{
				  	print_function("\n\n Network connection failed.");
				}

				print_function("\n\n Going to sleep.");
				fflush(stdout);
						
				/* Sleep for a long while */
				ThisThread::sleep_for(60000*60);
		   }
		}

    return 0;
}
```

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-09-16:
cc @ARMmbed/mbed-os-wan

AriParkkila, 2019-09-19:
@JRubis That could be due to LWIP [tcpip_thread](https://github.com/ARMmbed/mbed-os/blob/4f7339af78c59fd343cad0d54f2b5bae5dd5862e/features/lwipstack/lwip/src/api/lwip_tcpip.c#L127) is never quit.

JRubis, 2019-09-20:
@AriParkkila it does seems that its related to a timer in the tcpip_thread. Once a call is complete and tcpip is no longer needed is there a way to suspend this task until the next call?

closed 2019-10-25 by ciarmcom

felser, 2020-11-25:
I'm wondering if this topic has gone an further. I couldn't find anything on the mbed forum. I am seeing the same issue. I have also run into an issue where doing the same as above but with only 5s of sleep, the target frequently crashes with a tcpip stack overflow. That's with a 4096 stack size. I'd like to know if @JRubis has found a work around. You can email me direct at llindenfelser@multitech.com
Thank you,
Leon

Arjun765, 2022-02-09:
This is adding about 520uW of extra power during deep sleep, while this is insignificant i do not like how my deep sleeping controller is waking up every 100ms or so. One workaround is to misuse the `LWIP_TCPIP_THREAD_ALIVE()` function. Use it to halt the thread if networking has been shutdown.

john-rubis, 2022-02-09:
Unfortunately, I am no longer working on the project that drove this issue, so I won't be much help. To my knowledge, we never did find a decent workaround and the project moved in a different direction after I left the company.