ARMmbed/mbed-os #11501: Deepsleep issue since version mbed-os 5.12

elforce, 2019-09-17:
Hi 
I have a project with STM32L432KC microcontroller created in mbed-os version 5.11.2. The device sends data and then goes to sleep for eg. 1h. Power consumption in deepsleep mode approx. 4uA. "lpticker_lptim:" 0 to use RTC wakeup timer and "lpticker_delay_ticks": 4. After upgrading mbed-os to version 5.13.4 the program does not enter deepsleep mode at all. I have downgraded the mbed-os to version to 5.12.4 and the same problem occurs.
After that I downloaded latest mbed-os-example-blinky with version mbed-os 5.12.0. The effect was the same. Below are the statistics from mbed_stats_cpu_t after the program has been running for some time:
[image]
[image]
After that I changed mbed-os in blinky to version 5.11.2 from my project. Program enters deep-sleep mode without any problems. Below are the statistics from mbed_stats_cpu_t after the program ran for some time for mbed-os 5.11.2:
[image]
[image]
What has changed in mbed-os since version 5.12.0? Maybe someone from ST would suggest a solution to this issue?

0xc0170, 2019-09-18:
Have you checked the release notes for the last 2 releases if there are any relevant changes that could affect your app ? They could narrow down the issue.

What is your configuration, are you using tickless?

cc @ARMmbed/team-st-mcd

elforce, 2019-09-18:
I reviewed the release notes and found nothing special.
In the targets.json file for NUCLEO_L432KC in "macros_add" field there is the macro "MBED_TICKLESS". I did not modify this file for the tested versions: 5.11.2 and 5.12.4 and 5.13.4 where the problem occurred.
[image]
The configuration for the blinky on which I did the tests is standard. I just changed the UART baudrate to 115200, convert-newlines and changed the pins for UART_TX and UART_RX to PA_9 and PA_10
[image]

elforce, 2019-09-19:
I did more tests and I have interesting conclusions.
Due to the fact that every newly created program in mbed studio with mbed-os does not go into deep-sleep mode, I went to the online compiler. I loaded blinky with mbed-os 5.13.1 and the program enters deep-sleep mode without any problems. I tested for NUCLEO_L476RG, NUCLEO_L432KC and NUCLEO_WB55RG. I did the mbed-os upgrade to 5.13.4 and everything works too.
Therefore, I made fork of working program and Published it to my account. From mbed studio I created a new program from the URL of the program that worked in mbed compiler. After compiling and running the program on each of the three platforms, the program does not enter deep sleep mode.
Further I created an empty program in mbed studio and copied the program exported from mbed compier that worked and the same problem. Deep sleep does not work.
Programs exported from mbed compiler in the mbed-os directory have only sources for the given platform. Each program created in mbed studio has a complete mbed-os. Maybe this is the key. There is a problem because what was created on the online compiler does not work the same way in deep sleep mode as in mbed studio above mbed-os 5.11.2

0xc0170, 2019-09-19:
Can you summarize all possibilities you tested (deep sleep work vs do not work), I am slightly confused about the latest update.

LMESTM, 2019-09-19:
FYI 
On the 5.14 / master branch there definitely is an issue that is being discussed in #11509 and #11522
When it comes to mbed studio, one question is whether MBED_DEBUG is active (e.g. debug profile being used) when using mbed studio ? 
Because in debug profiles deep sleep is disabled  ...

elforce, 2019-09-19:
**Deep sleep does not work for**
blinky program https://github.com/ARMmbed/mbed-os-example-blinky loaded into **mbed studio** when the mbed-os version is higher than 5.11.2.
**Deep sleep works for**
blinky program loaded into the **mbed online compiler** with every version of mbed-os.

I do tests on blinky on purpose to choose the simplest possible program, because the behaviour is identical on every other program.

I refer this problem only to ST microcontrollers because I work only on such. The behavior is identical for three platforms NUCLEO_L476RG, NUCLEO_L432KC and NUCLEO_WB55RG.

elforce, 2019-09-19:
@LMESTM after what you wrote, I even commented the lines of code that do not allow to go into deep sleep in the file mbed_sleep_manager.c
[image]
It didn't change anything. The program does not enter deep sleep.

elforce, 2019-09-20:
The mbed studio 0.6.1 version has been released today. All programs that did not want to enter deep sleep for the mbed-os version above 5.11.2 began to work properly after compilation from the mbed studio 0.6.1 version. The problem has therefore been solved.

0xc0170, 2019-09-20:
@elforce This can be closed ?

elforce, 2019-09-23:
Yes @0xc0170. You can close it.

closed 2019-10-02 by adbridge