ARMmbed/mbed-os #11503: NUCLEO_F303RE fails tests-mbed_platform-wait_ns when tickless is enabled

gpsimenos, 2019-09-17:
### Description

When enabling tickless mode (by adding the MBED_TICKLESS macro) on the NUCLEO_F303RE, the tests-mbed_platform-wait_ns greentea test fails.

```[1568741320.60][GLRM][INF] remote resources reset...
[1568741324.00][GLRM][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1568741324.00][CONN][INF] sending up to 5 __sync packets (specified with --sync=5)
[1568741324.00][CONN][INF] sending preamble '1c2d92fd-e402-42f1-adc0-1a3146e37cfd'
[1568741324.17][GLRM][TXD] {{__sync;1c2d92fd-e402-42f1-adc0-1a3146e37cfd}}
[1568741324.22][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1568741324.31][CONN][INF] found SYNC in stream: {{__sync;1c2d92fd-e402-42f1-adc0-1a3146e37cfd}} it is #0 sent, queued...
[1568741324.32][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1568741324.32][CONN][INF] found KV pair in stream: {{__timeout;15}}, queued...
[1568741324.32][HTST][INF] sync KV found, uuid=1c2d92fd-e402-42f1-adc0-1a3146e37cfd, timestamp=1568741324.314810
[1568741324.32][HTST][INF] DUT greentea-client version: 1.3.0
[1568741324.32][HTST][INF] setting timeout to: 15 sec
[1568741324.42][CONN][RXD] >>> Running 2 test cases...
[1568741324.42][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1568741324.42][HTST][INF] host test class: '<class 'mbed_os_tools.test.host_tests.default_auto.DefaultAuto'>'
[1568741324.42][HTST][INF] host test setup() call...
[1568741324.42][HTST][INF] CALLBACKs updated
[1568741324.42][HTST][INF] host test detected: default_auto
[1568741324.52][CONN][RXD]
[1568741324.52][CONN][INF] found KV pair in stream: {{__testcase_name;Test: wait_ns - compare with lp_timer 1s}}, queued...
[1568741324.52][CONN][INF] found KV pair in stream: {{__testcase_name;Test: wait_ns - compare with us_timer 1s}}, queued...
[1568741324.62][CONN][RXD] >>> Running case #1: 'Test: wait_ns - compare with lp_timer 1s'...
[1568741324.72][CONN][INF] found KV pair in stream: {{__testcase_start;Test: wait_ns - compare with lp_timer 1s}}, queued...
mbedgt: :80::FAIL: Expected 1.200000 Was 1.626831
[1568741326.44][CONN][RXD] :80::FAIL: Expected 1.200000 Was 1.626831
[1568741326.44][CONN][INF] found KV pair in stream: {{__testcase_finish;Test: wait_ns - compare with lp_timer 1s;0;1}}, queued...
[1568741326.53][CONN][RXD] >>> 'Test: wait_ns - compare with lp_timer 1s': 0 passed, 1 failed with reason 'Assertion Failed'
[1568741326.53][CONN][RXD]
[1568741326.63][CONN][RXD] >>> Test cases: 0 passed, 1 failed with reason 'Assertion Failed'
[1568741326.63][CONN][RXD] >>> TESTS FAILED!
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-09-18:
cc @ARMmbed/team-st-mcd

LMESTM, 2019-09-18:
The reason tickless mode has not been enabled for F3 devices is that some of the test cases are failing with TICKLESS.  Supporting Tickless for F3 is a future enhancement. If you find out the reason why few tests are failing, this would help us making this happen earlier.

jeromecoutant, 2019-09-19:
Could we close this issue ?
Thx

0xc0170, 2019-09-19:
> Could we close this issue ?

Isn't this valid or how to proceed? As it is not enabled, there is a reason and it's known, thus this is not a bug but.

jeromecoutant, 2019-10-02:
Adding TICKLESS for targets which are not designed as low power platform is not planned.

@MarceloSalazar

jeromecoutant, 2019-10-11:
@gpsimenos Is it OK to close the issue ?
Thx

gpsimenos, 2019-10-11:
Tickless could enable great power savings in this platform but if there are no plans to ever look into the issue, I suppose you can close it.