ARMmbed/mbed-os #11505: Cellular: clear error

pilotak, 2019-09-18:
When using ublox G350 in AT command mode it fails to connect to the network. Specifically in newly introduced function `clear()`

https://github.com/ARMmbed/mbed-os/blob/8f1bf433351d8101dc5691e92ee41fa3b6c8ef8f/features/cellular/framework/AT/AT_CellularNetwork.cpp#L635

```
[INFO][CELL]: Modem ready
[INFO][CELL]: AT_CellularNetwork::clear
[INFO][CELL]: AT TX (11): AT+CGDCONT?
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 8): <cr><ln>+CME E
[INFO][CELL]: AT RX ( 9): RROR: 3<cr><ln>
[WARN][CELL]: AT error code 3
[DBG ][CELL]: AT error -3012
[WARN][CELL]: CellularDevice clear failed
```

same thing happes in setting up
https://github.com/ARMmbed/mbed-os/blob/8f1bf433351d8101dc5691e92ee41fa3b6c8ef8f/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT.cpp#L123
 again it's related to PDP and fails, fortunatelly `UBLOX_AT::init()` can be overriden but still it's not ideal

0xc0170, 2019-09-18:
@pilotak What is an issue here - is it generic problem in cellular network or ublox related?

0xc0170, 2019-09-18:
cc @ARMmbed/team-ublox

pilotak, 2019-09-18:
Hard to judge if this is an ublox related or generic. It happens on ublox but error is in `AT_CellularNetwork`

Second part of issue if definitely ublox related it was discussed here https://github.com/ARMmbed/mbed-os/issues/11170#issuecomment-519836084 but with no thoughts this PDP issue

fahim-ublox, 2019-09-18:
@pilotak can you please attached full logs ? It will be helpful to narrow down. Meanwhile we are trying to reproduce the issue. 

Internal reference ticket PGM_NO_MBD-281 and PGM_NO_MBD-282.

pilotak, 2019-09-18:
```
[INFO][CELL]: New CellularContext  (0x200082a8)
[INFO][CELL]: CellularContext set blocking 0
[INFO][CELL]: CellularContext connect
[DBG ][CELL]: callback: 4106, err: 0, data: 4096
[INFO][CELL]: Start connecting (timeout 1000 ms)
[INFO][CELL]: AT TX ( 2): AT
[INFO][CELL]: AT TX ( 1): <cr>
[DBG ][CELL]: AT error -3012
[INFO][CELL]: AT TX ( 2): AT
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 2): AT
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[DBG ][CELL]: Init => Device ready
[DBG ][CELL]: callback: 4106, err: 0, data: 4096
[DBG ][CELL]: AT flush
[INFO][CELL]: AT TX ( 2): AT
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 2): AT
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[INFO][CELL]: AT TX ( 4): ATE0
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 2): AT
[INFO][CELL]: AT RX ( 2): E0
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[INFO][CELL]: AT TX ( 9): AT+CMEE=1
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 5): <ln>OK<cr><ln>
[INFO][CELL]: AT TX ( 9): AT+CFUN=1
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[INFO][CELL]: Modem ready
[INFO][CELL]: AT_CellularNetwork::clear
[INFO][CELL]: AT TX (11): AT+CGDCONT?
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX (16): <ln>+CME ERROR: 3<cr><ln>
[WARN][CELL]: AT error code 3
[DBG ][CELL]: AT error -3012
[WARN][CELL]: CellularDevice clear failed
[WARN][CELL]: Power cycle CellularDevice and restart connecting
[INFO][CELL]: AT TX (10): AT+CPWROFF
[INFO][CELL]: AT TX ( 1): <cr>
[DBG ][CELL]: Device ready => Init
[DBG ][CELL]: callback: 4106, err: 0, data: 4096
[INFO][CELL]: Start connecting (timeout 1000 ms)
[INFO][CELL]: AT TX ( 2): AT
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 5): <cr><ln>OK<cr>
[INFO][CELL]: AT RX ( 1): <ln>
[INFO][CELL]: AT TX ( 2): AT
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 4): <cr><ln>OK
[INFO][CELL]: AT RX ( 2): <cr><ln>
[DBG ][CELL]: Init => Device ready
[DBG ][CELL]: callback: 4106, err: 0, data: 4096
[WARN][CELL]: URC already added with prefix: +CGEV: NW DEACT
[WARN][CELL]: URC already added with prefix: +CGEV: ME DEACT
[WARN][CELL]: URC already added with prefix: +CGEV: NW PDN D
[WARN][CELL]: URC already added with prefix: +CGEV: ME PDN D
[DBG ][CELL]: AT flush
[INFO][CELL]: AT TX ( 2): AT
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 3): <cr><ln>O
[INFO][CELL]: AT RX ( 3): K<cr><ln>
[INFO][CELL]: AT TX ( 4): ATE0
[... 106 lines cut ...]
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[INFO][CELL]: AT TX ( 4): ATE0
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 2): A\
[INFO][CELL]: AT RX ( 3): E0<cr>
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 5): <ln>OK<cr><ln>
[INFO][CELL]: AT TX ( 9): AT+CMEE=1
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 5): <cr><ln>OK<cr>
[INFO][CELL]: AT RX ( 1): <ln>
[INFO][CELL]: AT TX ( 9): AT+CFUN=1
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[INFO][CELL]: Modem ready
[INFO][CELL]: AT_CellularNetwork::clear
[INFO][CELL]: AT TX (11): AT+CGDCONT?
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX (10): <cr><ln>+CME ERR
[INFO][CELL]: AT RX ( 7): OR: 3<cr><ln>
[WARN][CELL]: AT error code 3
[DBG ][CELL]: AT error -3012
[WARN][CELL]: CellularDevice clear failed
```

AriParkkila, 2019-09-19:
@pilotak This is fixed in PR #11469, waiting for merge now.

pilotak, 2019-09-19:
@AriParkkila that only disables `clear` function but doesn't fix it. Meanwhile @fahim-ublox fixed it, thanks a lot

pilotak, 2019-09-19:
@fahim-ublox second part of my issue is when using G350 but not on C027 `init` automatically goes to https://github.com/ARMmbed/mbed-os/blob/8f1bf433351d8101dc5691e92ee41fa3b6c8ef8f/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT.cpp#L111 which is not supported on G350 . Should there not be a better way of handling modem specific code? at the moment it's tied to dev board only. In my case i have a STM32F412 & F303 with G350 and i have to override a lot function from ublox but if i could somehow specify which modem i have regardless the processor it would be more generic. And of course when i override it i can't use `UBLOX_AT::config_authentication_parameters` because it's private

Do you want me to create another issue? or is it ok to continue here

fahim-ublox, 2019-09-19:
@pilotak 
First if it is OK for you to specify your target in targets.json such that you inherit from C027. Like what we did with UBLOX_C030_R412M  https://github.com/ARMmbed/mbed-os/blob/4f7339af78c59fd343cad0d54f2b5bae5dd5862e/targets/targets.json#L4889
We are using same MCU on different C030s variant and while you have a different MCU for custom board. What you suggest about it ? 

Second point, in any case #else in init method is a bug and need proper usage while you have created https://github.com/ARMmbed/mbed-os/issues/11170, fix can be added against that issue.

pilotak, 2019-09-19:
If i specify my custom target with inherit from C027 my target macro becomes `TARGET_MY_CUSTOM_TARGET` which would not help.
I noticed the thing about C030 and inherit, it's perfect. This is how it meant to be used but as you say MCU stays the same which is not my case.

I realise it would be complex change but it would be nice if you specify let's say in mbed_app.json that you have G350, N211, R410, R412 etc. and the lib will adjust itself regardless MCU

fahim-ublox, 2019-10-01:
@pilotak My team member has generated a pull request. Can you please close this ticket if you are happy now ?

closed 2019-10-01 by pilotak