ARMmbed/mbed-os #11506: STM32-F429: Ethernet driver possibly wasting RAM

Arjun765, 2019-09-18:
### Description

When i ran `nm` on my elf, i saw 2 variables Rx_buff and Tx_buff (from stm32xx_emac.cpp) which was using up 6096 bytes each, even though I'm not using ethernet at all. I could not find any way to disable them, best i could do is override `stm32-emac.eth-txbufnb` parameter from 4 to 1 bringing down memory to 1524 bytes.

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

0xc0170, 2019-09-18:
@Arjun765 Can you share more? Is this in the final binary? 

Looking at emac implementation, the buffers are on stack or I've missed something?

cc @ARMmbed/team-st-mcd Please review

Arjun765, 2019-09-18:
>Can you share more? Is this in the final binary?

Yes, its the final ELF file.

>Looking at emac implementation, the buffers are on stack or I've missed something?

These are not (stm32xx_emac.cpp line 56 and 61)
`__ALIGN_BEGIN uint8_t Tx_Buff[ETH_TXBUFNB][ETH_TX_BUF_SIZE] __ALIGN_END;`
`__ALIGN_BEGIN uint8_t Rx_Buff[ETH_RXBUFNB][ETH_RX_BUF_SIZE] __ALIGN_END;`

0xc0170, 2019-09-18:
Whats your toolchain (assuming gcc arm) and the compile command - mainly what profile you use?

LMESTM, 2019-09-18:
NUCLEO_F429ZI has 
`"network-default-interface-type": "ETHERNET"`
Maybe overriding this value from mbed_app.json could disable the Ethernet ?
`"target.network-default-interface-type": null`
Laurent

Arjun765, 2019-09-18:
I can reproduce this on a new empty project with `mbed-os-5.13.4`, `GCC 8-2019-q3-update` compiled with `mbed compile --target NUCLEO_F429ZI --toolchain GCC_ARM --clean`

0xc0170, 2019-09-18:
Can you try `--profile release` as well?

Arjun765, 2019-09-18:
> Can you try `--profile release` as well?

Nope, still there. I don't think it will be removed without flto.

0xc0170, 2019-09-18:
That one is still due to be added. As I remember, we need to first update to GCC 8 , it should happen the following releases as we had some build issues prior the current one.

0xc0170, 2019-09-18:
A workaround would be : "https://github.com/ARMmbed/mbed-os/issues/11506#issuecomment-532582145" - does that work for you?

kjbracey-arm, 2019-09-18:
Personally, I don't see any reason those buffer arrays should be retained. A plain C-style array should vanish during linking if no-one uses it. Lots of the codebase relies on that.

Not sure what's happening here.

Maybe try literally deleting the definitions, leaving just `extern uint8_t RxBuff[];` declarations. With GNU ld, if it thinks someone is referencing it, it will fail to link and tell you so.

jeromecoutant, 2019-09-19:
Hi

> Maybe overriding this value from mbed_app.json could disable the Ethernet ?
> "target.network-default-interface-type": null

You have to remove also STM_EMAC from extra_labels, and EMAC from device_has...

            "target.extra_labels_remove": ["STM_EMAC"],
            "target.device_has_remove": ["EMAC"],

SeppoTakalo, 2019-09-24:
STM32 Emac driver seem to have arrays of structs: https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/emac-drivers/TARGET_STM_EMAC/stm32xx_emac.cpp#L54 in the global namespace.

Struct is a class in C++ and therefore have constructor.
And this constructor initialises the object and therefore refers to it... so the linker cannot drop.
`SingletonPtr` might help.

kjbracey-arm, 2019-09-26:
> Struct is a class in C++ and therefore have constructor.

In C++ it should still be dropped if the constructor and destructor are trivial or constant. For a simple C-like structure, the implicit constructor is trivial.

It looks like `ETH_DMADescTypeDef` only contains `uint32_t`s so, that's trivial. Only oddity I can see is that `Status` is `volatile` (yuck), but that shouldn't be sufficient to upset it.

kjbracey-arm, 2019-09-26:
C++11 complicates the rules.

Trivial covers "zero-init", so this can be eliminated, in C++98 or C++11:

     struct Foo {
         int a;
         int b;
     };

     Foo array[10];

C++11 adds constant initialisation, so in C++11 this can be eliminated

     struct Foo {
         constexpr Foo() a(2), b(4) { }
         int a;
         int b;
     };
     // or, equivalently
     struct Foo {
         int a = 2;
         int b = 4;
     };

     Foo array[10];

But you can't mix. If it's not zero or constant, it goes back to being treated as dynamic:

     struct Bar {
        int a = 4;
     };

     struct Foo {
         Bar a;
         int b;
     };

     Foo array[10];

That can't be eliminated, because Foo is neither zero-init nor constant-init.

closed 2019-10-31 by ciarmcom