ARMmbed/mbed-os #11508: LSI as RTC clock source on certain STM targets

Funky185540, 2019-09-18:
### Description

Currently it is possible through a switch in *mbed_app.json* (specific to the `FAMILY_STM32` of devices) to control whether to use *LSE* or *LSI* as clock source for the RTC.

Today however I realized that configuring `"lse_available" : 0` in the overrides section of the STM32H743ZI will cause the build to fail in **mbed_overrides.c** at line 63 with the definition of `RCC_RTCCLKSOURCE_NO_CLK` seemingly missing. Inspecting the codes of other STM32XY boards revealed that this is defined in **stm32xy_hal_rcc.h**, but this is seemingly missing for the H7.
Thus, my first question is:

May I go ahead and simply create a merge request adding the required line directly into this HAL file or does the H7 deploy another mechanism/symbol for that?

On a sidenote I also found out that the definition of `RCC_RTCCLKSOURCE_NO_CLK` is absent in the HAL files of the STM32L4 and STM32WB family of devices as well. 
For the STM32L4 family this is solved in the respective **stm32_hal_legacy.h**, in line 2805 where it says:
```
#if defined(STM32L4)
#define RCC_RTCCLKSOURCE_NO_CLK     RCC_RTCCLKSOURCE_NONE
```
whereas the STM32WB family of devices doesn not have a section like this.
So I went ahead and started a compilation after adapting **targets.json** to NOT use the LSE for a STM32WB target which resulted in a compilation error in file **mbed_overrides.c** with the same error as for the H7: No definition of `RCC_RTCCLKSOURCE_NO_CLK` present.

Which leads me to my second question:

In the same or another merge request, should I expand the line 2805 (`#if defined(STM32L4)`) to something like `#if defined(STM32L4) || defined(STM32WB)`?  
Because as far as I could determine the **stm32wbxx_hal_rcc.h** does contain a definition of `RCC_RTCCLKSOURCE_NONE`, but `RCC_RTCCLKSOURCE_NO_CLK` is required by **mbed_overrides.c**.

Or am I missing something here?

### Further information
- Related Targets: STM32H7, STM32L4, STM32WB
- Toolchain I tested with: GCC_ARM, Version 6. Release Q2 2017
- Tools I used: *mbed-cli*, Version 0.22.0

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-09-18:
cc @ARMmbed/team-st-mcd Please review

jeromecoutant, 2019-09-19:
Hi
I confirm the compilation issues when lse_available is set to 0

PR #11525 (2019-09-20): STM32WB/STM32H7 : LSI selection when LSE is not available
### Description

When LSE is not available, LSI selection was not possible due to build issues

PR fixes #11508 

@Funky185540

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

### Release Notes
closed 2019-09-20 by 0xc0170